espressif/esp-adf #747: ADF环境安装完成后编译出现问题

CYCHEN97, 2022-01-12:
[image]
ADF安装好后进行编译自动语音识别demo时，提示python版本问题，还没安装ADF之前linux中只有python3，安装好ADF后多了几个python版本
[image]

[image]
ADF安装成功可以在VSCODE显示example

[image]
但是点这个配置后提示错误

CYCHEN97, 2022-01-12:
@HengYongChao
麻烦帮忙看一下

CYCHEN97, 2022-01-13:
@HengYongChao
@jason-mao 
@binartech
@espressif-zhanghu
@krzychb
@ahhfzhang
你好，我用装好的这个环境编译了MP3的example是没问题的，也可以正常播放音乐，但是编译配置ASR语音识别这个example时就出现问题了，麻烦帮我看一下这是怎么回事，谢谢
[image]

jason-mao, 2022-01-13:
@CYCHEN97 建议你先把 IDF 升级到 release/4.4，同时使用最新ADF master里面的wwe example 体验ASR 的功能。

CYCHEN97, 2022-01-13:
@jason-mao
IDF不需要使用ADF下的那个吗？

jason-mao, 2022-01-13:
> @jason-mao
> IDF不需要使用ADF下的那个吗？
ADF下IDF 只是指定了版本，可以在里面切换到其他分支。参考example README, 
https://github.com/espressif/esp-adf/blob/master/examples/speech_recognition/wwe/README_CN.md#编译和下载

CYCHEN97, 2022-01-13:
@jason-mao
ADF用master版本吗？

CYCHEN97, 2022-01-13:
[image]

CYCHEN97, 2022-01-13:
@jason-mao
ADF更新到master，ADF下的IDF更新到4.4版本后，编译那个wee例程成功了，但是跑到开发板上出问题了，开发板用的是ESP32-LyraTD-MSC_A V2.2
[image]
还有就是默认分区太小了，生成的.bin文件有3.2M，但是那个分区表默认只给了0x300000，我增大了那个程序分区后才能烧写程序到板子上
[image]

jason-mao, 2022-01-13:
@CYCHEN97 你板子的批次比较久远，内存使用上有些限制，建议只开wakenet，不要开multi net了。另外建议你使用的S3开发板体验效果会更佳。

CYCHEN97, 2022-01-13:
@jason-mao 
关闭wakenet，只开启wakenet
[image]
然后还是一样的问题
[image]

CYCHEN97, 2022-01-13:
@jason-mao

CYCHEN97, 2022-01-13:
@jason-mao
跑ASR的例程就一直在复位
[image]

CYCHEN97, 2022-01-14:
@ jason-mao
支持了就跟我说一下哈

jason-mao, 2022-01-14:
@CYCHEN97 请基于这个ID测试 34af12428ea6fa691e9b2cf9259cd86b1c62c0c2,

closed 2022-02-07 by jason-mao