espressif/esp-adf #748: ADF的asr工程烧录进去后一直重启

Zhengqinze05, 2022-01-14:
刚更新的adf版本，使用的是esp32_s3_korvo2_v3板子
编译没有问题，烧录之后串口工具显示板子一直在重启。
这是串口log

Rebooting...
ESP-ROM:esp32s3-20210327
Build:Mar 27 2021
rst:0xc (RTC_SW_CPU_RST),boot:0x9 (SPI_FAST_FLASH_BOOT)
Saved PC:0x40375960
SPIWP:0xee
mode:DIO, clock div:1
load:0x3fcd0108,len:0x1718
load:0x403b6000,len:0x8c8
load:0x403ba000,len:0x2eb0
entry 0x403b61c0
[0;32mI (29) boot: ESP-IDF v4.4-dev-3401-gb86fe0c66c-dirty 2nd stage bootloader[0m
[0;32mI (29) boot: compile time 14:45:40[0m
[0;32mI (29) boot: chip revision: 0[0m
[0;32mI (33) boot.esp32s3: Boot SPI Speed : 80MHz[0m
[0;32mI (38) boot.esp32s3: SPI Mode       : DIO[0m
[0;32mI (42) boot.esp32s3: SPI Flash Size : 8MB[0m
[0;32mI (47) boot: Enabling RNG early entropy source...[0m
[0;33mW (53) bootloader_random: RNG for ESP32-S3 not currently supported[0m
[0;32mI (59) boot: Partition Table:[0m
[0;32mI (63) boot: ## Label            Usage          Type ST Offset   Length[0m
[0;32mI (70) boot:  0 nvs              WiFi data        01 02 00009000 00006000[0m
[0;32mI (78) boot:  1 phy_init         RF data          01 01 0000f000 00001000[0m
[0;32mI (85) boot:  2 factory          factory app      00 00 00010000 00400000[0m
[0;32mI (93) boot:  3 flash_tone       Unknown data     01 ff 00410000 00004000[0m
[0;32mI (100) boot: End of partition table[0m
[0;32mI (104) esp_image: segment 0: paddr=00010020 vaddr=3c050020 size=2034ch (131916) map[0m
[0;32mI (137) esp_image: segment 1: paddr=00030374 vaddr=3fc93780 size=02854h ( 10324) load[0m
[0;32mI (139) esp_image: segment 2: paddr=00032bd0 vaddr=40374000 size=0d448h ( 54344) load[0m
[0;32mI (154) esp_image: segment 3: paddr=00040020 vaddr=42000020 size=4f5ach (325036) map[0m
[0;32mI (213) esp_image: segment 4: paddr=0008f5d4 vaddr=40381448 size=02334h (  9012) load[0m
[0;32mI (215) esp_image: segment 5: paddr=00091910 vaddr=50000000 size=00010h (    16) load[0m
[0;32mI (225) boot: Loaded app from partition at offset 0x10000[0m
[0;32mI (225) boot: Disabling RNG early entropy source...[0m
[0;33mW (230) bootloader_random: RNG for ESP32-S3 not currently supported[0m
[0;32mI (249) cpu_start: Pro cpu up.[0m
[0;32mI (249) cpu_start: Starting app cpu, entry point is 0x40375354[0m
[0;32mI (227) cpu_start: App cpu up.[0m
[0;32mI (263) cpu_start: Pro cpu start user code[0m
[0;32mI (263) cpu_start: cpu freq: 240000000[0m
[0;32mI (263) cpu_start: Application information:[0m
[0;32mI (266) cpu_start: Compile time:     Jan 14 2022 14:45:22[0m
[0;32mI (272) cpu_start: ELF file SHA256:  604a4ff4bcb3aede...[0m
[0;32mI (278) cpu_start: ESP-IDF:          v4.4-dev-3401-gb86fe0c66c-dirty[0m
[0;32mI (285) heap_init: Initializing. RAM available for dynamic allocation:[0m
[0;32mI (292) heap_init: At 3FC97088 len 00048F78 (291 KiB): D/IRAM[0m
[0;32mI (298) heap_init: At 3FCE0000 len 0000EE34 (59 KiB): STACK/DRAM[0m
[0;32mI (305) heap_init: At 3FCF0000 len 00008000 (32 KiB): DRAM[0m
[0;32mI (312) spi_flash: detected chip: winbond[0m
[0;32mI (316) spi_flash: flash io: dio[0m
[0;32mI (321) cpu_start: Starting scheduler on PRO CPU.[0m
[0;32mI (0) cpu_start: Starting scheduler on APP CPU.[0m
[0;32mI (351) example_asr_keywords: Initialize SR wn handle[0m
[0;31mE (351) RECORDER_ENG_HELPER: No valid neural network model selected[0m
[0;31mE (361) RECORDER_ENG_HELPER: No valid wake word selected[0m
Guru Meditation Error: Core  0 panic'ed (LoadProhibited). Exception was unhandled.

Core  0 register dump:
PC      : 0x42006774  PS      : 0x00060430  A0      : 0x8204f34a  A1      : 0x3fcf3f80  
A2      : 0x00000005  A3      : 0x3fcf3fa0  A4      : 0x3c052642  A5      : 0x10624dd3  
A6      : 0x3fc94d2c  A7      : 0x00000001  A8      : 0x8200676e  A9      : 0x3fcf3f60  
A10     : 0x3fcf41a0  A11     : 0x3c052642  A12     : 0x3c052657  A13     : 0x0000015f  
A14     : 0x3c052642  A15     : 0x00000000  SAR     : 0x00000004  EXCCAUSE: 0x0000001c  
EXCVADDR: 0x00000005  LBEG    : 0x400556d5  LEND    : 0x400556e5  LCOUNT  : 0xfffffffb  

Backtrace:0x42006771:0x3fcf3f800x4204f347:0x3fcf41c0 

ELF file SHA256: 604a4ff4bcb3aede

使用xtensa-esp32s3-elf-addr2line -pfiaC -e build/speech_recognition_example.elf 0x42006771:0x3fcf3f800x4204f347:0x3fcf41c0 显示报错在
0x42006771: app_main at /mnt/f/Workspace/espadf/esp-adf/examples/speech_recognition/asr/build/../main/example_asr_main.c:128
工程代码报错语句：    model_wn_data = wakenet->create(model_coeff_getter, DET_MODE_90);
请问如何解决，希望得到你们的帮助，谢谢

Zhengqinze05, 2022-01-14:
刚才发现adf更新了，只剩下wwe和vad工程，但是wwe工程编译不通过
FAILED: /mnt/f/Workspace/project/esp-adf/components/esp-sr/model/target/_MODEL_INFO_
cd /mnt/f/Workspace/project/esp-adf/examples/speech_recognition/wwe/build/esp-idf/esp-sr && python /mnt/f/Workspace/project/esp-adf/components/esp-sr/model/movemodel.py -d1 /mnt/f/Workspace/project/esp-adf/examples/speech_recognition/wwe -d2 /mnt/f/Workspace/project/esp-adf/components/esp-sr
/mnt/f/Workspace/project/esp-adf/examples/speech_recognition/wwe/sdkconfig
/mnt/f/Workspace/project/esp-adf/components/esp-sr/model
hilexin7q8
mn3cn
Traceback (most recent call last):
  File "/mnt/f/Workspace/project/esp-adf/components/esp-sr/model/movemodel.py", line 83, in <module>
    shutil.copytree(model_path + '/wakenet_model/' + wakenet_model, target_model+'/'+wakenet_model)
  File "/usr/lib/python3.8/shutil.py", line 557, in copytree
    return _copytree(entries=entries, src=src, dst=dst, symlinks=symlinks,
  File "/usr/lib/python3.8/shutil.py", line 513, in _copytree
    raise Error(errors)
shutil.Error: [('/mnt/f/Workspace/project/esp-adf/components/esp-sr/model/wakenet_model/hilexin7q8', '/mnt/f/Workspace/project/esp-adf/components/esp-sr/model/target/hilexin7q8', "[Errno 13] Permission denied: '/mnt/f/Workspace/project/esp-adf/components/esp-sr/model/target/hilexin7q8'")]
[5/1192] Generating ../../partition_table/partition-table.bin
Partition table binary generated. Contents:
请问这是什么问题呢

jason-mao, 2022-01-17:
@Zhengqinze05 ADF 更新完了，需要git submodule update --init 把对应的子仓库也拉到最新。

Zhengqinze05, 2022-01-17:
已经执行过了呢

jason-mao, 2022-01-17:
'/mnt/f/Workspace/project/esp-adf/components/esp-sr/model/target/hilexin7q8', "[Errno 13] Permission denied:,这个权限也需要改下

Zhengqinze05, 2022-01-17:
[image]
[image]
我这边看权限都是打开的呢

HengYongChao, 2022-01-28:
Hi @Zhengqinze05 

ASR 这个例程已经被 WWE 替代， 请更新ADF到最新版本，使用 WWE 吧， WWE 是使用了 recorder high level 的高级接口，开发起来更方便和更容易构建项目。

jason-mao, 2022-02-07:
@Zhengqinze05 

> [image] [image] 我这边看权限都是打开的呢

这个问题有解决吗？

Zhengqinze05, 2022-02-10:
> 

使用了esp-skainet之前版本的movemodel.py文件，就没有这个问题了

sapoiuyt, 2022-03-21:
您好，我想问一下 你是使用那个版本的  movemodel.py  我也遇到了一样的问题。

Zhengqinze05, 2022-03-24:
import io
import os
import argparse

def calculate_total_size(folder_path):
    total_size = 0
    for file_name in os.listdir(folder_path):
        path = os.path.join(folder_path, file_name)
        if os.path.isdir(path):
            total_size = total_size + calculate_total_size(path)
        if os.path.isfile(path):
            total_size = total_size + os.path.getsize(path)
    return total_size

if __name__ == '__main__':
    # input parameter
    parser = argparse.ArgumentParser(description='Model generator tool')
    parser.add_argument('-d1', '--project_path')
    parser.add_argument('-d2', '--model_path')
    args = parser.parse_args()

    sdkconfig_path = args.project_path + '/sdkconfig'
    model_path = args.model_path + '/model'

    print(sdkconfig_path)
    print(model_path)

with io.open(sdkconfig_path, "r") as f:
    WN_STRING = ''
    MN_STRING = ''
    for label in f:
        label = label.strip("\n")
        if 'CONFIG_SR_WN' in label and label[0] != '#':
            WN_STRING += label
        if 'CONFIG_SR_MN' in label and label[0] != '#':
            MN_STRING += label

wakenet_model = 'null'
if "CONFIG_SR_WN_WN7_XIAOAITONGXUE" in WN_STRING and "CONFIG_SR_WN_MODEL_WN7_QUANT8" in WN_STRING:
    wakenet_model = 'xiaoaitongxue7q8'
elif "CONFIG_SR_WN_WN7_XIAOAITONGXUE" in WN_STRING and "CONFIG_SR_WN_MODEL_WN7_QUANT" in WN_STRING:
    wakenet_model = 'xiaoaitongxue7'
elif "CONFIG_SR_WN_WN7_HILEXIN" in WN_STRING and "CONFIG_SR_WN_MODEL_WN7_QUANT8" in WN_STRING:
    wakenet_model = 'hilexin7q8'
elif "CONFIG_SR_WN_WN7_HILEXIN" in WN_STRING and "CONFIG_SR_WN_MODEL_WN7_QUANT" in WN_STRING:
    wakenet_model = 'hilexin7'
elif "CONFIG_SR_WN_WN7_ALEXA" in WN_STRING and "CONFIG_SR_WN_MODEL_WN7_QUANT8" in WN_STRING:
    wakenet_model = 'alexa7q8'
elif "CONFIG_SR_WN_WN7_ALEXA" in WN_STRING and "CONFIG_SR_WN_MODEL_WN7_QUANT" in WN_STRING:
    wakenet_model = 'alexa7'
elif "CONFIG_SR_WN_WN8_ALEXA" in WN_STRING and "CONFIG_SR_WN_MODEL_WN8_QUANT" in WN_STRING:
    wakenet_model = 'alexa8'
elif "CONFIG_SR_WN_WN8_HIESP" in WN_STRING and "CONFIG_SR_WN_MODEL_WN8_QUANT8" in WN_STRING:
    wakenet_model = 'hiesp8q8'
elif "CONFIG_SR_WN_WN8_HIESP" in WN_STRING and "CONFIG_SR_WN_MODEL_WN8_QUANT" in WN_STRING:
    wakenet_model = 'hiesp8'
else:
    print('choose no wakenet mode')

multinet_model = 'null'
if "CONFIG_SR_MN_CN_MULTINET3_SINGLE_RECOGNITION" in MN_STRING or "CONFIG_SR_MN_CN_MULTINET3_CONTINUOUS_RECOGNITION" in MN_STRING:
    multinet_model = 'mn3cn'
elif "CONFIG_SR_MN_CN_MULTINET4_SINGLE_RECOGNITION" in MN_STRING:
    multinet_model = 'mn4cn'
elif "CONFIG_SR_MN_EN_MULTINET5_SINGLE_RECOGNITION_QUANT8" in MN_STRING:
    multinet_model = 'mn5q8en'
elif "CONFIG_SR_MN_EN_MULTINET5_SINGLE_RECOGNITION" in MN_STRING:
    multinet_model = 'mn5en'
else:
    print('choose no multinet mode')

if wakenet_model != 'null':
    wakenet_model = model_path + '/wakenet_model/' + wakenet_model
if multinet_model != 'null':
    multinet_model = model_path + '/multinet_model/' + multinet_model

print(wakenet_model)
print(multinet_model)

target_model = model_path + '/target'
if not os.path.exists(target_model):
    os.makedirs(target_model)
os.system("rm -rf %s/*" % target_model)
if wakenet_model != 'null':
    os.system("cp -rf %s %s" % (wakenet_model, target_model))
if multinet_model != 'null':
    os.system("cp -rf %s %s" % (multinet_model, target_model))
# os.system("cp %s %s" % (wakenet_model+'/_MODEL_INFO_', target_model))

total_size = calculate_total_size(target_model)
print("Recommended model partition size: ", str(int((total_size / 1024 + 500) / 4 ) * 4) + 'KB')
你试一下

HengYongChao, 2022-04-01:
Hi @Zhengqinze05 

最新 ADF 版本的 asr 例程已经被去除了，因为 esp-sr 子仓库已经更新到最新的神经网络模型了，不在支持以前的调用。

jason-mao, 2022-08-09:
关闭该问题，如还有问题，请重新打开或者创建新 issue。

closed 2022-08-09 by jason-mao