espressif/esp-adf #750: ESP32-C3 ADF not supported

pffelix, 2022-01-16:
Hi 
I wan to run the example app play_mp3_control on my ESP32-C3, however the build does not compile:

In file included from C:/Users/Felix/esp/esp-adf/components/audio_board/lyrat_v4_3/board_def.h:28,
                 from C:/Users/Felix/esp/esp-adf/components/audio_board/lyrat_v4_3/board.h:29,
                 from C:/Users/Felix/esp/esp-adf/components/audio_hal/driver/zl38063/zl38063.c:32:
C:/Users/Felix/esp/esp-idf/components/driver/include/driver/touch_pad.h:17:10: fatal error: driver/touch_sensor.h: No such file or directory
 #include "driver/touch_sensor.h"
          ^~~~~~~~~~~~~~~~~~~~~~~
compilation terminated.
ninja: build stopped: subcommand failed.
ninja failed with exit code 1

Can you reproduce the error? In idf.py menuconfig there is currently no ESP32-C3 board selectable, however support for ESP32-C3 should be given since https://github.com/espressif/esp-adf/commit/7a729dd6acbbafdbab256799ecc95d4cf8baaf11 .

jason-mao, 2022-01-17:
@pffelix Thanks for your report, let me check.

nezen, 2022-01-18:
The default board for the example 'play_mp3_control' is lyrat4.3. But it also provides the option for a custom board. You can enable it via menuconfig. 
menuconfig > Audio HAL > My board v1.0 
Then you need to modify the files under play_mp3_control/components/my_board/my_board_v1_0/ according to your connections.

I hope this will help.

pffelix, 2022-01-20:
I got further on the issue. Before I can set the board connection configuration, I get following error: 

_5/77] Building C object esp-idf/audio_recorder/CMakeFiles/__idf_audio_recorder.dir/recorder_sr.c.obj
FAILED: esp-idf/audio_recorder/CMakeFiles/__idf_audio_recorder.dir/recorder_sr.c.obj
ccache C:\Users\Felix\.espressif\tools\riscv32-esp-elf\esp-2021r2-8.4.0\riscv32-esp-elf\bin\riscv32-esp-elf-gcc.exe  -Iconfig -IC:/Users/Felix/esp/esp-adf/components/audio_recorder/include -IC:/Users/Felix/esp/esp-idf/components/newlib/platform_include -IC:/Users/Felix/esp/esp-idf/components/freertos/include -IC:/Users/Felix/esp/esp-idf/components/freertos/port/riscv/include -IC:/Users/Felix/esp/esp-idf/components/esp_hw_support/include -IC:/Users/Felix/esp/esp-idf/components/esp_hw_support/port/esp32c3/. -IC:/Users/Felix/esp/esp-idf/components/esp_hw_support/port/esp32c3/private_include -IC:/Users/Felix/esp/esp-idf/components/heap/include -IC:/Users/Felix/esp/esp-idf/components/log/include -IC:/Users/Felix/esp/esp-idf/components/lwip/include/apps -IC:/Users/Felix/esp/esp-idf/components/lwip/include/apps/sntp -IC:/Users/Felix/esp/esp-idf/components/lwip/lwip/src/include -IC:/Users/Felix/esp/esp-idf/components/lwip/port/esp32/include -IC:/Users/Felix/esp/esp-idf/components/lwip/port/esp32/include/arch -IC:/Users/Felix/esp/esp-idf/components/soc/include -IC:/Users/Felix/esp/esp-idf/components/soc/esp32c3/. -IC:/Users/Felix/esp/esp-idf/components/soc/esp32c3/include -IC:/Users/Felix/esp/esp-idf/components/hal/esp32c3/include -IC:/Users/Felix/esp/esp-idf/components/hal/include -IC:/Users/Felix/esp/esp-idf/components/esp_rom/include -IC:/Users/Felix/esp/esp-idf/components/esp_rom/esp32c3 -IC:/Users/Felix/esp/esp-idf/components/esp_rom/include/esp32c3 -IC:/Users/Felix/esp/esp-idf/components/esp_common/include -IC:/Users/Felix/esp/esp-idf/components/esp_system/include -IC:/Users/Felix/esp/esp-idf/components/esp32c3/include -IC:/Users/Felix/esp/esp-idf/components/driver/include -IC:/Users/Felix/esp/esp-idf/components/driver/esp32c3/include -IC:/Users/Felix/esp/esp-idf/components/esp_ringbuf/include -IC:/Users/Felix/esp/esp-idf/components/efuse/include -IC:/Users/Felix/esp/esp-idf/components/efuse/esp32c3/include -IC:/Users/Felix/esp/esp-idf/components/riscv/include -IC:/Users/Felix/esp/esp-idf/components/espcoredump/include -IC:/Users/Felix/esp/esp-idf/components/esp_timer/include -IC:/Users/Felix/esp/esp-idf/components/esp_ipc/include -IC:/Users/Felix/esp/esp-idf/components/esp_pm/include -IC:/Users/Felix/esp/esp-idf/components/vfs/include -IC:/Users/Felix/esp/esp-idf/components/esp_wifi/include -IC:/Users/Felix/esp/esp-idf/components/esp_wifi/esp32c3/include -IC:/Users/Felix/esp/esp-idf/components/esp_event/include -IC:/Users/Felix/esp/esp-idf/components/esp_netif/include -IC:/Users/Felix/esp/esp-idf/components/esp_eth/include -IC:/Users/Felix/esp/esp-idf/components/tcpip_adapter/include -IC:/Users/Felix/esp/esp-adf/components/audio_sal/include -IC:/Users/Felix/esp/esp-adf/components/audio_pipeline/include -IC:/Users/Felix/esp/esp-adf/components/esp-adf-libs/esp_audio/include -IC:/Users/Felix/esp/esp-adf/components/esp-adf-libs/esp_codec/include/codec -IC:/Users/Felix/esp/esp-adf/components/esp-adf-libs/esp_codec/include/processing -IC:/Users/Felix/esp/esp-adf/components/esp-adf-libs/media_lib_sal/include -IC:/Users/Felix/esp/esp-adf/components/esp-adf-libs/media_lib_sal/include/port -IC:/Users/Felix/esp/esp-adf/components/esp-adf-libs/recorder_engine/include -IC:/Users/Felix/esp/esp-adf/components/esp-adf-libs/audio_misc/include -IC:/Users/Felix/esp/esp-adf/components/esp-adf-libs/esp_media_protocols/include -IC:/Users/Felix/esp/esp-idf/components/nghttp/port/include -IC:/Users/Felix/esp/esp-idf/components/nghttp/nghttp2/lib/includes -IC:/Users/Felix/esp/esp-idf/components/tcp_transport/include -IC:/Users/Felix/esp/esp-idf/components/esp-tls -IC:/Users/Felix/esp/esp-idf/components/esp-tls/esp-tls-crypto -IC:/Users/Felix/esp/esp-idf/components/mbedtls/port/include -IC:/Users/Felix/esp/esp-idf/components/mbedtls/mbedtls/include -IC:/Users/Felix/esp/esp-idf/components/mbedtls/esp_crt_bundle/include -march=rv32imc   -ffunction-sections -fdata-sections -Wall -Werror=all -Wno-error=unused-function -Wno-error=unused-variable -Wno-error=deprecated-declarations -Wextra -Wno-unused-parameter -Wno-sign-compare -ggdb -Wno-error=format= -nostartfiles -Wno-format -Og -fstrict-volatile-bitfields -Wno-error=unused-but-set-variable -std=gnu99 -Wno-old-style-declaration -D_GNU_SOURCE -DIDF_VER=\"v4.3.2\" -DESP_PLATFORM -MD -MT esp-idf/audio_recorder/CMakeFiles/__idf_audio_recorder.dir/recorder_sr.c.obj -MF esp-idf\audio_recorder\CMakeFiles\__idf_audio_recorder.dir\recorder_sr.c.obj.d -o esp-idf/audio_recorder/CMakeFiles/__idf_audio_recorder.dir/recorder_sr.c.obj   -c C:/Users/Felix/esp/esp-adf/components/audio_recorder/recorder_sr.c
C:/Users/Felix/esp/esp-adf/components/audio_recorder/recorder_sr.c:38:10: fatal error: esp_afe_sr_iface.h: No such file or directory
 #include "esp_afe_sr_iface.h"
          ^~~~~~~~~~~~~~~~~~~~
compilation terminated._

The error results as there is no include path set for `${IDF_TARGET} STREQUAL "esp32c3"` in C:\Users\Felix\esp\esp-adf\components\esp-sr\CMakeLists.txt. For ESP32, ESP32S2 and ESP32C3 include paths are given. Is this correct?

jason-mao, 2022-02-07:
> I got further on the issue. Before I can set the board connection configuration, I get following error:
> 
> _5/77] Building C object esp-idf/audio_recorder/CMakeFiles/__idf_audio_recorder.dir/recorder_sr.c.obj FAILED: esp-idf/audio_recorder/CMakeFiles/__idf_audio_recorder.dir/recorder_sr.c.obj ccache C:\Users\Felix.espressif\tools\riscv32-esp-elf\esp-2021r2-8.4.0\riscv32-esp-elf\bin\riscv32-esp-elf-gcc.exe -Iconfig -IC:/Users/Felix/esp/esp-adf/components/audio_recorder/include -IC:/Users/Felix/esp/esp-idf/components/newlib/platform_include -IC:/Users/Felix/esp/esp-idf/components/freertos/include -IC:/Users/Felix/esp/esp-idf/components/freertos/port/riscv/include -IC:/Users/Felix/esp/esp-idf/components/esp_hw_support/include -IC:/Users/Felix/esp/esp-idf/components/esp_hw_support/port/esp32c3/. -IC:/Users/Felix/esp/esp-idf/components/esp_hw_support/port/esp32c3/private_include -IC:/Users/Felix/esp/esp-idf/components/heap/include -IC:/Users/Felix/esp/esp-idf/components/log/include -IC:/Users/Felix/esp/esp-idf/components/lwip/include/apps -IC:/Users/Felix/esp/esp-idf/components/lwip/include/apps/sntp -IC:/Users/Felix/esp/esp-idf/components/lwip/lwip/src/include -IC:/Users/Felix/esp/esp-idf/components/lwip/port/esp32/include -IC:/Users/Felix/esp/esp-idf/components/lwip/port/esp32/include/arch -IC:/Users/Felix/esp/esp-idf/components/soc/include -IC:/Users/Felix/esp/esp-idf/components/soc/esp32c3/. -IC:/Users/Felix/esp/esp-idf/components/soc/esp32c3/include -IC:/Users/Felix/esp/esp-idf/components/hal/esp32c3/include -IC:/Users/Felix/esp/esp-idf/components/hal/include -IC:/Users/Felix/esp/esp-idf/components/esp_rom/include -IC:/Users/Felix/esp/esp-idf/components/esp_rom/esp32c3 -IC:/Users/Felix/esp/esp-idf/components/esp_rom/include/esp32c3 -IC:/Users/Felix/esp/esp-idf/components/esp_common/include -IC:/Users/Felix/esp/esp-idf/components/esp_system/include -IC:/Users/Felix/esp/esp-idf/components/esp32c3/include -IC:/Users/Felix/esp/esp-idf/components/driver/include -IC:/Users/Felix/esp/esp-idf/components/driver/esp32c3/include -IC:/Users/Felix/esp/esp-idf/components/esp_ringbuf/include -IC:/Users/Felix/esp/esp-idf/components/efuse/include -IC:/Users/Felix/esp/esp-idf/components/efuse/esp32c3/include -IC:/Users/Felix/esp/esp-idf/components/riscv/include -IC:/Users/Felix/esp/esp-idf/components/espcoredump/include -IC:/Users/Felix/esp/esp-idf/components/esp_timer/include -IC:/Users/Felix/esp/esp-idf/components/esp_ipc/include -IC:/Users/Felix/esp/esp-idf/components/esp_pm/include -IC:/Users/Felix/esp/esp-idf/components/vfs/include -IC:/Users/Felix/esp/esp-idf/components/esp_wifi/include -IC:/Users/Felix/esp/esp-idf/components/esp_wifi/esp32c3/include -IC:/Users/Felix/esp/esp-idf/components/esp_event/include -IC:/Users/Felix/esp/esp-idf/components/esp_netif/include -IC:/Users/Felix/esp/esp-idf/components/esp_eth/include -IC:/Users/Felix/esp/esp-idf/components/tcpip_adapter/include -IC:/Users/Felix/esp/esp-adf/components/audio_sal/include -IC:/Users/Felix/esp/esp-adf/components/audio_pipeline/include -IC:/Users/Felix/esp/esp-adf/components/esp-adf-libs/esp_audio/include -IC:/Users/Felix/esp/esp-adf/components/esp-adf-libs/esp_codec/include/codec -IC:/Users/Felix/esp/esp-adf/components/esp-adf-libs/esp_codec/include/processing -IC:/Users/Felix/esp/esp-adf/components/esp-adf-libs/media_lib_sal/include -IC:/Users/Felix/esp/esp-adf/components/esp-adf-libs/media_lib_sal/include/port -IC:/Users/Felix/esp/esp-adf/components/esp-adf-libs/recorder_engine/include -IC:/Users/Felix/esp/esp-adf/components/esp-adf-libs/audio_misc/include -IC:/Users/Felix/esp/esp-adf/components/esp-adf-libs/esp_media_protocols/include -IC:/Users/Felix/esp/esp-idf/components/nghttp/port/include -IC:/Users/Felix/esp/esp-idf/components/nghttp/nghttp2/lib/includes -IC:/Users/Felix/esp/esp-idf/components/tcp_transport/include -IC:/Users/Felix/esp/esp-idf/components/esp-tls -IC:/Users/Felix/esp/esp-idf/components/esp-tls/esp-tls-crypto -IC:/Users/Felix/esp/esp-idf/components/mbedtls/port/include -IC:/Users/Felix/esp/esp-idf/components/mbedtls/mbedtls/include -IC:/Users/Felix/esp/esp-idf/components/mbedtls/esp_crt_bundle/include -march=rv32imc -ffunction-sections -fdata-sections -Wall -Werror=all -Wno-error=unused-function -Wno-error=unused-variable -Wno-error=deprecated-declarations -Wextra -Wno-unused-parameter -Wno-sign-compare -ggdb -Wno-error=format= -nostartfiles -Wno-format -Og -fstrict-volatile-bitfields -Wno-error=unused-but-set-variable -std=gnu99 -Wno-old-style-declaration -D_GNU_SOURCE -DIDF_VER="v4.3.2" -DESP_PLATFORM -MD -MT esp-idf/audio_recorder/CMakeFiles/__idf_audio_recorder.dir/recorder_sr.c.obj -MF esp-idf\audio_recorder\CMakeFiles__idf_audio_recorder.dir\recorder_sr.c.obj.d -o esp-idf/audio_recorder/CMakeFiles/__idf_audio_recorder.dir/recorder_sr.c.obj -c C:/Users/Felix/esp/esp-adf/components/audio_recorder/recorder_sr.c C:/Users/Felix/esp/esp-adf/components/audio_recorder/recorder_sr.c:38:10: fatal error: esp_afe_sr_iface.h: No such file or directory #include "esp_afe_sr_iface.h" ^~~~~~~~~~~~~~~~~~~~ compilation terminated._
> 
> The error results as there is no cmake include path set for `${IDF_TARGET} STREQUAL "esp32c3"` in esp\esp-adf\components\esp-sr\CMakeLists.txt. For ESP32, ESP32S2 and ESP32S3 include paths are given. Is this correct?

C3 can't use the esp-sr APIs, should be disabled. ADF fix is coming, thanks.

closed 2022-02-09 by espressif-bot