espressif/esp-adf #754: The algorithm example not working on my lyrat4.3 board

nezen, 2022-01-26:
I was trying to experiment with the algorithm example but it was unsuccessful. Initially, there seemed to be an issue related to the default PSRAM config. After I enabled the PSRAM via menuconfig, I was able to compile the example and execute it on my lyrat4.3 board. But unfortunately, nothing happened.

I was using the pretty latest frameworks:
IDF4.4 f3e0c8bc41098f79baf2cf9f45dcbb6c1d41b3fe
latest ADF c5ef39d7c3ae3e3f4fb997b81dfc5d14b36ba299

here is the log:

I (1344) spiram: SPI SRAM memory test OK
I (1352) cpu_start: Pro cpu start user code
I (1352) cpu_start: cpu freq: 240000000
I (1352) cpu_start: Application information:
I (1355) cpu_start: Project name:     algorithm_examples
I (1361) cpu_start: App version:      1
I (1365) cpu_start: Compile time:     Jan 26 2022 14:05:41
I (1372) cpu_start: ELF file SHA256:  4c0bd48573dd025f...
I (1378) cpu_start: ESP-IDF:          v4.4-rc1
I (1383) heap_init: Initializing. RAM available for dynamic allocation:
I (1390) heap_init: At 3FFAE6E0 len 00001920 (6 KiB): DRAM
I (1396) heap_init: At 3FFB4C68 len 0002B398 (172 KiB): DRAM
I (1402) heap_init: At 3FFE0440 len 00003AE0 (14 KiB): D/IRAM
I (1409) heap_init: At 3FFE4350 len 0001BCB0 (111 KiB): D/IRAM
I (1415) heap_init: At 40097344 len 00008CBC (35 KiB): IRAM
I (1422) spiram: Adding pool of 4095K of external SPI memory to heap allocator
I (1430) spi_flash: detected chip: generic
I (1434) spi_flash: flash io: qio
W (1438) spi_flash: Detected size(8192k) larger than the size in the binary image header(4096k). Using the size in the binary image header.
I (1452) cpu_start: Starting scheduler on PRO CPU.
I (0) cpu_start: Starting scheduler on APP CPU.
I (1467) spiram: Reserving pool of 32K of internal memory for DMA/internal allocations
I (1475) ALGORITHM_EXAMPLES: [1.0] Mount sdcard 111
I (1481) SDCARD: Using 1-line SD mode, 4-line SD mode,  base path=/sdcard
I (1550) SDCARD: CID name SD16G!

I (1989) ALGORITHM_EXAMPLES: [2.0] Start codec chip
I (1989) gpio: GPIO[19]| InputEn: 1| OutputEn: 0| OpenDrain: 0| Pullup: 1| Pulldown: 0| Intr:3
E (1992) gpio: gpio_install_isr_service(449): GPIO isr service already installed
I (2012) gpio: GPIO[21]| InputEn: 0| OutputEn: 1| OpenDrain: 0| Pullup: 0| Pulldown: 0| Intr:0
I (2012) ES8388_DRIVER: init,out:02, in:00
I (2022) AUDIO_HAL: Codec mode is 3, Ctrl:1
I (2027) ALGORITHM_EXAMPLES: [3.0] Create audio pipeline_rec for recording
I (2029) ALGORITHM_EXAMPLES: [3.1] Create i2s stream to read audio data from codec chip
I (2045) I2S: APLL Enabled, coefficient: sdm0=28, sdm1=8, sdm2=5, odir=6
I (2046) I2S: DMA Malloc info, datalen=blocksize=1200, dma_buf_count=3
I (2053) I2S: DMA Malloc info, datalen=blocksize=1200, dma_buf_count=3
I (2060) I2S: I2S0, MCLK output by GPIO0
I (2064) LYRAT_V4_3: I2S0, MCLK output by GPIO0
I (2070) ALGORITHM_EXAMPLES: [3.2] Create mp3 encoder to encode mp3 format
I (2077) ALGORITHM_EXAMPLES: [3.3] Create fatfs stream to write data to sdcard
I (2085) ALGORITHM_EXAMPLES: [3.4] Register all elements to audio pipeline
I (2093) ALGORITHM_EXAMPLES: [3.5] Link it together [codec_chip]-->i2s_stream-->mp3_encoder-->fatfs_stream-->[sdcard]
I (2104) AUDIO_PIPELINE: link el->rb, el:0x3f80739c, tag:i2s_rd, rb:0x3f808cec
I (2112) AUDIO_PIPELINE: link el->rb, el:0x3f8074e0, tag:algo, rb:0x3f80ad2c
I (2120) AUDIO_PIPELINE: link el->rb, el:0x3f808a2c, tag:wav_encoder, rb:0x3f80cd6c
I (2128) ALGORITHM_EXAMPLES: [3.6] Set up  uri (file as fatfs_stream, mp3 as mp3 encoder)
I (2137) ALGORITHM_EXAMPLES: [4.0] Create audio pipeline_rec for recording
I (2144) ALGORITHM_EXAMPLES: [4.1] Create fatfs stream to read data from sdcard
I (2152) ALGORITHM_EXAMPLES: [4.2] Create mp3 decoder to decode mp3 file
I (2160) MP3_DECODER: MP3 init
I (2163) ALGORITHM_EXAMPLES: [4.3] Create i2s stream to write data to codec chip
**E (2171) I2S: register I2S object to platform failed**
I (2177) I2S: I2S0, MCLK output by GPIO0
I (2182) LYRAT_V4_3: I2S0, MCLK output by GPIO0
I (2187) ALGORITHM_EXAMPLES: [4.4] Register all elements to audio pipeline
I (2194) ALGORITHM_EXAMPLES: [4.5] Link it together [sdcard]-->fatfs_stream-->mp3_decoder-->i2s_stream-->[codec_chip]
I (2207) AUDIO_PIPELINE: link el->rb, el:0x3f80edf0, tag:file_rd, rb:0x3f80f26c
I (2214) AUDIO_PIPELINE: link el->rb, el:0x3f80ef68, tag:mp3_decoder, rb:0x3f8112ac
I (2222) ALGORITHM_EXAMPLES: [4.6] Set up  uri (file as fatfs_stream, mp3 as mp3 decoder, and default output is i2s)
I (2233) ALGORITHM_EXAMPLES: [5.0] Set up event listener
I (2239) ALGORITHM_EXAMPLES: [5.1] Listening event from all elements of pipeline
I (2247) ALGORITHM_EXAMPLES: [5.2] Listening event from peripherals
I (2254) ALGORITHM_EXAMPLES: [6.0] Start audio_pipeline
I (2261) AUDIO_ELEMENT: [file_rd-0x3f80edf0] Element task created
I (2267) AUDIO_ELEMENT: [mp3_decoder-0x3f80ef68] Element task created
I (2274) AUDIO_ELEMENT: [i2s_wd-0x3f80f0fc] Element task created
I (2281) AUDIO_PIPELINE: Func:audio_pipeline_run, Line:359, MEM Total:4341563 Bytes, Inter:280791 Bytes, Dram:246915 Bytes

I (2293) AUDIO_ELEMENT: [file_rd] AEL_MSG_CMD_RESUME,state:1
I (2300) AUDIO_ELEMENT: [mp3_decoder] AEL_MSG_CMD_RESUME,state:1
I (2306) MP3_DECODER: MP3 opened
I (2310) AUDIO_ELEMENT: [i2s_wd] AEL_MSG_CMD_RESUME,state:1
I (2316) I2S_STREAM: AUDIO_STREAM_WRITER
I (2321) AUDIO_PIPELINE: Pipeline started
I (2329) FATFS_STREAM: File size: 12996 byte, file position: 0
I (2357) AUDIO_ELEMENT: [i2s_rd-0x3f80739c] Element task created
I (2358) AUDIO_ELEMENT: [algo-0x3f8074e0] Element task created
I (2365) AUDIO_ELEMENT: [wav_encoder-0x3f808a2c] Element task created
I (2367) AUDIO_ELEMENT: [mp3_wd_file-0x3f808b68] Element task created
I (2375) AUDIO_PIPELINE: Func:audio_pipeline_run, Line:359, MEM Total:4284495 Bytes, Inter:263971 Bytes, Dram:230095 Bytes

I (2390) AUDIO_ELEMENT: [i2s_rd] AEL_MSG_CMD_RESUME,state:1
I (2392) I2S_STREAM: AUDIO_STREAM_READER,Rate:44100,ch:2
I (2402) AUDIO_ELEMENT: [algo] AEL_MSG_CMD_RESUME,state:1
I (2481) I2S: APLL Enabled, coefficient: sdm0=28, sdm1=8, sdm2=5, odir=6
I (2493) AUDIO_ELEMENT: [wav_encoder] AEL_MSG_CMD_RESUME,state:1
I (2494) AUDIO_ELEMENT: [mp3_wd_file] AEL_MSG_CMD_RESUME,state:1
I (2502) AUDIO_PIPELINE: Pipeline started
I (2502) ALGORITHM_EXAMPLES: [7.0] Listen for all pipeline events
Initial ONE-MIC auido front-end for speech recognition, mode:1, (Dec 10 2021 20:29:24)

As you can see, there is an I2S error message, but I'm not sure if that matters and what went wrong.
Please help me out. Thanks.

nezen, 2022-01-27:
I also tested this on my lyrat-mini board, still no luck.
There was a sound coming out from the speaker and rec_out.wav was generated, but no valid data was in the wav file and the length was only 5 sec while my test.mp3 was 10 sec.
To make sure my lyrat-mini board was good, I tested the example 'pipeline_recording_to_sdcard' on it and my voice can be recorded.

Here is the log:

entry 0x400806b4
I (27) boot: ESP-IDF v4.4-rc1 2nd stage bootloader
I (28) boot: compile time 09:40:53
I (28) boot: chip revision: 1
I (30) boot_comm: chip revision: 1, min. bootloader chip revision: 0
I (38) qio_mode: Enabling default flash chip QIO
I (43) boot.esp32: SPI Speed      : 80MHz
I (47) boot.esp32: SPI Mode       : QIO
I (52) boot.esp32: SPI Flash Size : 4MB
I (57) boot: Enabling RNG early entropy source...
I (62) boot: Partition Table:
I (66) boot: ## Label            Usage          Type ST Offset   Length
I (73) boot:  0 nvs              WiFi data        01 02 00009000 00006000
I (80) boot:  1 phy_init         RF data          01 01 0000f000 00001000
I (88) boot:  2 factory          factory app      00 00 00010000 00100000
I (95) boot: End of partition table
I (99) boot_comm: chip revision: 1, min. application chip revision: 0
I (107) esp_image: segment 0: paddr=00010020 vaddr=3f400020 size=6c2cch (443084) map
I (237) esp_image: segment 1: paddr=0007c2f4 vaddr=3ffb0000 size=02bc4h ( 11204) load
I (241) esp_image: segment 2: paddr=0007eec0 vaddr=40080000 size=01158h (  4440) load
I (244) esp_image: segment 3: paddr=00080020 vaddr=400d0020 size=76b28h (486184) map
I (385) esp_image: segment 4: paddr=000f6b50 vaddr=40081158 size=16248h ( 90696) load
I (415) esp_image: segment 5: paddr=0010cda0 vaddr=50000000 size=00010h (    16) load
I (426) boot: Loaded app from partition at offset 0x10000
I (426) boot: Disabling RNG early entropy source...
I (438) psram: This chip is ESP32-D0WD
I (439) spiram: Found 64MBit SPI RAM device
I (439) spiram: SPI RAM mode: flash 80m sram 40m
I (444) spiram: PSRAM initialized, cache is in low/high (2-core) mode.
I (451) cpu_start: Pro cpu up.
I (455) cpu_start: Starting app cpu, entry point is 0x40081894
0x40081894: call_start_cpu1 at C:/Users/yujonath/esp/latest_esp_idf_adf/esp-idf/components/esp_system/port/cpu_start.c:156

I (0) cpu_start: App cpu up.
I (1344) spiram: SPI SRAM memory test OK
I (1352) cpu_start: Pro cpu start user code
I (1352) cpu_start: cpu freq: 240000000
I (1352) cpu_start: Application information:
I (1355) cpu_start: Project name:     algorithm_examples
I (1361) cpu_start: App version:      1
I (1365) cpu_start: Compile time:     Jan 27 2022 09:39:49
I (1372) cpu_start: ELF file SHA256:  14b3b12beb98a703...
I (1378) cpu_start: ESP-IDF:          v4.4-rc1
I (1383) heap_init: Initializing. RAM available for dynamic allocation:
I (1390) heap_init: At 3FFAE6E0 len 00001920 (6 KiB): DRAM
I (1396) heap_init: At 3FFB4CE0 len 0002B320 (172 KiB): DRAM
I (1403) heap_init: At 3FFE0440 len 00003AE0 (14 KiB): D/IRAM
I (1409) heap_init: At 3FFE4350 len 0001BCB0 (111 KiB): D/IRAM
I (1416) heap_init: At 400973A0 len 00008C60 (35 KiB): IRAM
I (1422) spiram: Adding pool of 4095K of external SPI memory to heap allocator
I (1430) spi_flash: detected chip: gd
I (1434) spi_flash: flash io: qio
W (1438) spi_flash: Detected size(8192k) larger than the size in the binary image header(4096k). Using the size in the binary image header.
I (1452) cpu_start: Starting scheduler on PRO CPU.
I (0) cpu_start: Starting scheduler on APP CPU.
I (1467) spiram: Reserving pool of 32K of internal memory for DMA/internal allocations
I (1475) ALGORITHM_EXAMPLES: [1.0] Mount sdcard
I (1982) ALGORITHM_EXAMPLES: [2.0] Start codec chip
E (2003) DRV8311: SET: volume:46 (0x00 = -95.5dB, 0xBF = 0dB, 0xFF = +32dB)
W (2044) I2C_BUS: i2c_bus_create:57: I2C bus has been already created, [port:0]
E (2052) DRV8311: SET: volume:5a (0x00 = -95.5dB, 0xBF = 0dB, 0xFF = +32dB)
I (2053) ALGORITHM_EXAMPLES: [3.0] Create audio pipeline_rec for recording
I (2058) ALGORITHM_EXAMPLES: [3.1] Create i2s stream to read audio data from codec chip
I (2069) ALGORITHM_EXAMPLES: [3.2] Create mp3 encoder to encode mp3 format
[... 17 lines cut ...]
Initial ONE-MIC auido front-end for speech recognition, mode:1, (Dec 10 2021 20:29:24)
W (3019) FATFS_STREAM: No more data, ret:0
W (11870) ALGORITHM_EXAMPLES: [ * ] Stop event received
W (11871) AUDIO_ELEMENT: IN-[algo] AEL_IO_ABORT
W (11871) AUDIO_ELEMENT: IN-[wav_encoder] AEL_IO_ABORT
afe_se_task quit
afe task destroy finished
E (11877) AUDIO_ELEMENT: [wav_encoder] Element already stopped
W (11887) AUDIO_ELEMENT: IN-[mp3_wd_file] AEL_IO_ABORT
E (11911) AUDIO_ELEMENT: [file_rd] Element already stopped
E (11911) AUDIO_ELEMENT: [mp3_decoder] Element already stopped
E (11913) AUDIO_ELEMENT: [i2s_wd] Element already stopped
I (11919) ALGORITHM_EXAMPLES: [8.0] Stop audio_pipeline
W (11925) AUDIO_ELEMENT: [algo] Element has not create when AUDIO_ELEMENT_TERMINATE
W (11933) AUDIO_PIPELINE: There are no listener registered
W (11939) AUDIO_ELEMENT: [algo] Element has not create when AUDIO_ELEMENT_TERMINATE
W (11948) AUDIO_ELEMENT: [i2s_rd] Element has not create when AUDIO_ELEMENT_TERMINATE
W (11957) AUDIO_ELEMENT: [wav_encoder] Element has not create when AUDIO_ELEMENT_TERMINATE
W (11965) AUDIO_ELEMENT: [mp3_wd_file] Element has not create when AUDIO_ELEMENT_TERMINATE
W (11974) AUDIO_PIPELINE: There are no listener registered
W (11980) AUDIO_ELEMENT: [file_rd] Element has not create when AUDIO_ELEMENT_TERMINATE
W (11989) AUDIO_ELEMENT: [mp3_decoder] Element has not create when AUDIO_ELEMENT_TERMINATE
W (11998) AUDIO_ELEMENT: [i2s_wd] Element has not create when AUDIO_ELEMENT_TERMINATE

Again, please fix these issues, thanks.

HengYongChao, 2022-01-27:
Hi @nezen 

Sorry, I have reproduced the problem, it does exist, and I will fix it ASAP.

tozon, 2022-02-09:
I'm debugging now pipeline_passthrough example, that gives the same error message as algorithm example.
The reason is i2s_priv_register_object() call. First call during creation I2S write stream works fine, then it fails during creation of read stream (the same p_i2s[port_id] object is used).
Dear ESP32 support, please treat your customers seriously. How long it can take to debug this simple issue? @HengYongChao

HengYongChao, 2022-02-10:
> I'm debugging now pipeline_passthrough example, that gives the same error message as algorithm example. The reason is i2s_priv_register_object() call. First call during creation I2S write stream [...]

Hi @tozon, 

Can you tell me more details (running, building, logs)?  
I use the IDF and ADF versions you mentioned, use the Lyrat v4.3 development board, compile and run the pipeline_passthru routine, and there is no error you mentioned. Here is the log.[pipeline_passthru_run.log](https://github.com/espressif/esp-adf/files/8039499/pipeline_passthru_run.log)

And about @nezen, He said that the problem mentioned is caused by the upgrade of the [esp-sr](https://github.com/espressif/esp-sr/tree/release/v1.0) submodule, which does not seem to be the same problem as the i2s problem you mentioned.

tozon, 2022-02-10:
> I use the IDF and ADF versions you mentioned, use the Lyrat v4.3 development board, compile and run the pipeline_passthru routine, and there is no error you mentioned. Here is the log.[pipeline_passthru_run.log](https://github.com/espressif/esp-adf/files/8039499/pipeline_passthru_run.log)
I can see it in your log. Line 75:
"E (444) I2S: register I2S object to platform failed"
that's the same result I'm getting. It's an error, not warning, something went wrong.

I have reverted to an older IDF version (that, what is present in ADF repository) and now this error is replaced by warning with "W (405) I2S: I2S driver already installed".
The example seems to work: if I connect music source to AUX_IN and headphones to PHONEJACK I can hear it. But if I press reset button and hold it, the music is still there. If I do a short circuit on I2S line the music also is playing. So I think there is something wrong with this example. It should sample input using codec chip, send it via I2S to ESP32 and send it back via I2S to codec chip.
How can you explain that?

closed 2022-02-17 by espressif-bot