espressif/esp-adf #755: CONSOLE_EXAMPLE: Fail to save sdcard url to sdcard playlist

SinglWolf, 2022-01-28:
I am using example https://github.com/espressif/esp-adf/tree/master/examples/cli
If remove and reinsert the SD card, cannot get the playlist.
```
I (29147) SDCARD: Using 1-line SD mode, 4-line SD mode,  base path=/sdcard
I (29147) gpio: GPIO[13]| InputEn: 0| OutputEn: 1| OpenDrain: 0| Pullup: 0| Pulldown: 0| Intr:0 
I (29337) SDCARD: Name: SU08G
I (29337) SDCARD: Type: SDHC/SDXC
I (29337) SDCARD: Speed: high speed
I (29337) SDCARD: Size: 7580MB
I (29347) SDCARD: CSD: ver=1, sector_size=512, capacity=15523840 read_bl_len=9
I (29347) SDCARD: SCR: sd_spec=2, bus_width=5
I (29357) SDCARD: CID name SU08G!

W (29357) AUDIO_EVT: There is no space in external queue
W (34087) AUDIO_EVT: There is no space in external queue
I (37857) SDCARD: Using 1-line SD mode, 4-line SD mode,  base path=/sdcard
I (37857) gpio: GPIO[13]| InputEn: 0| OutputEn: 1| OpenDrain: 0| Pullup: 0| Pulldown: 0| Intr:0 
I (38047) SDCARD: Name: SU08G
I (38047) SDCARD: Type: SDHC/SDXC
I (38047) SDCARD: Speed: high speed
I (38047) SDCARD: Size: 7580MB
I (38057) SDCARD: CSD: ver=1, sector_size=512, capacity=15523840 read_bl_len=9
I (38057) SDCARD: SCR: sd_spec=2, bus_width=5
I (38067) SDCARD: CID name SU08G!

W (38067) AUDIO_EVT: There is no space in external queue
W (44037) AUDIO_EVT: There is no space in external queue
I (45867) SDCARD: Using 1-line SD mode, 4-line SD mode,  base path=/sdcard
I (45867) gpio: GPIO[13]| InputEn: 0| OutputEn: 1| OpenDrain: 0| Pullup: 0| Pulldown: 0| Intr:0 
I (46057) SDCARD: Name: SU08G
I (46057) SDCARD: Type: SDHC/SDXC
I (46057) SDCARD: Speed: high speed
I (46057) SDCARD: Size: 7580MB
I (46067) SDCARD: CSD: ver=1, sector_size=512, capacity=15523840 read_bl_len=9
I (46067) SDCARD: SCR: sd_spec=2, bus_width=5
I (46077) SDCARD: CID name SU08G!

W (46077) AUDIO_EVT: There is no space in external queue
esp32> scan /sdcard/m
E (67317) PLAYLIST_SDCARD: unexpect action in sdcard list, line: 77
E (67317) CONSOLE_EXAMPLE: Fail to save sdcard url to sdcard playlist
E (67337) PLAYLIST_SDCARD: unexpect action in sdcard list, line: 77
E (67337) CONSOLE_EXAMPLE: Fail to save sdcard url to sdcard playlist
E (67347) PLAYLIST_SDCARD: unexpect action in sdcard list, line: 77
E (67347) CONSOLE_EXAMPLE: Fail to save sdcard url to sdcard playlist
E (67357) PLAYLIST_SDCARD: unexpect action in sdcard list, line: 77
E (67357) CONSOLE_EXAMPLE: Fail to save sdcard url to sdcard playlist
E (67367) PLAYLIST_SDCARD: unexpect action in sdcard list, line: 77
E (67377) CONSOLE_EXAMPLE: Fail to save sdcard url to sdcard playlist
E (67387) PLAYLIST_SDCARD: unexpect action in sdcard list, line: 77
E (67397) CONSOLE_EXAMPLE: Fail to save sdcard url to sdcard playlist
E (67397) PLAYLIST_SDCARD: unexpect action in sdcard list, line: 77
E (67407) CONSOLE_EXAMPLE: Fail to save sdcard url to sdcard playlist
E (67407) PLAYLIST_SDCARD: unexpect action in sdcard list, line: 77
E (67417) CONSOLE_EXAMPLE: Fail to save sdcard url to sdcard playlist
E (67417) PLAYLIST_SDCARD: unexpect action in sdcard list, line: 77
E (67437) CONSOLE_EXAMPLE: Fail to save sdcard url to sdcard playlist
E (67447) PLAYLIST_SDCARD: unexpect action in sdcard list, line: 77
E (67447) CONSOLE_EXAMPLE: Fail to save sdcard url to sdcard playlist
E (67457) PLAYLIST_SDCARD: unexpect action in sdcard list, line: 77
E (67457) CONSOLE_EXAMPLE: Fail to save sdcard url to sdcard playlist
E (67467) PLAYLIST_SDCARD: unexpect action in sdcard list, line: 77
E (67467) CONSOLE_EXAMPLE: Fail to save sdcard url to sdcard playlist
E (67487) PLAYLIST_SDCARD: unexpect action in sdcard list, line: 77
E (67497) CONSOLE_EXAMPLE: Fail to save sdcard url to sdcard playlist
E (67497) PLAYLIST_SDCARD: unexpect action in sdcard list, line: 77
E (67507) CONSOLE_EXAMPLE: Fail to save sdcard url to sdcard playlist
E (67507) PLAYLIST_SDCARD: unexpect action in sdcard list, line: 77
E (67517) CONSOLE_EXAMPLE: Fail to save sdcard url to sdcard playlist
E (67517) PLAYLIST_SDCARD: unexpect action in sdcard list, line: 77
[... 149 lines cut ...]
E (68577) PLAYLIST_SDCARD: unexpect action in sdcard list, line: 77
E (68587) CONSOLE_EXAMPLE: Fail to save sdcard url to sdcard playlist
E (68597) PLAYLIST_SDCARD: unexpect action in sdcard list, line: 77
E (68607) CONSOLE_EXAMPLE: Fail to save sdcard url to sdcard playlist
E (68617) PLAYLIST_SDCARD: unexpect action in sdcard list, line: 77
E (68617) CONSOLE_EXAMPLE: Fail to save sdcard url to sdcard playlist
E (68627) PLAYLIST_SDCARD: unexpect action in sdcard list, line: 77
E (68627) CONSOLE_EXAMPLE: Fail to save sdcard url to sdcard playlist
E (68637) PLAYLIST_SDCARD: unexpect action in sdcard list, line: 77
E (68637) CONSOLE_EXAMPLE: Fail to save sdcard url to sdcard playlist
E (68657) PLAYLIST_SDCARD: unexpect action in sdcard list, line: 77
E (68667) CONSOLE_EXAMPLE: Fail to save sdcard url to sdcard playlist
E (68667) PLAYLIST_SDCARD: unexpect action in sdcard list, line: 77
E (68677) CONSOLE_EXAMPLE: Fail to save sdcard url to sdcard playlist
E (68677) PLAYLIST_SDCARD: unexpect action in sdcard list, line: 77
E (68687) CONSOLE_EXAMPLE: Fail to save sdcard url to sdcard playlist
E (68687) PLAYLIST_SDCARD: unexpect action in sdcard list, line: 77
E (68697) CONSOLE_EXAMPLE: Fail to save sdcard url to sdcard playlist
E (68717) PLAYLIST_SDCARD: unexpect action in sdcard list, line: 77
E (68717) CONSOLE_EXAMPLE: Fail to save sdcard url to sdcard playlist
E (68727) PLAYLIST_SDCARD: unexpect action in sdcard list, line: 203
esp32>
```
Judging by the number of lines of output after command  `scan /sdcard/m`, the SD card is scanned the directory with music files.
Help solve the problem.

jason-mao, 2022-02-07:
@SinglWolf Did you see  /sdcard/__playlist/_playlist_url and /sdcard/__playlist/_offset  files in your SDCARD? Please ensure 
 have the authorization to write for sdcard.

SinglWolf, 2022-02-07:
> Did you see /sdcard/__playlist/_playlist_url and /sdcard/__playlist/_offset files in your SDCARD? Please ensure
> have the authorization to write for sdcard.

Thanks for the answer. I no longer hoped.
The files `/sdcard/__playlist/_playlist_url` and `/sdcard/__playlist/_offset` are present on the `SD-card`. It follows from this that writing to the `SD-card` is taking place. And what is the point for me to prohibit writing to the `SD-card` and write to you about the problem that has arisen?

HengYongChao, 2022-02-18:
Hi @SinglWolf 

I think the problem is that the sdcard is deliberately ejected during use. Then insert it again (sdcard detection interruption will automatically reinitialize the sdcard), it may be necessary to re-execute the playlist scan operation.

closed 2023-09-05 by jason-mao