espressif/esp-adf #757: esp32s2 not compiling, due to missing declaration of AFE_CONFIG_DEFAULT

moefear85, 2022-01-31:
Hi,

I get the following error midway through compilation:

> esp-adf/components/audio_stream/algorithm_stream.c:120:31: error: implicit declaration of function 'AFE_CONFIG_DEFAULT'

There is no way to turn off AFE in menuconfig. There is only the option of choosing 1 or 2 microphones. Does this mean esp-adf doesn't support the esp32s2? I'm using the latest master branch (as of 01.31.2022). It works fine with the esp32.

ISR-AlanYang, 2022-02-04:
@moefear85 I got the same issue with esp32s2. After diving into the source code, I am pretty sure this is a bug in adf (latest master branch). Simply change line 51 of algorithm_stream.h  to: 
#if CONFIG_IDF_TARGET_ESP32 || CONFIG_IDF_TARGET_ESP32S2
will fix the bug. 
Apparently the type of esp32s2 is missed in definition of AFE_CONFIG_DEFAULT. I don't know which definition is suitable for esp32s2, so I simply make it the same as esp32.  
It works for me. Hope it helps.

HengYongChao, 2022-02-11:
HI @ISR-AlanYang @moefear85 

ADF: 407499f88, release/v4.4
IDF: ddc44956b, master

1. `git submodule update -f` update the submodules.
2. after `idf.py set-target esp32-s2`
3. then `idf.py menuconfig`, select  `ESP32-S2-Kaluga-1 v1.2`
4. at laest cd `algorithm` example, `idf.py build` , 

it build sesseccful. 
If you are still having compilation issues, please give me some more details, thanks!

closed 2022-08-09 by jason-mao