espressif/esp-adf #758: The esp-adf/examples/get-started/play_mp3_control project does not work.

zip1982b, 2022-02-08:
idf.py menuconfig -> Failed to resolve component 'jsmn'.
My os Ubuntu 20.04. esp idf worked correct.
what could be the problem?

HengYongChao, 2022-02-09:
Hi @zip1982b 

I don't see your complete compilation log,  I guess you may have forgotten to pull the full IDF submodules.

you can try it by following the steps below:
1.  cd $IDF_PATH 
2.  git submodule update -f 

Then try to `idf.py menuconfig` again.

zip1982b, 2022-02-16:
~/esp/play_mp3_control$ idf.py menuconfig
Executing action: menuconfig
Running cmake in directory /home/zip1982b/esp/play_mp3_control/build
Executing "cmake -G Ninja -DPYTHON_DEPS_CHECKED=1 -DESP_PLATFORM=1 -DCCACHE_ENABLE=0 /home/zip1982b/esp/play_mp3_control"...
fatal: не найден git репозиторий (или один из его каталогов вплоть до точки монтирования /)
Останавливаю поиск на границе файловой системы (так как GIT_DISCOVERY_ACROSS_FILESYSTEM не установлен).
-- IDF_TARGET not set, using default target: esp32
-- Found Git: /usr/bin/git (found version "2.25.1") 
-- The C compiler identification is GNU 8.4.0
-- The CXX compiler identification is GNU 8.4.0
-- The ASM compiler identification is GNU
-- Found assembler: /home/zip1982b/.espressif/tools/xtensa-esp32-elf/esp-2021r2-8.4.0/xtensa-esp32-elf/bin/xtensa-esp32-elf-gcc
-- Check for working C compiler: /home/zip1982b/.espressif/tools/xtensa-esp32-elf/esp-2021r2-8.4.0/xtensa-esp32-elf/bin/xtensa-esp32-elf-gcc
-- Check for working C compiler: /home/zip1982b/.espressif/tools/xtensa-esp32-elf/esp-2021r2-8.4.0/xtensa-esp32-elf/bin/xtensa-esp32-elf-gcc -- works
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Detecting C compile features
-- Detecting C compile features - done
-- Check for working CXX compiler: /home/zip1982b/.espressif/tools/xtensa-esp32-elf/esp-2021r2-8.4.0/xtensa-esp32-elf/bin/xtensa-esp32-elf-g++
-- Check for working CXX compiler: /home/zip1982b/.espressif/tools/xtensa-esp32-elf/esp-2021r2-8.4.0/xtensa-esp32-elf/bin/xtensa-esp32-elf-g++ -- works
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Project is not inside a git repository, or git repository has no commits; will not use 'git describe' to determine PROJECT_VER.
-- Building ESP-IDF components for target esp32
CMake Error at /home/zip1982b/esp/esp-idf/tools/cmake/build.cmake:197 (message):
  Failed to resolve component 'jsmn'.
Call Stack (most recent call first):
  /home/zip1982b/esp/esp-idf/tools/cmake/build.cmake:231 (__build_resolve_and_add_req)
  /home/zip1982b/esp/esp-idf/tools/cmake/build.cmake:512 (__build_expand_requirements)
  /home/zip1982b/esp/esp-idf/tools/cmake/project.cmake:422 (idf_build_process)
  CMakeLists.txt:10 (project)

-- Configuring incomplete, errors occurred!
See also "/home/zip1982b/esp/play_mp3_control/build/CMakeFiles/CMakeOutput.log".
cmake failed with exit code 1

zip1982b, 2022-02-16:
I used the installation instructions.
1. Installed esp-idf. Checked hello-world. The project compiles successfully.
2. installed esp-adf. Project fails at idf.py menuconfig stage.

HengYongChao, 2022-02-17:
> I used the installation instructions.
> 
> 1. Installed esp-idf. Checked hello-world. The project compiles successfully.
> 2. installed esp-adf. Project fails at idf.py menuconfig stage.

Hi @zip1982b 

When you use ADF, please set $ADF_PATH to your ADF folder first, then try it again.

zip1982b, 2022-02-20:
> > I used the installation instructions.
> > 
> > 1. Installed esp-idf. Checked hello-world. The project compiles successfully.
> > 2. installed esp-adf. Project fails at idf.py menuconfig stage.
> 
> Hi @zip1982b
> 
> When you use ADF, please set $ADF_PATH to your ADF folder first, then try to compile the example again.

i did that.
get_idf 
get_adf > /home/zip1982b/esp/esp-adf
idf.py menuconfig > Failed to resolve component 'jsmn'.

zip1982b, 2022-02-20:
I completely removed esp-idf, esp-adf and installed esp-idf, esp-adf.
When installing esp-idf, esp-adf I completely followed the instructions:
1.https://docs.espressif.com/projects/esp-idf/en/latest/esp32/get-started/index.html. 
2.https://docs.espressif.com/projects/esp-adf/en/latest/get-started/index.html#get-started-get-esp-adf
All environment variables have been set and configured ($IDF_PATH, $ADF_PATH). 
Projects on esp-idf compile successfully. But there is no esp-adf. Please try to install esp-adf on ubuntu 20.04 anyone please.

zip1982b, 2022-02-20:
CMakeOutput.log
The target system is: Generic -  - 
The host system is: Linux - 5.13.0-28-generic - x86_64
Compiling the C compiler identification source file "CMakeCCompilerId.c" succeeded.
Compiler: /home/zip1982b/.espressif/tools/xtensa-esp32-elf/esp-2021r2-8.4.0/xtensa-esp32-elf/bin/xtensa-esp32-elf-gcc 
Build flags: -mlongcalls;-Wno-frame-address;
Id flags:  

The output was:
0

Compilation of the C compiler identification source "CMakeCCompilerId.c" produced "a.out"

The C compiler identification is GNU, found in "/home/zip1982b/esp/play_mp3_control/build/CMakeFiles/3.16.3/CompilerIdC/a.out"

Compiling the CXX compiler identification source file "CMakeCXXCompilerId.cpp" succeeded.
Compiler: /home/zip1982b/.espressif/tools/xtensa-esp32-elf/esp-2021r2-8.4.0/xtensa-esp32-elf/bin/xtensa-esp32-elf-g++ 
Build flags: -mlongcalls;-Wno-frame-address;
Id flags:  

The output was:
0

Compilation of the CXX compiler identification source "CMakeCXXCompilerId.cpp" produced "a.out"

The CXX compiler identification is GNU, found in "/home/zip1982b/esp/play_mp3_control/build/CMakeFiles/3.16.3/CompilerIdCXX/a.out"

Checking whether the ASM compiler is GNU using "--version" matched "(GNU assembler)|(GCC)|(Free Software Foundation)":
xtensa-esp32-elf-gcc (crosstool-NG esp-2021r2) 8.4.0
Copyright (C) 2018 Free Software Foundation, Inc.
This is free software; see the source for copying conditions.  There is NO
warranty; not even for MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.

Determining if the C compiler works passed with the following output:
Change Dir: /home/zip1982b/esp/play_mp3_control/build/CMakeFiles/CMakeTmp

Run Build Command(s):/usr/bin/ninja cmTC_f8ed1 && [1/2] Building C object CMakeFiles/cmTC_f8ed1.dir/testCCompiler.c.obj
[2/2] Linking C executable cmTC_f8ed1

Detecting C compiler ABI info compiled with the following output:
Change Dir: /home/zip1982b/esp/play_mp3_control/build/CMakeFiles/CMakeTmp

Run Build Command(s):/usr/bin/ninja cmTC_1c26b && [1/2] Building C object CMakeFiles/cmTC_1c26b.dir/CMakeCCompilerABI.c.obj
Using built-in specs.
COLLECT_GCC=/home/zip1982b/.espressif/tools/xtensa-esp32-elf/esp-2021r2-8.4.0/xtensa-esp32-elf/bin/xtensa-esp32-elf-gcc
Target: xtensa-esp32-elf
Configured with: /builds/idf/crosstool-NG/.build/xtensa-esp32-elf/src/gcc/configure --build=x86_64-build_pc-linux-gnu --host=x86_64-build_pc-linux-gnu --target=xtensa-esp32-elf --prefix=/builds/idf/crosstool-NG/builds/xtensa-esp32-elf --with-local-prefix=/builds/idf/crosstool-NG/builds/xtensa-esp32-elf/xtensa-esp32-elf --with-headers=/builds/idf/crosstool-NG/builds/xtensa-esp32-elf/xtensa-esp32-elf/include --with-newlib --enable-threads=no --disable-shared --with-pkgversion='crosstool-NG esp-2021r2' --disable-__cxa_atexit --enable-cxx-flags=-ffunction-sections --disable-libgomp --disable-libmudflap --disable-libmpx --disable-libssp --disable-libquadmath --disable-libquadmath-support --with-gmp=/builds/idf/crosstool-NG/.build/xtensa-esp32-elf/buildtools --with-mpfr=/builds/idf/crosstool-NG/.build/xtensa-esp32-elf/buildtools --with-mpc=/builds/idf/crosstool-NG/.build/xtensa-esp32-elf/buildtools --with-isl=/builds/idf/crosstool-NG/.build/xtensa-esp32-elf/buildtools --enable-lto --enable-target-optspace --without-long-double-128 --disable-nls --enable-multiarch --enable-languages=c,c++ --disable-libstdcxx-verbose --enable-threads=posix --enable-gcov-custom-rtio --enable-libstdcxx-time=yes
Thread model: posix
gcc version 8.4.0 (crosstool-NG esp-2021r2) 
COLLECT_GCC_OPTIONS='-mlongcalls' '-Wno-frame-address' '-v' '-o' 'CMakeFiles/cmTC_1c26b.dir/CMakeCCompilerABI.c.obj' '-c'
 /home/zip1982b/.espressif/tools/xtensa-esp32-elf/esp-2021r2-8.4.0/xtensa-esp32-elf/bin/../libexec/gcc/xtensa-esp32-elf/8.4.0/cc1 -quiet -v -iprefix /home/zip1982b/.espressif/tools/xtensa-esp32-elf/esp-2021r2-8.4.0/xtensa-esp32-elf/bin/../lib/gcc/xtensa-esp32-elf/8.4.0/ /usr/share/cmake-3.16/Modules/CMakeCCompilerABI.c -quiet -dumpbase CMakeCCompilerABI.c -mlongcalls -auxbase-strip CMakeFiles/cmTC_1c26b.dir/CMakeCCompilerABI.c.obj -Wno-frame-address -version -o /tmp/ccI1KD93.s
GNU C17 (crosstool-NG esp-2021r2) version 8.4.0 (xtensa-esp32-elf)
	compiled by GNU C version 4.9.2, GMP version 6.2.0, MPFR version 4.0.2, MPC version 1.1.0, isl version isl-0.19-GMP

GGC heuristics: --param ggc-min-expand=100 --param ggc-min-heapsize=131072
ignoring duplicate directory "/home/zip1982b/.espressif/tools/xtensa-esp32-elf/esp-2021r2-8.4.0/xtensa-esp32-elf/bin/../lib/gcc/../../lib/gcc/xtensa-esp32-elf/8.4.0/include"
ignoring duplicate directory "/home/zip1982b/.espressif/tools/xtensa-esp32-elf/esp-2021r2-8.4.0/xtensa-esp32-elf/bin/../lib/gcc/../../lib/gcc/xtensa-esp32-elf/8.4.0/include-fixed"
ignoring duplicate directory "/home/zip1982b/.espressif/tools/xtensa-esp32-elf/esp-2021r2-8.4.0/xtensa-esp32-elf/bin/../lib/gcc/../../lib/gcc/xtensa-esp32-elf/8.4.0/../../../../xtensa-esp32-elf/sys-include"
ignoring duplicate directory "/home/zip1982b/.espressif/tools/xtensa-esp32-elf/esp-2021r2-8.4.0/xtensa-esp32-elf/bin/../lib/gcc/../../lib/gcc/xtensa-esp32-elf/8.4.0/../../../../xtensa-esp32-elf/include"
#include "..." search starts here:
#include <...> search starts here:
 /home/zip1982b/.espressif/tools/xtensa-esp32-elf/esp-2021r2-8.4.0/xtensa-esp32-elf/bin/../lib/gcc/xtensa-esp32-elf/8.4.0/include
 /home/zip1982b/.espressif/tools/xtensa-esp32-elf/esp-2021r2-8.4.0/xtensa-esp32-elf/bin/../lib/gcc/xtensa-esp32-elf/8.4.0/include-fixed
[... cut ...]
  collapse library dir [/home/zip1982b/.espressif/tools/xtensa-esp32-elf/esp-2021r2-8.4.0/xtensa-esp32-elf/bin/../lib/gcc/xtensa-esp32-elf/8.4.0/../../../../xtensa-esp32-elf/lib] ==> [/home/zip1982b/.espressif/tools/xtensa-esp32-elf/esp-2021r2-8.4.0/xtensa-esp32-elf/xtensa-esp32-elf/lib]
  implicit libs: [stdc++;m;gcc;c;nosys;c;gcc]
  implicit dirs: [/home/zip1982b/.espressif/tools/xtensa-esp32-elf/esp-2021r2-8.4.0/xtensa-esp32-elf/lib/gcc/xtensa-esp32-elf/8.4.0;/home/zip1982b/.espressif/tools/xtensa-esp32-elf/esp-2021r2-8.4.0/xtensa-esp32-elf/lib/gcc;/home/zip1982b/.espressif/tools/xtensa-esp32-elf/esp-2021r2-8.4.0/xtensa-esp32-elf/xtensa-esp32-elf/lib]
  implicit fwks: []

zip1982b, 2022-02-20:
if there is a working version for docker, please give a link.

ftab, 2022-05-24:
This issue is caused by the change in esp-idf v5.0 to move some components out of the esp-idf repository and into the component registry. adf examples have not yet been updated for this upcoming version of esp-idf.

5.0 has not yet been released, but here's the relevant portion of the migration guide: https://docs.espressif.com/projects/esp-idf/en/latest/esp32/migration-guides/removed-components.html

The component manager has been around for a little while: https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-guides/tools/idf-component-manager.html

You may have better luck trying some current release of esp-idf such as 4.4.1. Try this:

```
cd ~/esp/esp-adf/esp-idf
git checkout v4.4.1
git submodule update --recursive --init
git clean -ffd
~/esp/esp-adf/esp-idf/install.sh
```

If you run into any errors during the submodule update, cd into the component directory it's complaining about and also do a `git clean -ffd` there. (This can happen often when switching between major versions, as a lot of submodule differences can happen and confuse git terribly)

And then make sure you're using the export.sh from ~/esp/esp-adf/esp-idf and not from the separate instance you installed at ~/esp/esp-idf

If you have an IDF_PATH variable set in your .bashrc or .profile, I'd recommend removing it, keeping only the ADF_PATH variable. (the export.sh adds the IDF_PATH variable for you)

Otherwise, if you need a newer version of esp-idf, I've found the workaround is fairly straightforward:

Either:

```
idf.py add-dependency jsmn
idf.py add-dependency nghttp
```

Or

Place a file called idf_component.yml in your main/ directory, e.g.:

```yml
dependencies:
  espressif/jsmn: "*"
  espressif/nghttp: "*"
  idf:
    version: ">=4.1.0"
```

(The add-dependency commands above create the same file)

If you're on a recent version of master, you will likely also need to enable backward compatibility legacy APIs in the FreeRTOS section of the menuconfig

closed 2022-08-09 by jason-mao