espressif/esp-adf #759: 语音识别wwe工程移植进micropython，启动micropython报错

Zhengqinze05, 2022-02-10:
单独在C的主函数中使用是没问题的，开启了micropython线程之后，就报错重启

Zhengqinze05, 2022-02-10:
单独开启
 xTaskCreatePinnedToCore(voice_read_task, "read_task", 4 * 1024, NULL, 5, NULL, 0);
线程没报错
[image]
加了micropython线程就报错重启了
以下是报错log
[image]
[image]

请问这是什么问题，如何解决呢。期待你们的回复

jason-mao, 2022-02-11:
@Zhengqinze05 可能是内存不足了，你把这个启动往前放，让它先起来然后加载其他任务，试试。

Zhengqinze05, 2022-02-21:
很感谢你的回复。我在C层面，把语音识别的任务调最高可以实现。但是现在的问题是，我在micropython中使用语音识别的线程，还是有内存错误。我打印出了内存信息，有6M左右的内存可以供语音识别使用，但还是内存申请失败。在C层面我看语音识别内存竟然要用到4M的内存吗？

Zhengqinze05, 2022-02-21:
[0;31mE (2311) I2S: register I2S object to platform failed[0m
[0;32mI (2321) MY_KORVO: I2S0, MCLK output by GPIO0[0m
[0;32mI (2321) PRINT_MEM10: Func:start_recorder, Line:249, MEM Total:6278163 Bytes, Inter:172591 Bytes, Dram:172591 Bytes
[0m
[0;32mI (2341) speech_cn: Recorder has been created[0m
[0;32mI (2341) PRINT_MEM11: Func:start_recorder, Line:261, MEM Total:6253659 Bytes, Inter:166403 Bytes, Dram:166403 Bytes
[0m
[0;32mI (2351) PRINT_MEM12: Func:start_recorder, Line:270, MEM Total:6253659 Bytes, Inter:166403 Bytes, Dram:166403 Bytes
[0m
[0;32mI (2361) PRINT_MEM_SR1: Func:recorder_sr_create, Line:556, MEM Total:6253583 Bytes, Inter:166403 Bytes, Dram:166403 Bytes
[0m

[17:49:02.019]收←◆model_name: hilexin7q8 model_data: /srmodel/hilexin7q8/wn7q8_data

[17:49:02.780]收←◆MC Quantized-8 wakeNet7: wakeNet7Q8_v2_hilexin_5_0.97_0.90, mode:2, p:3, (Jan  5 2022 11:31:01)

[17:49:02.879]收←◆Initial TWO-MIC auido front-end for speech recognition, mode:0, (Dec 10 2021 11:08:00)
[0;32mI (3351) PRINT_MEM_SR2: Func:recorder_sr_create, Line:558, MEM Total:4810271 Bytes, Inter:136739 Bytes, Dram:136739 Bytes
[0m
[0;32mI (3351) PRINT_MEM_SR3: Func:recorder_sr_create, Line:568, MEM Total:4810271 Bytes, Inter:136739 Bytes, Dram:136739 Bytes
[0m
model_name: mn3cn model_data: /srmodel/mn3cn/mn3cn_data

[17:49:04.815]收←◆Item psram alloc failed. Size: 240020 = 120000 x 2 + 16 + 4
item buff malloc fail
Guru Meditation Error: Core  0 panic'ed (StoreProhibited). Exception was unhandled.

Zhengqinze05, 2022-02-23:
你好，我发现语音识别初始化竟然需要占用7M的内存，之后才会释放出来，剩下4M内存可以用。如果有其他需要内存超过1M的任务先启动了，语音识别初始化的内存就不够用了。请问语音识别真的需要这么大内存吗，可以进行优化裁剪吗。
在recorder_sr.c文件的这部分代码中内存需要7M。
#ifdef CONFIG_USE_MULTINET
    if (cfg->multinet_init) {
        recorder_sr->mn_handle = multinet->create((const model_coeff_getter_t *)&MULTINET_COEFF, 5760);
        recorder_sr->mn_enable = true;
        AUDIO_NULL_CHECK(TAG, recorder_sr->mn_handle, goto _failed);
    }
#endif
[image]
在这个地方还有8M内存

[image]
这里还剩下7M

[image]
到了这里就只剩下1M多的内存，之后会释放掉3M左右的内存

JosephTang, 2022-02-25:
@Zhengqinze05 你好，因为现在sr的模型数据全部放到了psram中，在sr初始化的过程中的确会用到6 - 7m的psram空间，并在初始化完成后释放一部分。所以需要等sr的全部初始化之后再去初始化micropython，并将micropython使用的psram大小限制到4M以内。

jason-mao, 2022-08-09:
@Zhengqinze05 关闭该问题，如还有问题，请重新打开或者创建新 issue。

closed 2022-08-09 by jason-mao