espressif/esp-adf #760: idfv4.4 and latest adf compile error

achao1104, 2022-02-10:
D:/project/esp32/esp-adf/components/audio_stream/pwm_stream.c: In function 'timer_group_isr':
D:/project/esp32/esp-adf/components/audio_stream/pwm_stream.c:230:25: error: 'timg_dev_t' {aka 'volatile struct timg_dev_s'} has no member named 'int_st_timers'
     if (handle->timg_dev->int_st_timers.val & BIT(handle->config.timer_num)) {
                         ^~
D:/project/esp32/esp-adf/components/audio_stream/pwm_stream.c:231:27: error: 'timg_dev_t' {aka 'volatile struct timg_dev_s'} has no member named 'int_clr_timers'; did you mean 'int_clr'?
         handle->timg_dev->int_clr_timers.val |= (1UL << handle->config.timer_num);
                           ^~~~~~~~~~~~~~
                           int_clr
D:/project/esp32/esp-adf/components/audio_stream/pwm_stream.c:234:65: error: 'volatile union <anonymous>' has no member named 'tx_alarm_en'; did you mean 'alarm_en'?
     handle->timg_dev->hw_timer[handle->config.timer_num].config.tx_alarm_en = TIMER_ALARM_EN;
                                                                 ^~~~~~~~~~~
                                                                 alarm_en
D:/project/esp32/esp-adf/components/audio_stream/pwm_stream.c: In function 'audio_pwm_set_sample_rate':
D:/project/esp32/esp-adf/components/audio_stream/pwm_stream.c:440:81: error: 'volatile union <anonymous>' has no member named 'tx_divider'; did you mean 'divider'?
     div = (uint16_t)handle->timg_dev->hw_timer[handle->config.timer_num].config.tx_divider;

achao1104, 2022-02-10:
audio board:esp32-s2-kaluga-1 v1.2

jason-mao, 2022-02-11:
@achao1104 Thanks for your report, please provide full log for us, thanks.

jason-mao, 2022-08-09:
Closed, successfully build esp32-s2-kaluga board on v4.4.

closed 2022-08-09 by jason-mao