espressif/esp-adf #761: VoIP example掉线

davidHuW, 2022-02-10:
IDF:V4.4
ADF:Master
服务器：yate
开发板：esp32-lyrat
现象：esp32注册到yate，过15分钟左右掉线，需复位

【esp32日志】：
[0;32mI (35864) SIP: [1970-01-01/00:00:17]=======WRITE 0827 bytes>>[0m
[0;32mI (35864) SIP: 

OPTIONS sip:100@192.168.3.93:5060 SIP/2.0

Via: SIP/2.0/UDP 192.168.3.176:14040;branch=z9hG4bK--291103790;rport

From: <sip:100@192.168.3.93:5060>;tag=-1831371918

To: <sip:100@192.168.3.93:5060>

Contact: <sip:100@192.168.3.176:14040;transport=UDP>

Max-Forwards: 70

Call-ID: 62164195B9DF78A201378694790BA70B926D3E7652C2

CSeq: 3 OPTIONS

Expires: 3600

User-Agent: ESP32 SIP/2.0

Content-Length: 0

Allow: INVITE, ACK, CANCEL, BYE, UPDATE, REFER, MESSAGE, OPTIONS, INFO, SUBSCRIBE

Supported: replaces, norefersub, extended-refer, timer, X-cisco-serviceuri

Allow-Events: presence, kpml

Authorization: Digest username="100", realm="Yate", nonce="4f7f179a02d285ed5c686d254de38f0d.1644533087", uri="sip:192.168.3.93:5060", response="501b4e5ac6c331886c82a2530bebb7c7", algorithm=md5, nc=00000002, cnonce="fcca4917460b2935"

[0m
[0;32mI (35934) SIP: [1970-01-01/00:00:17]=======================>>[0m
[0;32mI (35964) SIP: [1970-01-01/00:00:17]<<=====READ 0316 bytes==[0m
[0;32mI (35964) SIP: 

SIP/2.0 100 Trying

Via: SIP/2.0/UDP 192.168.3.176:14040;branch=z9hG4bK--291103790;rport=14040;received=192.168.3.176

From: <sip:100@192.168.3.93:5060>;tag=-1831371918

To: <sip:100@192.168.3.93:5060>

Call-ID: 62164195B9DF78A201378694790BA70B926D3E7652C2

CSeq: 3 OPTIONS

Server: YATE/6.1.0

Content-Length: 0

[0m
[0;32mI (35984) SIP: [1970-01-01/00:00:17]<<======================[0m
[0;32mI (35994) SIP: [1970-01-01/00:00:17]<<=====READ 0392 bytes==[0m
[0;32mI (35994) SIP: 

SIP/2.0 200 OK

Via: SIP/2.0/UDP 192.168.3.176:14040;branch=z9hG4bK--291103790;rport=14040;received=192.168.3.176

[... 2661 lines cut ...]
[0;32mI (665134) SIP: 

【yate日志】：
2022-02-11_06:43:59.045644 <CpuLoad:STUB> System CPU load is not implemented for this OS
2022-02-11_06:43:59.045644 <cpuload:NOTE> System Cpu load not supported!
2022-02-11_06:44:47.376987 <sip:NOTE> Registered user '100' expires in 3600 s
2022-02-11_06:45:24.135978 <sip:NOTE> Registered user '100' expires in 3600 s
2022-02-11_06:59:14.404729 <sip:NOTE> Registered user '102' expires in 3600 s
2022-02-11_06:59:18.926463 <sipengine:MILD> Received To tag in 100 answer! (sender bug)
2022-02-11_06:59:32.731015 <sip:NOTE> Registered user '100' expires in 3600 s
2022-02-11_06:59:40.431887 <sipengine:MILD> Received To tag in 100 answer! (sender bug)
2022-02-11_06:59:55.910242 <sip:NOTE> Registered user '100' expires in 3600 s
2022-02-11_07:00:09.411720 <sipengine:MILD> Received To tag in 100 answer! (sender bug)
2022-02-11_07:01:14.391310 <sip:NOTE> Registered user '102' expires in 3600 s
2022-02-11_07:03:14.391891 <sip:NOTE> Registered user '102' expires in 3600 s
2022-02-11_07:05:14.391877 <sip:NOTE> Registered user '102' expires in 3600 s
2022-02-11_07:07:14.396759 <sip:NOTE> Registered user '102' expires in 3600 s
2022-02-11_07:09:14.404580 <sip:NOTE> Registered user '102' expires in 3600 s
2022-02-11_07:10:54.235430 <sipengine:MILD> Received To tag in 100 answer! (sender bug)
2022-02-11_07:11:14.395820 <sip:NOTE> Registered user '102' expires in 3600 s
2022-02-11_07:13:14.390013 <sip:NOTE> Registered user '102' expires in 3600 s
2022-02-11_07:13:31.415965 <sipengine:MILD> Received To tag in 100 answer! (sender bug)
2022-02-11_07:15:14.397306 <sip:NOTE> Registered user '102' expires in 3600 s
2022-02-11_07:17:14.393284 <sip:NOTE> Registered user '102' expires in 3600 s
2022-02-11_07:19:14.398493 <sip:NOTE> Registered user '102' expires in 3600 s

davidHuW, 2022-02-13:
[22/02/13 11:22:05.276] eSipUAFsmBj2() - SIP:UA[15], call[1012] timeout to receive (1XX) for INVITE - <200=>100> 1644722513563343401-0@192.168.2.218
[22/02/13 11:22:05.276] eSipUAFsmBj2() - inivte timeout, clear ip ext (14) reg info: b02a8c0:14173, exp:3600!

lonelyjesse, 2022-02-13:
Hi, @Jason-mao 
I have the same problem with my envrionment.  Would you please help me out.
It's an emergency, our projects are based on voip example.

ahhfzhang, 2022-02-14:
Hi, @davidHuW 

The YATE server has not been updated for many years. 
It is recommended to use FreeSWITCH to test first and modify some config according to the README. 
We will build a YATE server to test it later.

Thanks.

Bruce213, 2022-02-14:
Hi，@ahhfzhang
i have tried myvoipserver，yate and a Commercial ip-pbx hardware ，
esp-sip with all of these platforms will lost connection after a while.
Please help to check whether it is the setting problem or the protocol stack problem，
thanks.

ahhfzhang, 2022-02-15:
Hi,  @davidHuW 

We have been running on the YATE server for more than 10 hours without any offline problems. 

```
ADF: master 
IDF: release/v4.4
```

But I also saw some abnormal logs, like `SIP: Retransmit again`, maybe you can try using `TCP` or `TLS` transport.

ahhfzhang, 2022-02-15:
> Hi, @jason-mao I have the same problem with my envrionment. Would you please help me out. It's an emergency, our projects are based on voip example.

Hi @lonelyjesse  

You can create a new issue and give us as much information and logs as possible.

Thanks.

Bruce213, 2022-02-15:
> Hi, @davidHuW
> 
> We have been running on the YATE server for more than 10 hours without any offline problems.
> 
> ```
> ADF: master 
> IDF: release/v4.4
> ```
> 
> But I also saw some abnormal logs, like `SIP: Retransmit again`, maybe you can try using `TCP` or `TLS` transport.

Hi @ahhfzhang 
you mean it is a network problem ?

ahhfzhang, 2022-02-15:
Yes, maybe the network is not very stable, so the server does not receive the UDP packets sent by ESP.

Bruce213, 2022-02-15:
> Yes, maybe the network is not very stable, so the server does not receive the UDP packets sent by ESP.

then , does esp have some timeout vailable about sip message transmission?
or can i switch the transmission to eth channel？ what & how should i do to get it?
thanks in advance.

ahhfzhang, 2022-02-15:
Yes, we will retransmit several times if it was time out.

about eth you can refer to idf eth example : 

https://github.com/espressif/esp-idf/tree/master/examples/ethernet

Bruce213, 2022-02-15:
> Yes, we will retransmit several times if it was time out.
> 
> about eth you can refer to idf eth example :
> 
> https://github.com/espressif/esp-idf/tree/master/examples/ethernet

should i add the eth driver to components/esp_peripherals and register the peripheral to esp_peripherals.c ?
is there any specification about the driver transplant ?
thanks.

ahhfzhang, 2022-02-21:
Hi @davidHuW 

No, the connection is stable, but the server does not respond or the response is timeout, we will retransmit again.

I also found a similar problem because the YATE server did not reply to the esp client's requests, which caused the retransmission.

jason-mao, 2022-08-09:
Hi guys，hope and appreciate any progress update from your side.

closed 2023-01-19 by espressif-bot