espressif/esp-adf #762: VoIP register become offline after a while

davidHuW, 2022-02-13:
IDF:V4.4
ADF:Master

rst:0x1 (POWERON_RESET),boot:0x1f (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0030,len:6948
load:0x40078000,len:15512
load:0x40080400,len:4492
entry 0x400806b4
[0;32mI (27) boot: ESP-IDF v4.4-157-gc7b9f94-dirty 2nd stage bootloader[0m
[0;32mI (27) boot: compile time 10:52:39[0m
[0;32mI (27) boot: chip revision: 3[0m
[0;32mI (32) boot_comm: chip revision: 3, min. bootloader chip revision: 0[0m
[0;32mI (39) qio_mode: Enabling default flash chip QIO[0m
[0;32mI (44) boot.esp32: SPI Speed      : 80MHz[0m
[0;32mI (49) boot.esp32: SPI Mode       : QIO[0m
[0;32mI (53) boot.esp32: SPI Flash Size : 8MB[0m
[0;32mI (58) boot: Enabling RNG early entropy source...[0m
[0;32mI (63) boot: Partition Table:[0m
[0;32mI (67) boot: ## Label            Usage          Type ST Offset   Length[0m
[0;32mI (74) boot:  0 nvs              WiFi data        01 02 00009000 00006000[0m
[0;32mI (81) boot:  1 phy_init         RF data          01 01 0000f000 00001000[0m
[0;32mI (89) boot:  2 factory          factory app      00 00 00010000 00200000[0m
[0;32mI (96) boot:  3 flash_tone       Unknown data     01 ff 00210000 0007d000[0m
[0;32mI (104) boot: End of partition table[0m
[0;32mI (108) boot_comm: chip revision: 3, min. application chip revision: 0[0m
[0;32mI (115) esp_image: segment 0: paddr=00010020 vaddr=3f400020 size=30ad8h (199384) map[0m
[0;32mI (179) esp_image: segment 1: paddr=00040b00 vaddr=3ffb0000 size=03b10h ( 15120) load[0m
[0;32mI (184) esp_image: segment 2: paddr=00044618 vaddr=40080000 size=0ba00h ( 47616) load[0m
[0;32mI (200) esp_image: segment 3: paddr=00050020 vaddr=400d0020 size=d48e4h (870628) map[0m
[0;32mI (440) esp_image: segment 4: paddr=0012490c vaddr=4008ba00 size=0b704h ( 46852) load[0m
[0;32mI (456) esp_image: segment 5: paddr=00130018 vaddr=50000000 size=00010h (    16) load[0m
[0;32mI (467) boot: Loaded app from partition at offset 0x10000[0m
[0;32mI (467) boot: Disabling RNG early entropy source...[0m
[0;32mI (479) psram: This chip is ESP32-D0WD[0m
[0;32mI (479) spiram: Found 64MBit SPI RAM device[0m
[0;32mI (480) spiram: SPI RAM mode: flash 80m sram 80m[0m
[0;32mI (482) spiram: PSRAM initialized, cache is in low/high (2-core) mode.[0m
[0;32mI (489) cpu_start: Pro cpu up.[0m
[0;32mI (493) cpu_start: Starting app cpu, entry point is 0x400816c8[0m
[0;32mI (0) cpu_start: App cpu up.[0m
[0;32mI (995) spiram: SPI SRAM memory test OK[0m
[0;32mI (1003) cpu_start: Pro cpu start user code[0m
[0;32mI (1003) cpu_start: cpu freq: 240000000[0m
[0;32mI (1004) cpu_start: Application information:[0m
[0;32mI (1007) cpu_start: Project name:     voip_app[0m
[0;32mI (1012) cpu_start: App version:      v2.3-205-gecf62b4-dirty[0m
[0;32mI (1018) cpu_start: Compile time:     Feb 13 2022 10:53:37[0m
[0;32mI (1024) cpu_start: ELF file SHA256:  d102097703c17189...[0m
[0;32mI (1030) cpu_start: ESP-IDF:          v4.4-157-gc7b9f94-dirty[0m
[0;32mI (1037) heap_init: Initializing. RAM available for dynamic allocation:[0m
[0;32mI (1044) heap_init: At 3FFAE6E0 len 00001920 (6 KiB): DRAM[0m
[0;32mI (1050) heap_init: At 3FFB5230 len 0002ADD0 (171 KiB): DRAM[0m
[0;32mI (1057) heap_init: At 3FFE0440 len 00003AE0 (14 KiB): D/IRAM[0m
[0;32mI (1063) heap_init: At 3FFE4350 len 0001BCB0 (111 KiB): D/IRAM[0m
[0;32mI (1070) heap_init: At 40097104 len 00008EFC (35 KiB): IRAM[0m
[0;32mI (1076) spiram: Adding pool of 4084K of external SPI memory to heap allocator[0m
[0;32mI (1084) spi_flash: detected chip: gd[0m
[0;32mI (1088) spi_flash: flash io: qio[0m
[0;32mI (1093) cpu_start: Starting scheduler on PRO CPU.[0m
[0;32mI (0) cpu_start: Starting scheduler on APP CPU.[0m
[0;32mI (1103) spiram: Reserving pool of 18K of internal memory for DMA/internal allocations[0m
[0;32mI (1143) VOIP_EXAMPLE: [1.0] Initialize peripherals management[0m
[0;32mI (1143) VOIP_EXAMPLE: [1.1] Initialize and start peripherals[0m
[0;32mI (1143) gpio: GPIO[36]| InputEn: 1| OutputEn: 0| OpenDrain: 0| Pullup: 1| Pulldown: 0| Intr:3 [0m
[0;32mI (1153) gpio: GPIO[39]| InputEn: 1| OutputEn: 0| OpenDrain: 0| Pullup: 1| Pulldown: 0| Intr:3 [0m
[0;32mI (1163) VOIP_EXAMPLE: [1.2] Create and start input key service[0m
[0;33mW (1183) PERIPH_TOUCH: _touch_init[0m
[... 2216 lines cut ...]
[0;32mI (482703) ESP_AUDIO_CTRL: Request_CMD_Queue CMD:2, Available:5, que:0x3ffaf7a4[0m
[0;32mI (482713) ESP_AUDIO_TASK: MEDIA_CTRL_EVT_STOP, status:RUNNING[0m
[0;31mE (482713) AUDIO_ELEMENT: [IN_flash] Element already stopped[0m
[0;31mE (482723) AUDIO_ELEMENT: [DEC_mp3] Element already stopped[0m
[0;31mE (482723) AUDIO_ELEMENT: [OUT_iis] Element already stopped[0m
[0;31mE (482733) AUDIO_ELEMENT: [Audio_forge] Element already stopped[0m
[0;32mI (482743) ESP_AUDIO_TASK: MEDIA_CTRL_EVT_STOP Done, dat:0[0m
[0;32mI (482743) ESP_AUDIO_TASK: Recv Element[IN_flash-0x3f804e90] MSG,type:20000,cmd:8,len:4,status:AEL_STATUS_STATE_STOPPED[0m
[0;33mW (482763) AUDIO_PIPELINE: Check AEL STOPPED, pl:0x3f803d2c, el:0x3f80501c, tag:         DEC_mp3, state:15, wanted:14[0m
[0;33mW (482773) AUDIO_PIPELINE: Check AEL STOPPED, pl:0x3f803d2c, el:0x3f818014, tag:     Audio_forge, state:15, wanted:14[0m
[0;32mI (482783) ESP_AUDIO_TASK: Received last time: 3056 ms[0m
[0;32mI (482783) ESP_AUDIO_TASK: Recv Element[OUT_iis-0x3f8051c4] MSG,type:20000,cmd:8,len:4,status:AEL_STATUS_STATE_FINISHED[0m
[0;33mW (482803) AUDIO_PIPELINE: Check AEL FINISHED, pl:0x3f803d2c, el:0x3f804e90, tag:        IN_flash, state:14, wanted:15[0m
[0;32mI (482813) ESP_AUDIO_TASK: ESP_AUDIO status is AEL_STATUS_STATE_FINISHED, 0, src:0, is_stopping:1[0m
[0;32mI (482823) ESP_AUDIO_TASK: Func:media_ctrl_task, Line:984, MEM Total:4239651 Bytes, Inter:233203 Bytes, Dram:198751 Bytes

[0m
[0;33mW (482833) ESP_AUDIO_TASK: Destroy the old pipeline, FINISHED[0m
[0;33mW (482833) ESP_AUDIO_TASK: The old pipeline destroyed, FINISHED[0m
[0;32mI (482843) ESP_AUDIO_TASK: Release the seeking process due to stopping[0m
[0;32mI (482843) ESP_AUDIO_CTRL: Exit media_ctrl_stop procedure, ret:0[0m
[0;32mI (482853) ESP_AUDIO_TASK: Recv Element[DEC_mp3-0x3f80501c] MSG,type:20000,cmd:8,len:4,status:AEL_STATUS_STATE_STOPPED[0m
[0;32mI (482873) ESP_AUDIO_TASK: Recv Element[OUT_iis-0x3f8051c4] MSG,type:20000,cmd:8,len:4,status:AEL_STATUS_STATE_STOPPED[0m
[0;32mI (482883) ESP_AUDIO_TASK: Recv Element[Audio_forge-0x3f818014] MSG,type:20000,cmd:8,len:4,status:AEL_STATUS_STATE_STOPPED[0m
[0;32mI (482893) ESP_AUDIO_TASK: Recv Element[IN_flash-0x3f804e90] MSG,type:20000,cmd:8,len:4

ahhfzhang, 2022-02-15:
Hi @davidHuW 

I don't see where the offline appears, because the `INVITE` message is still received at the end.

But I also saw some abnormal logs, such as `403 Forbidden` message when the ESP client sent the option to the server, so I guess your server doesn't support `OPTION` message, you can close it by setting `send_options = false` in `sip_config_t`.

hope it works for you.

Bruce213, 2022-02-15:
hi @ahhfzhang 
i have tried some three sip server include myvoipserver,yate and om20(one ipbx server) , 
event if i don't do anything about esp32 after it registed on the sip server , 
the esp client will drop the connection from sip server, but the heart message is still send and the sip server ack is ok,
it is strange . 
hope some help.
thanks.

ahhfzhang, 2022-02-15:
Hi @davidHuW  

You can try Freeswitch, the yate server sometimes responds too slow.

br.

jason-mao, 2022-08-09:
@davidHuW So appreciate any progress update from your side.

closed 2023-01-19 by espressif-bot