espressif/esp-adf #763: ADF eth support

Bruce213, 2022-02-14:
hi，
i have a project which needs eth communication，but i can not find eth peripheral in the adf drivers，
if i want to use adf in my project with eth，should i need to develop eth driver by myself？
thanks.

HengYongChao, 2022-02-15:
Hi @Bruce213 

I have written a demo for uploading recording data using Ethernet, please refer to the following.

[recorder_raw_http_ethernet.zip](https://github.com/espressif/esp-adf/files/8067334/recorder_raw_http_ethernet.zip)

davidHuW, 2022-02-15:
> Hi @Bruce213
> 
> I have written a demo for uploading recording data using Ethernet, please refer to the following.
> 
> [recorder_raw_http_ethernet.zip](https://github.com/espressif/esp-adf/files/8067334/recorder_raw_http_ethernet.zip)

Hi @HengYongChao 
thank you for the demo .
i have another question , i use esp lyrat V4.3 board and we have an ETH custom board with the same MCU,but some  differences in GPIO assignment . which Audio_HAL type should i choose ( lyrat v4.3 or custom board?) ,
Is there anything else i need to fit in?
thanks in advance.

HengYongChao, 2022-02-16:
Hi @davidHuW 

I am using the [ESP32-Ethernet-Kit](https://docs.espressif.com/projects/esp-idf/en/latest/esp32/hw-reference/esp32/get-started-ethernet-kit.html#get-started-esp32-ethernet-kit-v1-2) development board, the schematic diagram is as follows:https://docs.espressif.com/projects/esp-idf/en/latest/esp32/hw-reference/esp32/get-started-ethernet-kit.html

Bruce213, 2022-02-17:
[image]

ifi_service.dir/airkiss_config/airkiss_config.c.obj   -c /home/hu/esp/esp-adf/components/wifi_service/airkiss_config/airkiss_config.c
/home/hu/esp/esp-adf/components/wifi_service/airkiss_config/airkiss_config.c: In function 'airkiss_send_ack_task':
/home/hu/esp/esp-adf/components/wifi_service/airkiss_config/airkiss_config.c:310:15: error: implicit declaration of function 'tcpip_adapter_get_ip_info'; did you mean 'esp_netif_get_ip_info'? [-Werror=implicit-function-declaration]
         ret = tcpip_adapter_get_ip_info(TCPIP_ADAPTER_IF_STA, &local_ip);

Hi @HengYongChao 
i build you demo project in my idf v4.4 and adf(master 2022,1,14) , but some errors about IDF occured.
please help to check my environment.
thanks.