espressif/esp-adf #768: pipeline_passthru example not working

tozon, 2022-02-17:
I've tested pipeline_passthru example with LyraT v4.3 board, IDF v4.4 and ADF v2.3.
There are 2 problems with this example:
1. "E (444) I2S: register I2S object to platform failed" error message is visible in the log.
2. If ESP32 CPU is stopped by pressing and holding reset button or shortcut is made on LyraT board I2S connector, audio signal is still being played from input to output jack. These tests prove, that there is no real I2S transmission done, but internal codec chip connection is used to connect input and output jack directly. This is not what should be done according to example's decription.
[pipeline_passthru_run(1).log](https://github.com/espressif/esp-adf/files/8092243/pipeline_passthru_run.1.log)

HengYongChao, 2022-02-23:
Hi @tozon 

1. This error print was introduced by IDF v4.4, replacing the original reinstall warning, which is essentially a warning to repeat the installation of i2s. 
2. The current passthru example is an internal bypass connection configuration, which is indeed different from the readme documentation. It will be updated the code to config pass data  through the esp32 i2s interface for optional.

The passthru functions are required to configure registers 38 and register 39 of the es8388.
[image]

[image]

tozon, 2022-02-23:
OK, I understand. What I need is an example, where the same I2S interface is used for both recording and playback at the same time. I hope with the option you have mentioned it will be possible.

jason-mao, 2022-08-08:
@tozon So appreciate any progress update from your side.

closed 2023-09-05 by jason-mao