espressif/esp-adf #772: Dueros相关的例程wifi连接失败后自动重启

Beanv587, 2022-02-23:
IDF:release/4.4
ADF:master

IDF、ADF都是最新的,子模块也都是拉取的最新的

包含的此BUG的例子:examples\advanced_examples\esp_dispatcher_dueros、examples\dueros

idf.py menuconfig里已经正确配置CONFIG_WIFI_SSID、CONFIG_WIFI_PASSWORD

测试别的例子配置wifi后能正常运行.打印日志如下(PS:日志里的(3494)显示ssid、pwd都是空的...问题应该在这儿):

entry 0x40080694
I (27) boot: ESP-IDF v4.4-dirty 2nd stage bootloader
I (27) boot: compile time 15:42:51
I (27) boot: chip revision: 3
I (30) boot_comm: chip revision: 3, min. bootloader chip revision: 0
I (37) boot.esp32: SPI Speed      : 80MHz
I (42) boot.esp32: SPI Mode       : DIO
I (47) boot.esp32: SPI Flash Size : 8MB
I (51) boot: Enabling RNG early entropy source...
I (57) boot: Partition Table:
I (60) boot: ## Label            Usage          Type ST Offset   Length
I (67) boot:  0 nvs              WiFi data        01 02 00009000 00006000
I (75) boot:  1 phy_init         RF data          01 01 0000f000 00001000
I (82) boot:  2 factory          factory app      00 00 00010000 00300000
I (90) boot: End of partition table
I (94) boot_comm: chip revision: 3, min. application chip revision: 0
I (101) esp_image: segment 0: paddr=00010020 vaddr=3f400020 size=1a27c0h (1714112) map
I (628) esp_image: segment 1: paddr=001b27e8 vaddr=3ffb0000 size=042b4h ( 17076) load
I (633) esp_image: segment 2: paddr=001b6aa4 vaddr=40080000 size=09574h ( 38260) load
I (647) esp_image: segment 3: paddr=001c0020 vaddr=400d0020 size=130b70h (1248112) map
I (1024) esp_image: segment 4: paddr=002f0b98 vaddr=40089574 size=1404ch ( 81996) load
I (1053) esp_image: segment 5: paddr=00304bec vaddr=50000000 size=00010h (    16) load
I (1068) boot: Loaded app from partition at offset 0x10000
I (1068) boot: Disabling RNG early entropy source...
I (1080) psram: This chip is ESP32-D0WD
I (1080) spiram: Found 64MBit SPI RAM device
I (1080) spiram: SPI RAM mode: flash 80m sram 80m
I (1085) spiram: PSRAM initialized, cache is in low/high (2-core) mode.
I (1093) cpu_start: Pro cpu up.
I (1097) cpu_start: Starting app cpu, entry point is 0x40081904
0x40081904: call_start_cpu1 at E:/esp32/esp-idf-v4.4/components/esp_system/port/cpu_start.c:156

I (1090) cpu_start: App cpu up.
I (1602) spiram: SPI SRAM memory test OK
I (1611) cpu_start: Pro cpu start user code
I (1611) cpu_start: cpu freq: 240000000
I (1611) cpu_start: Application information:
I (1614) cpu_start: Project name:     esp_dispatcher_dueros
I (1620) cpu_start: App version:      1
I (1625) cpu_start: Compile time:     Feb 23 2022 15:42:29
I (1631) cpu_start: ELF file SHA256:  7a417e150da9ad85...
I (1637) cpu_start: ESP-IDF:          v4.4-dirty
I (1642) heap_init: Initializing. RAM available for dynamic allocation:
I (1649) heap_init: At 3FFAE6E0 len 00001920 (6 KiB): DRAM
I (1656) heap_init: At 3FFB7520 len 00028AE0 (162 KiB): DRAM
I (1662) heap_init: At 3FFE0440 len 00003AE0 (14 KiB): D/IRAM
I (1668) heap_init: At 3FFE4350 len 0001BCB0 (111 KiB): D/IRAM
I (1675) heap_init: At 4009D5C0 len 00002A40 (10 KiB): IRAM
I (1681) spiram: Adding pool of 4084K of external SPI memory to heap allocator
I (1690) spi_flash: detected chip: generic
I (1694) spi_flash: flash io: dio
I (1699) cpu_start: Starting scheduler on PRO CPU.
I (0) cpu_start: Starting scheduler on APP CPU.
I (1713) spiram: Reserving pool of 32K of internal memory for DMA/internal allocations
I (1758) DISPATCHER_DUEROS: ADF version is 
I (1758) DISPATCHER_DUEROS: Step 1. Create dueros_speaker instance
E (1759) DISPATCHER: exe first list: 0x0
I (1763) DISPATCHER: dispatcher_event_task is running...
I (1769) DISPATCHER_DUEROS: [Step 2.0] Create esp_periph_set_handle_t instance and initialize Touch, Button, SDcard
[... 119 lines cut ...]
I (3494) WIFI_SERV: Connect to wifi ssid: , pwd: 
ESP_ERROR_CHECK failed: esp_err_t 0x300a (ESP_ERR_WIFI_SSID) at 0x40090c44
0x40090c44: _esp_error_check_failed at E:/esp32/esp-idf-v4.4/components/esp_system/esp_err.c:42

file: "E:/esp32/esp-adf/components/wifi_service/src/wifi_service.c" line 425
func: wifi_task
expression: esp_wifi_connect()

abort() was called at PC 0x40090c47 on core 0
0x40090c47: _esp_error_check_failed at E:/esp32/esp-idf-v4.4/components/esp_system/esp_err.c:43

Backtrace:0x40082455:0x3f82cbb00x40090c51:0x3f82cbd0 0x40098175:0x3f82cbf0 0x40090c47:0x3f82cc60 0x400dff33:0x3f82cc80
0x40082455: panic_abort at E:/esp32/esp-idf-v4.4/components/esp_system/panic.c:402

0x40090c51: esp_system_abort at E:/esp32/esp-idf-v4.4/components/esp_system/esp_system.c:121

0x40098175: abort at E:/esp32/esp-idf-v4.4/components/newlib/abort.c:46

0x40090c47: _esp_error_check_failed at E:/esp32/esp-idf-v4.4/components/esp_system/esp_err.c:43

0x400dff33: wifi_task at E:/esp32/esp-adf/components/wifi_service/src/wifi_service.c:425 (discriminator 1)

ELF file SHA256: 7a417e150da9ad85

Rebooting...

HengYongChao, 2022-02-23:
Hi @Beanv587 

请问你的 WiFi 的 SSID 和 PWD 是特殊字符吗？ 可使用普通的英文字母做测试有没有一样的问题呢？ 另外可以擦除 整个 flash  再尝试。

你说可以使用别的例程链接成功。可否上传 sdkconfig 配置看一看呢， 我的本地测试链接正常，无法复现你的问题，我的sdkconfig 配置如下供你参考，[sdkconfig.zip](https://github.com/espressif/esp-adf/files/8123168/sdkconfig.zip)。

Beanv587, 2022-02-23:
> Hi @Beanv587
> 
> 关于你的 WiFi 配置是空这个问题的确有些奇怪，请问你的 WiFi 的 SSID 和 PWD 是特殊字符吗？ 可使用普通的英文字母做测试有没有一样的问题呢？ 另外可以擦除 整个 flash 再烧录尝试。
> 
> 你说可以使用别的例程链接成功。可否上传 sdkconfig 配置看一看呢， 我的本地测试链接正常，无法复现你的问题，我的sdkconfig 配置如下供你参考，[sdkconfig.zip](https://github.com/espressif/esp-adf/files/8123168/sdkconfig.zip)。

@HengYongChao 按你的建议,我把flash全部擦除后重新烧录,问题解决.但是很奇怪程序有异常.我稍后把详细日志发上来.

我的开发板是Lyrat V4.3。

Beanv587, 2022-02-23:
entry 0x400806b4
I (27) boot: ESP-IDF v4.4-dirty 2nd stage bootloader
I (27) boot: compile time 20:56:13
I (27) boot: chip revision: 3
I (30) boot_comm: chip revision: 3, min. bootloader chip revision: 0
I (38) qio_mode: Enabling default flash chip QIO
I (43) boot.esp32: SPI Speed      : 80MHz
I (47) boot.esp32: SPI Mode       : QIO
I (52) boot.esp32: SPI Flash Size : 8MB
I (57) boot: Enabling RNG early entropy source...
I (62) boot: Partition Table:
I (66) boot: ## Label            Usage          Type ST Offset   Length
I (73) boot:  0 nvs              WiFi data        01 02 00009000 00006000
I (80) boot:  1 phy_init         RF data          01 01 0000f000 00001000
I (88) boot:  2 factory          factory app      00 00 00010000 00300000
I (95) boot: End of partition table
I (99) boot_comm: chip revision: 3, min. application chip revision: 0
I (107) esp_image: segment 0: paddr=00010020 vaddr=3f400020 size=68738h (427832) map
I (233) esp_image: segment 1: paddr=00078760 vaddr=3ffb0000 size=041c4h ( 16836) load
I (238) esp_image: segment 2: paddr=0007c92c vaddr=40080000 size=036ech ( 14060) load
I (243) esp_image: segment 3: paddr=00080020 vaddr=400d0020 size=14272ch (1320748) map
I (611) esp_image: segment 4: paddr=001c2754 vaddr=400836ec size=134f0h ( 79088) load
I (637) esp_image: segment 5: paddr=001d5c4c vaddr=50000000 size=00010h (    16) load
I (649) boot: Loaded app from partition at offset 0x10000
I (649) boot: Disabling RNG early entropy source...
I (660) psram: This chip is ESP32-D0WD
I (660) spiram: Found 64MBit SPI RAM device
I (661) spiram: SPI RAM mode: flash 80m sram 80m
I (666) spiram: PSRAM initialized, cache is in low/high (2-core) mode.
I (673) cpu_start: Pro cpu up.
I (677) cpu_start: Starting app cpu, entry point is 0x400816b4
0x400816b4: call_start_cpu1 at E:/esp32/esp-idf-v4.4/components/esp_system/port/cpu_start.c:156

I (0) cpu_start: App cpu up.
I (1158) spiram: SPI SRAM memory test OK
I (1166) cpu_start: Pro cpu start user code
I (1166) cpu_start: cpu freq: 240000000
I (1167) cpu_start: Application information:
I (1170) cpu_start: Project name:     esp_dueros
I (1175) cpu_start: App version:      1
I (1179) cpu_start: Compile time:     Feb 23 2022 20:59:27
I (1185) cpu_start: ELF file SHA256:  6c60573d06ff1838...
I (1192) cpu_start: ESP-IDF:          v4.4-dirty
I (1197) heap_init: Initializing. RAM available for dynamic allocation:
I (1204) heap_init: At 3FFAE6E0 len 00001920 (6 KiB): DRAM
I (1210) heap_init: At 3FFB7458 len 00028BA8 (162 KiB): DRAM
I (1217) heap_init: At 3FFE0440 len 00003AE0 (14 KiB): D/IRAM
I (1223) heap_init: At 3FFE4350 len 0001BCB0 (111 KiB): D/IRAM
I (1230) heap_init: At 40096BDC len 00009424 (37 KiB): IRAM
I (1236) spiram: Adding pool of 4084K of external SPI memory to heap allocator
I (1244) spi_flash: detected chip: generic
I (1249) spi_flash: flash io: qio
I (1254) cpu_start: Starting scheduler on PRO CPU.
I (0) cpu_start: Starting scheduler on APP CPU.
I (1268) spiram: Reserving pool of 32K of internal memory for DMA/internal allocations
I (1294) DUEROS: ADF version is 
I (1295) gpio: GPIO[19]| InputEn: 1| OutputEn: 0| OpenDrain: 0| Pullup: 1| Pulldown: 0| Intr:3 
E (1296) gpio: gpio_install_isr_service(449): GPIO isr service already installed
I (1315) gpio: GPIO[21]| InputEn: 0| OutputEn: 1| OpenDrain: 0| Pullup: 0| Pulldown: 0| Intr:0 
I (1315) ES8388_DRIVER: init,out:02, in:00
I (1324) gpio: GPIO[36]| InputEn: 1| OutputEn: 0| OpenDrain: 0| Pullup: 1| Pulldown: 0| Intr:3 
I (1328) gpio: GPIO[39]| InputEn: 1| OutputEn: 0| OpenDrain: 0| Pullup: 1| Pulldown: 0| Intr:3 
W (1363) PERIPH_TOUCH: _touch_init
I (1363) SDCARD: Using 1-line SD mode, 4-line SD mode,  base path=/sdcard
I (1429) SDCARD: CID name ACLCE!

E (1837) DISPATCHER: exe first list: 0x0
I (1838) DISPATCHER: dispatcher_event_task is running...
I (1841) AUDIO_THREAD: The wifi_serv task allocate stack on external memory
I (1847) wifi:wifi driver task: 3ffcdc14, prio:23, stack:6656, core=0
[... 225 lines cut ...]
E (19184,tid:3ffde0a0) lightduer_dcs_system.c( 315):error: didp product not exsists
I (19470,tid:3ffde0a0) lightduer_dcs_router.c( 460):Directive name: ThrowException
E (19470,tid:3ffde0a0) lightduer_dcs_system.c( 34)rr code: INTERNAL_SERVICE_EXCEPTION
E (19474,tid:3ffde0a0) lightduer_dcs_system.c( 315):error: didp product not exsists
I (19768,tid:3ffde0a0) lightduer_dcs_router.c( 460):Directive name: ThrowException
E (19768,tid:3ffde0a0) lightduer_dcs_system.c( 314):error code: INTERNAL_SERVICE_EXCEPTION
E (19772,tid:3ffde0a0) lightduer_dcs_system.c( 315):error: didp product not exsists
I (20106,tid:3ffde0a0) lightduer_dcs_router.c( 460):Directive name: ThrowException
E (20106,tid:3ffde0a0) lightduer_dcs_system.c( 314):error code: INTERNAL_SERVICE_EXCEPTION
E (20112,tid:3ffde0a0) lightduer_dcs_system.c( 315):error: didp product not exsists
I (20346,tid:3ffde0a0) lightduer_dcs_router.c( 460):Directive name: ThrowException
E (20346,tid:3ffde0a0) lightduer_dcs_system.c( 314):error code: INTERNAL_SERVICE_EXCEPTION
E (20352,tid:3ffde0a0) lightduer_dcs_system.c( 315):error: didp product not exsists
I (21761) DUEROS: rec_engine_cb - AUDIO_REC_VAD_STOP, state:4
I (21762) DUEROS: Dueros DUER_CMD_STOP
W (21787) DUEROS: duer_callback: type:0, source:0x3f80a51c data:5, data_len:0
I (20615,tid:3ffde0a0) lightduer_dcs_router.c( 460):Directive name: ThrowException
E (20615,tid:3ffde0a0) lightduer_dcs_system.c( 34)err deINLIREESEXPTN
E 06,t:3fda0litdercs_syem( 5)rr: dproct t sis
I (20625,tid:3ffde0a0) lightduer_dcs_router.c( 460):Directe me DloeFisd
I (20633,tid:3ffde0a0) lightduer_dcs_router.c( 460):Directive name: ThrowException
E (20635,tid:3ffde0a0) lightduer_dcs_system.c( 314):error code: INTERNAL_SERVICE_EXCEPTION
E (20643,tid:3ffde0a0) lightduer_dcs_system.c( 315):error: didp product not exsists
W (20658,tid:3ffde0a0) lightduer_engine.c( 697):data cache has not sent, pending..., dcache_len:, qchele1
I (22662) DUEROS: rec_engine_cb - AUDIO_REC_WAKEUP_END

Beanv587, 2022-02-23:
唤醒前从5573开始出现了错误,唤醒时(17830)无任何错误,唤醒后(19450)比如说今天天气怎么样,过程中就不断出现异常

HengYongChao, 2022-02-24:
Hi @Beanv587 

17830 后面的异常是指？ 无法播放下发的音频吗？ 还指后面提示的错误吗？

>E (19768,tid:3ffde0a0) lightduer_dcs_system.c( 314):error code: INTERNAL_SERVICE_EXCEPTION
> E (19772,tid:3ffde0a0) lightduer_dcs_system.c( 315):error: didp product not exsists`

这个是 dueros 库抛出的 内部服务异常，产品不存在？ 可能和你的 profile 有关，重新更换或者申请新的 profile ，再试试有无异常。

Beanv587, 2022-02-24:
> didp product not exsists

@HengYongChao 看来确实是,百度当初给的20个Profile.当一个profile绑定了一个设备后就无法用新的Profile了.我换成当初老的Profile问题就解决.谢谢

closed 2022-02-24 by Beanv587