espressif/esp-adf #774: Opus Example Files Report It is not Opus File

chadrockey, 2022-03-02:
When trying to run the play_http_select_decoder_example with SELECT_OPUS_DECODER, I received an error message stating that this was not a valid Opus file.

I got this for both the long example and short example from Opus.  Most sources didn't work, but I was able to set:

`static const char *selected_file_to_play = "http://eu-a.icecast.edge.streaminginter.net:8000/worksradio/opus_lq.opus";`

And this internet radio station worked well enough for my purposes.

I'm on custom hardware and had to make a lot of changes and a custom board, but I'd like someone to verify if the sample files can be played correctly.

https://espressif-docs.readthedocs-hosted.com/projects/esp-adf/en/latest/design-guide/audio-samples.html

Specifically [ff-16b-2c-44100hz.opus](https://dl.espressif.com/dl/audio/ff-16b-2c-44100hz.opus) and [gs-16b-2c-44100hz.opus](https://dl.espressif.com/dl/audio/gs-16b-2c-44100hz.opus)

`I (3550) HTTP_SELECT_OPUS_EXAMPLE: [ 4 ] Set up  event listener
I (3550) HTTP_SELECT_OPUS_EXAMPLE: [4.1] Listening event from all elements of pipeline
I (3554) HTTP_SELECT_OPUS_EXAMPLE: [4.2] Listening event from peripherals
I (3562) HTTP_SELECT_OPUS_EXAMPLE: [ 5 ] Start audio_pipeline
Playing file: https://dl.espressif.com/dl/audio/gs-16b-2c-44100hz.opus
E (5369) OPUS_DECODER: It is not opus file
E (5369) AUDIO_ELEMENT: [opus] ERROR_PROCESS, AEL_IO_FAIL
W (5369) AUDIO_ELEMENT: [opus] audio_element_on_cmd_error,3`

HengYongChao, 2022-03-03:
Hi @chadrockey 
It works for me, envs is  adf master and idf v4.4, opus link `static const char *selected_file_to_play = "http://eu-a.icecast.edge.streaminginter.net:8000/worksradio/opus_lq.opus"`; with the play_http_select_decoder_example.

```c
I (1079) cpu_start: Starting scheduler on PRO CPU.
I (0) cpu_start: Starting scheduler on APP CPU.
I (1093) spiram: Reserving pool of 32K of internal memory for DMA/internal allocations
I (1134) HTTP_SELECT_OPUS_EXAMPLE: [ 1 ] Start audio codec chip
I (1154) HTTP_SELECT_OPUS_EXAMPLE: [2.0] Create audio pipeline for playback
I (1154) HTTP_SELECT_OPUS_EXAMPLE: [2.1] Create http stream to read data
I (1158) HTTP_SELECT_OPUS_EXAMPLE: [2.2] Create opus decoder to decode opus file
I (1166) HTTP_SELECT_OPUS_EXAMPLE: [2.3] Create i2s stream to write data to codec chip
I (1176) HTTP_SELECT_OPUS_EXAMPLE: [2.4] Register all elements to audio pipeline
I (1183) HTTP_SELECT_OPUS_EXAMPLE: [2.5] Link it together http_stream-->opus_decoder-->i2s_stream-->[codec_chip]
I (1265) HTTP_SELECT_OPUS_EXAMPLE: [2.6] Set up  uri (http as http_stream, opus as opus_decoder, and default output is i2s)
I (1266) HTTP_SELECT_OPUS_EXAMPLE: [ 3 ] Start and wait for Wi-Fi network
E (1274) gpio: gpio_install_isr_service(449): GPIO isr service already installed
W (1301) phy_init: failed to load RF calibration data (0xffffffff), falling back to full calibration
W (3098) PERIPH_WIFI: WiFi Event cb, Unhandle event_base:WIFI_EVENT, event_id:4
W (4149) wifi:<ba-add>idx:0 (ifx:0, fc:ec:da:b7:11:c7), tid:4, ssn:1, winSize:64
I (5135) HTTP_SELECT_OPUS_EXAMPLE: [ 4 ] Set up  event listener
I (5135) HTTP_SELECT_OPUS_EXAMPLE: [4.1] Listening event from all elements of pipeline
I (5140) HTTP_SELECT_OPUS_EXAMPLE: [4.2] Listening event from peripherals
W (5143) wifi:<ba-add>idx:1 (ifx:0, fc:ec:da:b7:11:c7), tid:0, ssn:0, winSize:64
I (5156) HTTP_SELECT_OPUS_EXAMPLE: [ 5 ] Start audio_pipeline
I (8013) HTTP_SELECT_OPUS_EXAMPLE: [ * ] Receive music info from opus decoder, sample_rates=48000, bits=16, ch=2
```

chadrockey, 2022-03-03:
@HengYongChao yes, that's exactly what I was also able to get working.

What I couldn't get working is the default files in the source tree:
https://github.com/espressif/esp-adf/blob/master/examples/player/pipeline_http_select_decoder/main/play_http_select_decoder_example.c#L71

HengYongChao, 2022-03-03:
> @HengYongChao yes, that's exactly what I was also able to get working.
> 
> What I couldn't get working is the default files in the source tree: https://github.com/espressif/esp-adf/blob/master/examples/player/pipeline_http_select_decoder/main/play_http_select_decoder_example.c#L71

Hi @chadrockey 

Thank you for your correction, I did reproduce the above problem, and I will submit a fix as soon as possible after identifying the problem.

HengYongChao, 2022-03-03:
Hi @chadrockey 

I forgot to tell you that you may need to apply a patch in adf folder, [adf_http_stream.patch](https://github.com/espressif/esp-adf/blob/master/idf_patches/adf_http_stream.patch) will fix this issue. Please try it after you apply the patch.

jason-mao, 2022-08-08:
@chadrockey So appreciate any progress update from your side.

chadrockey, 2022-08-08:
Sorry, the library as written was going to be difficult for my use case.  I needed to convert opus frame by opus frame and occasionally loop (like a video game), which could throw off this player because it’s too strict.  It’s enforces unnecessary restrictions like the stream is being consistent or a monotonically increasing time.

It might be alright for playing files, but I ended up compiling libopus directly and did so without great difficulty.  It now works exactly the same as the rest of the libopus wrappers and my application is working well.

jason-mao, 2022-08-09:
@chadrockey Thank you for your info. I will closed this issue, please create new one if you have any questions.

closed 2022-08-09 by jason-mao