espressif/esp-adf #776: Full Example for Audio Pipeline with Raw Stream set to AUDIO_STREAM_WRITER

chadrockey, 2022-03-03:
I'm trying to input data over a custom format so can't use http stream or sdcard or file stream and pass the bytes directly into the beginning of the audio pipeline.

This should be supported as this is a comment in raw_stream.h:

>  *        - AUDIO_STREAM_WRITER, e.g. [raw]->[codec-mp3]->[i2s]
https://github.com/espressif/esp-adf/blob/master/components/audio_stream/include/raw_stream.h#L43

This doesn't mean it only supports mp3 right?  I'm looking to use the Opus decoder.

I've found a few related issues, but my overall issue is that I can begin streaming the data, but I don't hear output (other examples work fine) and I don't get many events reported.

I see few others reporting some "fixes", but it's not clear to me if this still applies:
https://github.com/espressif/esp-adf/issues/175#issuecomment-500323647

And am I required to use "audio_player"?  Is there a solution similar to this but with audio pipeline?
https://github.com/espressif/esp-adf/issues/383#issuecomment-636417347

Here is the output of my modified script, I only receive 3 events and I'm not sure yet how to best interpret what they are telling me.

`I (0) cpu_start: Starting scheduler on APP CPU.
I (400) HTTP_SELECT_OPUS_EXAMPLE: [ 1 ] Start audio codec chip
I (401) HTTP_SELECT_OPUS_EXAMPLE: [2.0] Create audio pipeline for playback
I (404) HTTP_SELECT_OPUS_EXAMPLE: [2.1] Create RAW STREAM to read data
I (411) HTTP_SELECT_OPUS_EXAMPLE: [2.2] Create opus decoder to decode opus file
I (419) HTTP_SELECT_OPUS_EXAMPLE: [2.3] Create i2s stream to write data to codec chip
I (429) HTTP_SELECT_OPUS_EXAMPLE: [2.4] Register all elements to audio pipeline
I (436) HTTP_SELECT_OPUS_EXAMPLE: [2.5] Link it together raw_stream-->opus_decoder-->i2s_stream-->[codec_chip]
I (448) HTTP_SELECT_OPUS_EXAMPLE: [4.1] Listening event from all elements of pipeline
I (455) HTTP_SELECT_OPUS_EXAMPLE: [ 5 ] Start audio_pipeline
Top of loop
Starting opus task 
Cmd: 8   Data len: 4   Source Type: 131072   Need Free: 0
Reported State: 3
Raw Stream State: 3
Decoder State: 3
i2s State: 3
I (475) HTTP_SELECT_OPUS_EXAMPLE: [ * ] Receive music info from opus decoder, sample_rates=44100, bits=16, ch=2  total_bytes=0
Top of loop
Cmd: 8   Data len: 4   Source Type: 131072   Need Free: 0
Reported State: 3
Raw Stream State: 3
Decoder State: 3
i2s State: 3
I (500) HTTP_SELECT_OPUS_EXAMPLE: [ * ] Receive music info from opus decoder, sample_rates=44100, bits=16, ch=2  total_bytes=0
Top of loop
Cmd: 8   Data len: 4   Source Type: 131072   Need Free: 0
Reported State: 3
Raw Stream State: 3
Decoder State: 3
i2s State: 3
I (525) HTTP_SELECT_OPUS_EXAMPLE: [ * ] Receive music info from opus decoder, sample_rates=44100, bits=16, ch=2  total_bytes=0
Top of loop
Cmd: 8   Data len: 4   Source Type: 131072   Need Free: 0
Reported State: 3
Raw Stream State: 3
Decoder State: 3
i2s State: 3
I (549) HTTP_SELECT_OPUS_EXAMPLE: [ * ] Receive music info from opus decoder, sample_rates=44100, bits=16, ch=2  total_bytes=0
Top of loop
Writing Opus data 240 
Writing Opus data 240 
Writing Opus data 240 
Writing Opus data 240 
Writing Opus data 240 
Writing Opus data 240 `

chadrockey, 2022-03-03:
I received an email about not being able to sync this to JIRA?  Is there anything I need to do?

HengYongChao, 2022-03-04:
Hi @chadrockey,

I write a demo, its pipeline like " [opus_file_read_cb]-->opus_decoder-->i2s_stream-->[codec_chip] ", it works good for me. 

I uploaded it here for your reference. Please try it.
[play_opus_embed.zip](https://github.com/espressif/esp-adf/files/8184526/play_opus_embed.zip)

jason-mao, 2022-08-10:
@chadrockey So appreciate any progress update from your side.

chadrockey, 2022-10-11:
You can close thanks!

closed 2023-09-05 by jason-mao