espressif/esp-adf #777: pipeline block the voip process?

Bruce213, 2022-03-04:
Hi:
i use the voip example and transplant it to my custom board, 
i get some sip client to call my esp32 client , it ringing OK, 
and i can hung up the call prcocess before the esp32 client accept the call request （esp_sip_uac_invite(sip,true)）.
but if i accept the call request ( esp_sip_uac_invite(sip,true)), the whole process is blocking.

please give me some help.
thanks.

【My Environment】：
IDF : V4.3
ADF : lastest Master
OS : ubuntu 16.04

【Log】:
[0;32mI (10) boot: ESP-IDF v4.4-157-gc7b9f94-dirty 2nd stage bootloader[0m
[0;32mI (10) boot: compile time 21:03:12[0m
[0;32mI (11) boot: chip revision: 3[0m
[0;32mI (14) boot_comm: chip revision: 3, min. bootloader chip revision: 0[0m
[0;32mI (21) qio_mode: Enabling default flash chip QIO[0m
[0;32mI (26) boot.esp32: SPI Speed      : 80MHz[0m
[0;32mI (31) boot.esp32: SPI Mode       : QIO[0m
[0;32mI (35) boot.esp32: SPI Flash Size : 8MB[0m
[0;32mI (40) boot: Enabling RNG early entropy source...[0m
[0;32mI (45) boot: Partition Table:[0m
[0;32mI (49) boot: ## Label            Usage          Type ST Offset   Length[0m
[0;32mI (56) boot:  0 nvs              WiFi data        01 02 00009000 00006000[0m
[0;32mI (63) boot:  1 phy_init         RF data          01 01 0000f000 00001000[0m
[0;32mI (71) boot:  2 factory          factory app      00 00 00010000 00200000[0m
[0;32mI (78) boot:  3 flash_tone       Unknown data     01 ff 00210000 0007d000[0m
[0;32mI (86) boot: End of partition table[0m
[0;32mI (90) boot_comm: chip revision: 3, min. application chip revision: 0[0m
[0;32mI (97) esp_image: segment 0: paddr=00010020 vaddr=3f400020 size=4d708h (317192) map[0m
[0;32mI (193) esp_image: segment 1: paddr=0005d730 vaddr=3ffb0000 size=028e8h ( 10472) load[0m
[0;32mI (196) esp_image: segment 2: paddr=00060020 vaddr=400d0020 size=bd20ch (774668) map[0m
[0;32mI (412) esp_image: segment 3: paddr=0011d234 vaddr=3ffb28e8 size=00be8h (  3048) load[0m
[0;32mI (413) esp_image: segment 4: paddr=0011de24 vaddr=40080000 size=14d20h ( 85280) load[0m
[0;32mI (446) esp_image: segment 5: paddr=00132b4c vaddr=50000000 size=00010h (    16) load[0m
[0;32mI (456) boot: Loaded app from partition at offset 0x10000[0m
[0;32mI (456) boot: Disabling RNG early entropy source...[0m
[0;32mI (468) psram: This chip is ESP32-D0WD[0m
[0;32mI (468) spiram: Found 64MBit SPI RAM device[0m
[0;32mI (468) spiram: SPI RAM mode: flash 80m sram 80m[0m
[0;32mI (471) spiram: PSRAM initialized, cache is in low/high (2-core) mode.[0m
[0;32mI (479) cpu_start: Pro cpu up.[0m
[0;32mI (482) cpu_start: Starting app cpu, entry point is 0x4008170c[0m
[0;32mI (0) cpu_start: App cpu up.[0m
[0;32mI (988) spiram: SPI SRAM memory test OK[0m
[0;32mI (996) cpu_start: Pro cpu start user code[0m
[0;32mI (996) cpu_start: cpu freq: 240000000[0m
[0;32mI (996) cpu_start: Application information:[0m
[0;32mI (999) cpu_start: Project name:     voip_app[0m
[0;32mI (1004) cpu_start: App version:      v2.3-246-g6546a2d-dirty[0m
[0;32mI (1010) cpu_start: Compile time:     Mar  2 2022 21:04:21[0m
[0;32mI (1016) cpu_start: ELF file SHA256:  033166032a2d3cf5...[0m
[0;32mI (1023) cpu_start: ESP-IDF:          v4.4-157-gc7b9f94-dirty[0m
[0;32mI (1029) heap_init: Initializing. RAM available for dynamic allocation:[0m
[0;32mI (1036) heap_init: At 3FFAE6E0 len 00001920 (6 KiB): DRAM[0m
[0;32mI (1042) heap_init: At 3FFB5AB0 len 0002A550 (169 KiB): DRAM[0m
[0;32mI (1049) heap_init: At 3FFE0440 len 00003AE0 (14 KiB): D/IRAM[0m
[0;32mI (1055) heap_init: At 3FFE4350 len 0001BCB0 (111 KiB): D/IRAM[0m
[0;32mI (1062) heap_init: At 40094D20 len 0000B2E0 (44 KiB): IRAM[0m
[0;32mI (1068) spiram: Adding pool of 4085K of external SPI memory to heap allocator[0m
[0;32mI (1076) spi_flash: detected chip: generic[0m
[0;32mI (1081) spi_flash: flash io: qio[0m
[0;32mI (1086) cpu_start: Starting scheduler on PRO CPU.[0m
[0;32mI (0) cpu_start: Starting scheduler on APP CPU.[0m
[0;32mI (1095) spiram: Reserving pool of 18K of internal memory for DMA/internal allocations[0m
[0;32mI (1105) gpio: GPIO[33]| InputEn: 0| OutputEn: 0| OpenDrain: 0| Pullup: 1| Pulldown: 0| Intr:0 [0m
[0;31mE (1115) i2c: i2c_master_cmd_begin(1394): i2c driver not installed[0m
[... 98 lines cut ...]
[0;32mI (45495) AUDIO_PIPELINE: Pipeline started[0m
[0;32mI (45495) AUDIO_ELEMENT: [i2s-0x3f80a004] Element task created[0m
[0;32mI (45505) AUDIO_THREAD: The filter_r task allocate stack on external memory[0m
[0;32mI (45515) AUDIO_ELEMENT: [filter_r-0x3f80a130] Element task created[0m
[0;32mI (45515) AUDIO_ELEMENT: [algo-0x3f80a280] Element task created[0m
[0;32mI (45525) AUDIO_THREAD: The filter task allocate stack on external memory[0m
[0;32mI (45535) AUDIO_ELEMENT: [filter-0x3f80b848] Element task created[0m
[0;32mI (45545) AUDIO_THREAD: The sip_enc task allocate stack on external memory[0m
[0;32mI (45545) AUDIO_ELEMENT: [sip_enc-0x3f80b978] Element task created[0m
[0;32mI (45555) AUDIO_ELEMENT: [raw-0x3f80baa8] Element task created[0m
[0;32mI (45565) AUDIO_PIPELINE: Func:audio_pipeline_run, Line:359, MEM Total:4231335 Bytes, Inter:251971 Bytes, Dram:208331 Bytes

[0m
[0;32mI (45575) AUDIO_ELEMENT: [i2s] AEL_MSG_CMD_RESUME,state:1[0m
[0;32mI (45575) AUDIO_ELEMENT: [filter_r] AEL_MSG_CMD_RESUME,state:1[0m
[0;32mI (45585) RSP_FILTER: sample rate of source data : 16000, channel of source data : 2, sample rate of destination data : 16000, channel of destination data : 1[0m
[0;32mI (45595) AUDIO_ELEMENT: [algo] AEL_MSG_CMD_RESUME,state:1[0m
[0;32mI (45625) AUDIO_ELEMENT: [filter] AEL_MSG_CMD_RESUME,state:1[0m
[0;32mI (45635) RSP_FILTER: sample rate of source data : 16000, channel of source data : 1, sample rate of destination data : 8000, channel of destination data : 1[0m
[0;32mI (45635) AUDIO_ELEMENT: [sip_enc] AEL_MSG_CMD_RESUME,state:1[0m
[0;32mI (45645) AUDIO_PIPELINE: Pipeline started[0m
[0;32mI (45655) SIP_RTP: receive task, local addr=255.255.63.179:37444[0m
[0;32mI (45655) SIP_RTP: send task, remote addr=192.168.2.218
:10098[0m
Initial ONE-MIC auido front-end for speech recognition, mode:1, (Dec 10 2021 20:29:24)

ahhfzhang, 2022-03-04:
Hi @Bruce213 

1. You should use `esp_sip_uas_answer(sip, true)` to answer the call.
2. The local IP address returned to sip via `_get_network_ip` is wrong.

thanks.

davidHuW, 2022-03-04:
> esp_sip_uac_invite(sip,true)

Hi @ahhfzhang 
sorry,i made a mistake about the title and description:
i use esp_sip_uas_answer(sip,true) to accept the call request。
after the acceptance，i can't hung up the session，
just stop at the position：Initial ONE-MIC auido front-end for speech recognition, mode:1, (Dec 10 2021 20:29:24)

and _get_network_ip()  is called by sip session? i don't find the EVENT log：SIP_EVENT_REQUEST_NETWORK_IP.

thanks.

Bruce213, 2022-03-05:
> Hi @Bruce213
> 
> 1. You should use `esp_sip_uas_answer(sip, true)` to answer the call.
> 2. The local IP address returned to sip via `_get_network_ip` is wrong.
> 
> thanks.

Hi @ahhfzhang
sorry,i made a mistake about the title and description:
i use esp_sip_uas_answer(sip,true) to accept the call request。
after the acceptance，i can't hung up the session，
just stop at the position：Initial ONE-MIC auido front-end for speech recognition, mode:1, (Dec 10 2021 20:29:24)

if i comment out the pipeline_open  and pipeline_run (recorder && player) , the sip session will be OK .  

and _get_network_ip() is called by sip session? i don't find the EVENT log：SIP_EVENT_REQUEST_NETWORK_IP.

thanks.

Bruce213, 2022-03-08:
> 

Hi @ahhfzhang 

any feedback pls?

ahhfzhang, 2022-03-08:
Hi @Bruce213

Let's check the data path first, you can change ESP_LOGD(TAG, "SIP_EVENT_REQUEST_NETWORK_IP") to ESP_LOGI .. 
and debug what the IP address is.

`SIP_RTP: receive task, local addr=255.255.63.179:37444`
I think the IP address returned here is wrong.

thanks.

Bruce213, 2022-03-08:
> Hi @Bruce213
> 
> Let's check the data path first, you can change ESP_LOGD(TAG, "SIP_EVENT_REQUEST_NETWORK_IP") to ESP_LOGI .. and debug what the IP address is.
> 
> `SIP_RTP: receive task, local addr=255.255.63.179:37444` I think the IP address returned here is wrong.
> 
> thanks.

Hi @ahhfzhang 
i think IP address has no effect , i make correct IP address for RTP,but the result as same as before :
after my acceptance of the call request , i can't hung up the voip session (and my board can not hear any voice from the voip phone) ,  log stop at the position as below:

pls help me，thanks advance.

【LOG】
\0x1b[0;32mI (10) boot: ESP-IDF v4.4-157-gc7b9f94-dirty 2nd stage bootloader\0x1b[0m
\0x1b[0;32mI (10) boot: compile time 21:03:12\0x1b[0m
\0x1b[0;32mI (11) boot: chip revision: 3\0x1b[0m
\0x1b[0;32mI (14) boot_comm: chip revision: 3, min. bootloader chip revision: 0\0x1b[0m
\0x1b[0;32mI (21) qio_mode: Enabling default flash chip QIO\0x1b[0m
\0x1b[0;32mI (26) boot.esp32: SPI Speed      : 80MHz\0x1b[0m
\0x1b[0;32mI (31) boot.esp32: SPI Mode       : QIO\0x1b[0m
\0x1b[0;32mI (35) boot.esp32: SPI Flash Size : 8MB\0x1b[0m
\0x1b[0;32mI (40) boot: Enabling RNG early entropy source...\0x1b[0m
\0x1b[0;32mI (45) boot: Partition Table:\0x1b[0m
\0x1b[0;32mI (49) boot: ## Label            Usage          Type ST Offset   Length\0x1b[0m
\0x1b[0;32mI (56) boot:  0 nvs              WiFi data        01 02 00009000 00006000\0x1b[0m
\0x1b[0;32mI (63) boot:  1 phy_init         RF data          01 01 0000f000 00001000\0x1b[0m
\0x1b[0;32mI (71) boot:  2 factory          factory app      00 00 00010000 00200000\0x1b[0m
\0x1b[0;32mI (78) boot:  3 flash_tone       Unknown data     01 ff 00210000 0007d000\0x1b[0m
\0x1b[0;32mI (86) boot: End of partition table\0x1b[0m
\0x1b[0;32mI (90) boot_comm: chip revision: 3, min. application chip revision: 0\0x1b[0m
\0x1b[0;32mI (97) esp_image: segment 0: paddr=00010020 vaddr=3f400020 size=4d788h (317320) map\0x1b[0m
\0x1b[0;32mI (193) esp_image: segment 1: paddr=0005d7b0 vaddr=3ffb0000 size=02868h ( 10344) load\0x1b[0m
\0x1b[0;32mI (196) esp_image: segment 2: paddr=00060020 vaddr=400d0020 size=bd33ch (774972) map\0x1b[0m
\0x1b[0;32mI (412) esp_image: segment 3: paddr=0011d364 vaddr=3ffb2868 size=00c68h (  3176) load\0x1b[0m
\0x1b[0;32mI (414) esp_image: segment 4: paddr=0011dfd4 vaddr=40080000 size=14d20h ( 85280) load\0x1b[0m
\0x1b[0;32mI (446) esp_image: segment 5: paddr=00132cfc vaddr=50000000 size=00010h (    16) load\0x1b[0m
\0x1b[0;32mI (456) boot: Loaded app from partition at offset 0x10000\0x1b[0m
\0x1b[0;32mI (456) boot: Disabling RNG early entropy source...\0x1b[0m
\0x1b[0;32mI (468) psram: This chip is ESP32-D0WD\0x1b[0m
\0x1b[0;32mI (468) spiram: Found 64MBit SPI RAM device\0x1b[0m
\0x1b[0;32mI (468) spiram: SPI RAM mode: flash 80m sram 80m\0x1b[0m
\0x1b[0;32mI (471) spiram: PSRAM initialized, cache is in low/high (2-core) mode.\0x1b[0m
\0x1b[0;32mI (479) cpu_start: Pro cpu up.\0x1b[0m
\0x1b[0;32mI (482) cpu_start: Starting app cpu, entry point is 0x4008170c\0x1b[0m
\0x1b[0;32mI (0) cpu_start: App cpu up.\0x1b[0m
\0x1b[0;32mI (988) spiram: SPI SRAM memory test OK\0x1b[0m
\0x1b[0;32mI (996) cpu_start: Pro cpu start user code\0x1b[0m
\0x1b[0;32mI (996) cpu_start: cpu freq: 240000000\0x1b[0m
\0x1b[0;32mI (996) cpu_start: Application information:\0x1b[0m
\0x1b[0;32mI (999) cpu_start: Project name:     voip_app\0x1b[0m
\0x1b[0;32mI (1004) cpu_start: App version:      v2.3-246-g6546a2d-dirty\0x1b[0m
\0x1b[0;32mI (1010) cpu_start: Compile time:     Mar  2 2022 21:04:21\0x1b[0m
\0x1b[0;32mI (1016) cpu_start: ELF file SHA256:  83e901f248db6e4b...\0x1b[0m
\0x1b[0;32mI (1023) cpu_start: ESP-IDF:          v4.4-157-gc7b9f94-dirty\0x1b[0m
\0x1b[0;32mI (1029) heap_init: Initializing. RAM available for dynamic allocation:\0x1b[0m
\0x1b[0;32mI (1036) heap_init: At 3FFAE6E0 len 00001920 (6 KiB): DRAM\0x1b[0m
\0x1b[0;32mI (1042) heap_init: At 3FFB5AB0 len 0002A550 (169 KiB): DRAM\0x1b[0m
\0x1b[0;32mI (1049) heap_init: At 3FFE0440 len 00003AE0 (14 KiB): D/IRAM\0x1b[0m
\0x1b[0;32mI (1055) heap_init: At 3FFE4350 len 0001BCB0 (111 KiB): D/IRAM\0x1b[0m
\0x1b[0;32mI (1062) heap_init: At 40094D20 len 0000B2E0 (44 KiB): IRAM\0x1b[0m
\0x1b[0;32mI (1068) spiram: Adding pool of 4085K of external SPI memory to heap allocator\0x1b[0m
\0x1b[0;32mI (1076) spi_flash: detected chip: generic\0x1b[0m
\0x1b[0;32mI (1081) spi_flash: flash io: qio\0x1b[0m
\0x1b[0;32mI (1086) cpu_start: Starting scheduler on PRO CPU.\0x1b[0m
\0x1b[0;32mI (0) cpu_start: Starting scheduler on APP CPU.\0x1b[0m
\0x1b[0;32mI (1095) spiram: Reserving pool of 18K of internal memory for DMA/internal allocations\0x1b[0m
\0x1b[0;32mI (1105) gpio: GPIO[33]| InputEn: 0| OutputEn: 0| OpenDrain: 0| Pullup: 1| Pulldown: 0| Intr:0 \0x1b[0m
\0x1b[0;31mE (1115) i2c: i2c_master_cmd_begin(1394): i2c driver not installed\0x1b[0m
[... 69 lines cut ...]
\0x1b[0;32mI (48615) AUDIO_ELEMENT: [filter-0x3f80e50c] Element task created\0x1b[0m
\0x1b[0;32mI (48615) AUDIO_ELEMENT: [i2s-0x3f80e6a0] Element task created\0x1b[0m
\0x1b[0;32mI (48625) AUDIO_PIPELINE: Func:audio_pipeline_run, Line:359, MEM Total:4282379 Bytes, Inter:264027 Bytes, Dram:220387 Bytes

\0x1b[0m
\0x1b[0;32mI (48635) AUDIO_ELEMENT: [sip_dec] AEL_MSG_CMD_RESUME,state:1\0x1b[0m
\0x1b[0;32mI (48645) CODEC_ELEMENT_HELPER: The element is 0x3f80e39c. The reserve data 2 is 0x0.\0x1b[0m
\0x1b[0;32mI (48655) AUDIO_ELEMENT: [filter] AEL_MSG_CMD_RESUME,state:1\0x1b[0m
[... cut ...]
\0x1b[0;32mI (48715) AUDIO_PIPELINE: Func:audio_pipeline_run, Line:359, MEM Total:4268639 Bytes, Inter:258879 Bytes, Dram:215239 Bytes

\0x1b[0m
\0x1b[0;32mI (48725) AUDIO_ELEMENT: [i2s] AEL_MSG_CMD_RESUME,state:1\0x1b[0m
\0x1b[0;32mI (48735) AUDIO_ELEMENT: [sip_enc] AEL_MSG_CMD_RESUME,state:1\0x1b[0m
\0x1b[0;32mI (48745) AUDIO_PIPELINE: Pipeline started\0x1b[0m
\0x1b[0;32mI (48745) SIP_RTP: receive task, local addr=192.168.2.9:39444\0x1b[0m
\0x1b[0;32mI (48755) SIP_RTP: send task, remote addr=192.168.2.218
:10110\0x1b[0m

ahhfzhang, 2022-03-09:
Hi @Bruce213 

1. Grab network packets to see if there is data.
2. Remove the algo stream from the pipeline and see if there is any sound coming out.

Bruce213, 2022-03-10:
> Hi @Bruce213
> 
> 1. Grab network packets to see if there is data.
> 2. Remove the algo stream from the pipeline and see if there is any sound coming out.

Hi @ahhfzhang 
1.i grad network packets by wireshark , no RTP packages sent or received.
2.after calling esp_sip_uas_answer to accept the incoming invite session , my voip app is stocked and has no response anymore.

hope you can give me some help , thanks.

Bruce213, 2022-03-12:
> Hi @ahhfzhang 1.i grad network packets by wireshark , no RTP packages sent or received. 2.after calling esp_sip_uas_answer to accept the incoming invite session , my voip app is stocked and has no [...]

Hi @ahhfzhang 
i remove the algo stream from the pipeline ，and the session is still blocked at the RTP position 。
if i only open && run  player pipeline，it will reboot because some memory err？
if i only open && run recorder pipeline ， it will block as the same before.

hope you can give me some suggestions for solving the problem. thank you.

Bruce213, 2022-03-14:
Hi @ahhfzhang
any feedback pls?

Bruce213, 2022-03-17:
Hi @HengYongChao 
can you give me some help?
thanks.

Bruce213, 2022-03-26:
Hi @ahhfzhang @HengYongChao :
any feedback pls?

jason-mao, 2022-08-08:
@Bruce213 So appreciate any progress update from your side.

closed 2023-01-19 by espressif-bot