espressif/esp-adf #778: LyraT mini V1.2 corrupted wav files when running example

rylox, 2022-03-04:
Referring to this example [pipeline_recording_to_sdcard](https://github.com/espressif/esp-adf/tree/master/examples/recorder/pipeline_recording_to_sdcard.) I ran it as is with the default WAV config and configured the board to the LyraT mini. I read a blog with an official answer from the espressif devs confirming that the lyraT mini v1.1 and v1.2 are code compatible so that shouldn't be the problem. 

The recorded wav file sounds corrupted, it randomly varies in lenght per recording and somehow "compresses" the audio if it records it at all. By compress I mean that it's at about 4x speed. I modified the code and specifically configured the codec chip and i2s stream to the same bit and sample rate yet without any success.

The output log doesn't contain unusual erros or warnings.

HengYongChao, 2022-03-08:
Hi @rylox ,

I am not sure what idf version you use, but I use IDF release v4.4 and ADF master branch test the pipeline_recordering_to_sdcard example, it works for me.

I haven't listened to your recording files. But from your description, I guess it may be that the write data of the sdcard seems to have lost some data.

The correct recording file waveform is as follows:

[image]

[recording_to_sdcard.log](https://github.com/espressif/esp-adf/files/8205904/recording_to_sdcard.log)

My suggestion is: 
1. align the compilation environment I tested(ADF version,IDF version), 
2. replace another SDcard and test again please.

jason-mao, 2022-08-08:
@rylox So appreciate any progress update from your side.

closed 2023-09-05 by jason-mao