espressif/esp-adf #781: idf v4.2.2 adf

CAIJUN111, 2022-03-09:
[image]
[image]
hi/ everyone! When I run voip routines using IDF v4.2.2, patching failed. The patch I used is  idf_v4.4_freertos.patch.
I saw that  IDF v4.2.2 can patch successfully on the forum,but  I did fail .
I would be grateful if anyone could help me.

_Administrator@USER-20211223CY MINGW64 /f/CJ/Espressif/frameworks/esp-idf-v4.2.2 ((v4.2.2))
$ git apply ./idf_v4.4_freertos.patch
error: patch failed: components/freertos/include/freertos/task.h:701
error: components/freertos/include/freertos/task.h: patch does not apply
error: patch failed: components/freertos/tasks.c:812
error: components/freertos/tasks.c: patch does not apply_

HengYongChao, 2022-03-10:
Hi @CAIJUN111 

The new MR coming soon  to fix the above situation. You can temporarily use the following patches. 

[idf_v4.2_patches.zip](https://github.com/espressif/esp-adf/files/8220000/idf_v4.2_patches.zip)

CAIJUN111, 2022-03-10:
thanks,but I have uninstalled IDF V4. 2.2.Will this patch be available in 4.3.1?

CAIJUN111, 2022-03-10:
Can you give me a patch that can be installed in 4.3.1? Thank you very much

HengYongChao, 2022-03-10:
Hi @CAIJUN111 ,

Try IDF V4.3 patches please, as I mentioned above, there will be an MR commit to the ADF master branch, until then, you can use the patch I sent you currently.

[idf_v4.3_patches.zip](https://github.com/espressif/esp-adf/files/8220881/idf_v4.3_patches.zip)

closed 2022-03-10 by CAIJUN111