espressif/esp-adf #782: voip sample build failed with ESP32-S2-Kaluga-1

achao1104, 2022-03-09:
Hello,

env: idf4.2.2 or idf4.4 and adf master
I see it is green for ESP32-S2-Kaluga-1 dev board to run voip sample, but build failed as below.

Building in: D:\project\esp32\project\voip_esp32s2\build
cmake --build . -- -v
[1/5] cmd.exe /C "cd /D D:\project\esp32\project\voip_esp32s2\build\bootloader && D:\project\esp32\.espressif\tools\cmake\3.16.4\bin\cmake.exe --build ."
ninja: no work to do.
[2/3] cmd.exe /C "cd . && D:\project\esp32\.espressif\tools\xtensa-esp32s2-elf\esp-2020r3-8.4.0\xtensa-esp32s2-elf\bin\xtensa-esp32s2-elf-g++.exe  -mlongcalls   @CMakeFiles\voip_app.elf.rsp  -o voip_app.elf  && cd ."
FAILED: voip_app.elf 
cmd.exe /C "cd . && D:\project\esp32\.espressif\tools\xtensa-esp32s2-elf\esp-2020r3-8.4.0\xtensa-esp32s2-elf\bin\xtensa-esp32s2-elf-g++.exe  -mlongcalls   @CMakeFiles\voip_app.elf.rsp  -o voip_app.elf  && cd ."
d:/project/esp32/.espressif/tools/xtensa-esp32s2-elf/esp-2020r3-8.4.0/xtensa-esp32s2-elf/bin/../lib/gcc/xtensa-esp32s2-elf/8.4.0/../../../../xtensa-esp32s2-elf/bin/ld.exe: esp-idf/main/libmain.a(voip_app.c.obj):(.literal.recorder_pipeline_open+0x58): undefined reference to `algo_stream_init'
d:/project/esp32/.espressif/tools/xtensa-esp32s2-elf/esp-2020r3-8.4.0/xtensa-esp32s2-elf/bin/../lib/gcc/xtensa-esp32s2-elf/8.4.0/../../../../xtensa-esp32s2-elf/bin/ld.exe: esp-idf/main/libmain.a(voip_app.c.obj):(.literal._sip_event_handler+0x50): undefined reference to `algo_stream_set_delay'
d:/project/esp32/.espressif/tools/xtensa-esp32s2-elf/esp-2020r3-8.4.0/xtensa-esp32s2-elf/bin/../lib/gcc/xtensa-esp32s2-elf/8.4.0/../../../../xtensa-esp32s2-elf/bin/ld.exe: esp-idf/main/libmain.a(voip_app.c.obj): in function `recorder_pipeline_open':
D:\project\esp32\project\voip_esp32s2\build/../main/voip_app.c:106: undefined reference to `algo_stream_init'
d:/project/esp32/.espressif/tools/xtensa-esp32s2-elf/esp-2020r3-8.4.0/xtensa-esp32s2-elf/bin/../lib/gcc/xtensa-esp32s2-elf/8.4.0/../../../../xtensa-esp32s2-elf/bin/ld.exe: esp-idf/main/libmain.a(voip_app.c.obj): in function `player_pipeline_open':
D:\project\esp32\project\voip_esp32s2\build/../main/voip_app.c:211: undefined reference to `algo_stream_set_delay'
d:/project/esp32/.espressif/tools/xtensa-esp32s2-elf/esp-2020r3-8.4.0/xtensa-esp32s2-elf/bin/../lib/gcc/xtensa-esp32s2-elf/8.4.0/../../../../xtensa-esp32s2-elf/bin/ld.exe: D:\project\esp32\project\voip_esp32s2\build/../main/voip_app.c:213: undefined reference to `algo_stream_set_delay'
collect2.exe: error: ld returned 1 exit status
ninja: build stopped: subcommand failed.
Build complete (7 errors, 0 warnings): D:\project\esp32\project\voip_esp32s2\build
Total time taken to build the project: 12,712 ms

HengYongChao, 2022-03-10:
Hi @achao1104 

ESP32-S2-Kaluga-1 do not supported by esp-sr submodule, Please try use it with esp32 or esp32-s3 board.

closed 2022-03-14 by espressif-bot

CAIJUN111, 2022-03-26:
> Hi @achao1104
> 
> ESP32-S2-Kaluga-1 do not supported by esp-sr submodule, Please try use it with esp32 or esp32-s3 board.
> 
> https://github.com/espressif/esp-adf/blob/master/components/audio_stream/CMakeLists.txt#L15
> 
> ```c
> if((${IDF_TARGET} STREQUAL "esp32") OR (${IDF_TARGET} STREQUAL "esp32s3"))
>     list(APPEND COMPONENT_SRCS "algorithm_stream.c" "tts_stream.c")
>     list(APPEND COMPONENT_REQUIRES esp-sr)
> endif()
> ```

@HengYongChao  I try to change the chip to esp32s3,but it can not compile successfully.
[image]
what is wrong with it?

HengYongChao, 2022-04-01:
Hi @CAIJUN111 

I see that you are using the related plugins of vscode, I have not used similar tools, I suggest you use the native linux terminal environment to compile the above routines.

In addition, I see that the error reported in the compilation report is that there is no relevant GPIO definition. Please check whether you use `idf.py set-target esp32s3` to select the chip when compiling. And you also need to select the correct  board in `menuconfig` to compile successfully.