espressif/esp-adf #783: Execute the VOIP example,the client always offline in about three minutes

CAIJUN111, 2022-03-14:
[image]

100 is the phone number for ESP32lyrat. When the ESP32 program runs, the client will be online for more than three minutes, and then the server will show that the call is dropped and you can make a call, but not be called  
There are two clients to call ESP32, 101 is another computer, 102 is ios app 22CallSIP  
The server is miniSIPServer V39  
The ESP32 model is esp32LYrat V4.3  
If anyone can help me, I'd really appreciate it.

The code run record:

> `ts Jul 29 2019 12:21:46

rst:0x1 (POWERON_RESET),boot:0x1f (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0030,len:7316
load:0x40078000,len:15468
load:0x40080400,len:4472
0x40080400: _init at ??:?

entry 0x400806a0
I (27) boot: ESP-IDF v4.3.1-dirty 2nd stage bootloader
I (27) boot: compile time 18:38:24
I (27) boot: chip revision: 3
I (31) boot_comm: chip revision: 3, min. bootloader chip revision: 0
I (38) qio_mode: Enabling default flash chip QIO
I (43) boot.esp32: SPI Speed      : 80MHz
I (48) boot.esp32: SPI Mode       : QIO
I (52) boot.esp32: SPI Flash Size : 8MB
I (57) boot: Enabling RNG early entropy source...
I (62) boot: Partition Table:
I (66) boot: ## Label            Usage          Type ST Offset   Length
I (73) boot:  0 nvs              WiFi data        01 02 00009000 00006000
I (80) boot:  1 phy_init         RF data          01 01 0000f000 00001000
I (88) boot:  2 factory          factory app      00 00 00010000 00200000
I (95) boot:  3 flash_tone       Unknown data     01 ff 00210000 0007d000
I (103) boot: End of partition table
I (107) boot_comm: chip revision: 3, min. application chip revision: 0
I (114) esp_image: segment 0: paddr=00010020 vaddr=3f400020 size=51dech (335340) map
I (222) esp_image: segment 1: paddr=00061e14 vaddr=3ffb0000 size=04470h ( 17520) load
I (228) esp_image: segment 2: paddr=0006628c vaddr=40080000 size=09d8ch ( 40332) load
I (242) esp_image: segment 3: paddr=00070020 vaddr=400d0020 size=f52e8h (1004264) map
I (540) esp_image: segment 4: paddr=00165310 vaddr=40089d8c size=0c5c4h ( 50628) load
I (558) esp_image: segment 5: paddr=001718dc vaddr=50000000 size=00010h (    16) load
I (569) boot: Loaded app from partition at offset 0x10000
I (570) boot: Disabling RNG early entropy source...
I (581) psram: This chip is ESP32-D0WD
I (581) spiram: Found 64MBit SPI RAM device
I (581) spiram: SPI RAM mode: flash 80m sram 80m
I (584) spiram: PSRAM initialized, cache is in low/high (2-core) mode.
I (592) cpu_start: Pro cpu up.
I (595) cpu_start: Starting app cpu, entry point is 0x40081814
0x40081814: call_start_cpu1 at D:/A/esp-idf-v4.3.1/components/esp_system/port/cpu_start.c:141

I (0) cpu_start: App cpu up.
I (1101) spiram: SPI SRAM memory test OK
I (1109) cpu_start: Pro cpu start user code
I (1109) cpu_start: cpu freq: 240000000
I (1109) cpu_start: Application information:
I (1112) cpu_start: Project name:     cj_voip2
I (1117) cpu_start: App version:      1
I (1122) cpu_start: Compile time:     Mar 11 2022 15:12:50
I (1128) cpu_start: ELF file SHA256:  1d637086b09f599f...
I (1134) cpu_start: ESP-IDF:          v4.3.1-dirty
I (1140) heap_init: Initializing. RAM available for dynamic allocation:
I (1147) heap_init: At 3FFAE6E0 len 00001920 (6 KiB): DRAM
I (1153) heap_init: At 3FFB7068 len 00028F98 (163 KiB): DRAM
I (1159) heap_init: At 3FFE0440 len 00003AE0 (14 KiB): D/IRAM
I (1166) heap_init: At 3FFE4350 len 0001BCB0 (111 KiB): D/IRAM
[... 394 lines cut ...]
Max-Forwards: 70
Call-ID: 0F2689426BD1F2E9234E9025CABD38BB947CB61669AA
CSeq: 8 OPTIONS
Expires: 3600
User-Agent: ESP32 SIP/2.0
Content-Length: 0
Allow: INVITE, ACK, CANCEL, BYE, UPDATE, REFER, MESSAGE, OPTIONS, INFO, SUBSCRIBE
Supported: replaces, norefersub, extended-refer, timer, X-cisco-serviceuri
Allow-Events: presence, kpml
Authorization: Digest username="100", realm="myvoipapp.com", nonce="2FD691ACC3BFA0B10107841AAF861571", uri="sip:192.168.1.175:5060", response="b5d8d7e33204abbaba73b9fb74cff2a2", algorithm=MD5, nc=00000007, cnonce="107244a1d67001d6"

I (200785) SIP: [1970-01-01/00:03:09]=======================>>
I (200795) SIP: [1970-01-01/00:03:09]<<=====READ 0403 bytes==
I (200795) SIP: 

SIP/2.0 200 OK
Via: SIP/2.0/UDP 192.168.1.108:10357;branch=z9hG4bK-191880252;received=192.168.1.108;rport=10357
From: <sip:100@192.168.1.175:5060>;tag=-1829764343
To: <sip:100@192.168.1.175:5060>;tag=d656d86877462d97
CSeq: 8 OPTIONS
Call-ID: 0F2689426BD1F2E9234E9025CABD38BB947CB61669AA
Allow: ACK,BYE,CANCEL,INFO,INVITE,MESSAGE,NOTIFY,OPTIONS,PRACK,REFER,REGISTER,SUBSCRIBE
Content-Length: 0

I (200835) SIP: [1970-01-01/00:03:09]<<======================`

CAIJUN111, 2022-03-14:
I have one more question,could you recommend one of the audio analysis software you mentioned in Readme.md?
[image]

CAIJUN111, 2022-03-16:
Who can help me?

CAIJUN111, 2022-03-16:
I find that the PC client is always online

CAIJUN111, 2022-03-17:
[image]
I check the sip status every minute.
When the client goes offline, the function indicates that SIP is  online

ahhfzhang, 2022-03-18:
Hi @CAIJUN111 

Yes, the client is actually online all the time, the reason why this server thinks that the client is offline is that the registration interval set by the server may be too short. The registration interval set by this server is 120 seconds, and the commonly used servers are generally set for 1800 or 3600, so you can change the server registration interval, then try again.

btw, we will add a configuration later, so that the client can also configure this parameter.

thanks.

CAIJUN111, 2022-03-26:
> Hi @CAIJUN111
>
> Yes, the client is actually online all the time, the reason why this server thinks that the client is offline is that the registration interval set by the server may be too short. The [...]

You are right!I change the the registration interval.Client will not offline.Thanks!

closed 2022-03-26 by CAIJUN111

pierluigizagaria, 2022-06-14:
How did you change the registration interval? I'm facing the same issue!