espressif/esp-adf #784: Memory alloc failed with algorithm-stream

quickshat, 2022-03-14:
Hi,

i've been trying to run the **pipeline_a2dp_sink_with_hfp** example.
The basic example is working on the MSC2.2 LyraT board.

Additional specifications require an algo-stream with AEC on the encoder pipeline, which i added according to the demo applications from this repo.

I've also tested this setup with external PSRAM enabled / disabled. In both cases there's plenty of heap left to allocate.
Right before the call of the pipeline_run, there's something around 10kB heap available (already without PSRAM).

The audio-element tasks are created and as soon as the init-step for the algo-element begins to execute, some random allocation fail aborts the application.

The pipeline looks like this:

`I2S-Reader (44100-2) --> Resample Filter (16000-1) --> AlgoStream --> Raw`
           

Algo-stream uses default type-1 config with 16000Hz and single channel.

The code **executes without any issues**, as long the **algo-stream** element **isn't linked** - even init and register don't cause a core dump.

Last debug print **always** is: `AUDIO_ELEMENT: [algo] AEL_MSG_CMD_RESUME,state:1`

Did you have some ideas what might cause this problem ?

HengYongChao, 2022-03-18:
Hi @quickshat 

I recommend you to use module with psram( ESP32-WROVER ), I see that you only have 10k of ram left, it should not be enough for AlgoStream + bt application.

quickshat, 2022-03-18:
Hi @HengYongChao ,

i enabled PSRAM. Are any specific options necessary to make the PSRAM work correctly ?

quickshat, 2022-03-18:
Hi @HengYongChao ,

i enabled PSRAM but it makes no difference. Are any specific options necessary to make the PSRAM work correctly ?

quickshat, 2022-03-21:
In the meantime the Algo-Stream component does not cause crashes anymore but makes the microphone pipeline unusable. 
The Algo stream runs on a different core than the bluetooth hfp client and as soon as a voice call is active, the following output gets thrown:
`BT_BTM: SCO xmit Q overflow, pkt dropped`

I've tried to modify any performance values of the algo task, but no change at all.
When increasing the resamplers out_buffer size, i am able to delay the issue but no prevent. Also no sound is transmitted. Seems like it won't pass the algo stream.

HengYongChao, 2022-03-25:
Hi @quickshat 

1. Are you working on using algorithm stream to process HFP data for Bluetooth?

> BT_BTM: SCO xmit Q overflow, pkt dropped

2. This print is a reminder that the Bluetooth data packet is lost. Could it be that the system load is not balanced so that the Bluetooth cannot receive the packet? You may need to check your system's cpu load

3. Algorithm flow has different processing algorithms, NS, AGC, AEC, please note that their usage conditions are different.
You can refer to the routine [algorithm](https://github.com/espressif/esp-adf/tree/master/examples/advanced_examples/algorithm).  Pipeline like : ** hfp stream ---> filter ---> algorithm ---> i2s_stream ---> [speaker] **

5. And your current problem is no longer a memory malloc problem, If so please open a new issue describing your problem.

jason-mao, 2022-08-08:
@quickshat So appreciate any progress from your issue.

closed 2023-09-05 by jason-mao