espressif/esp-adf #788: OGG stop and resume issue.

iloop2020, 2022-03-20:
Hi ,

We are using MP3 decoder for audio playback, and it works fine.
And we are going to switch to OGG decoder, and it also works fine simple playback.

Issue:
We are facing issue to "Stop" and "Resume" ogg audio

1. When stop, there are some error message:
```
W (49112) VORBIS_DECODER: error  output -3, 4096
I (49112) VORBIS_DECODER: Closed
```

2. When resume, there are some error message:
```
I (50822) VORBIS_DECODER: VORBIS codec
I (50822) VORBIS_DECODER: the song resume
E (50822) VORBIS_DECODER: The reserve data is null. line 349
E (50832) AUDIO_ELEMENT: [ogg] AEL_STATUS_ERROR_OPEN,-1
I (50832) VORBIS_DECODER: Closed
```

Please help, Thank you very much.

Leo

HengYongChao, 2022-03-22:
Hi @iloop2020 

I tested the playback of the ogg file format on the cli example. It still works fine, and the action correctly pauses and resumes playback. I suggest you refer to the code operation of the cli.

```c
I (45848) ESP_AUDIO_CTRL: Enter media_ctrl_pause procedure, src:0, st:RUNNING
I (45848) ESP_AUDIO_CTRL: Request_CMD_Queue CMD:3, Available:5, que:0x3ffe2068
I (45868) ESP_AUDIO_TASK: MEDIA_CTRL_EVT_PAUSE
I (45898) AUDIO_FORGE: audio forge closed
I (45898) AUDIO_ELEMENT: [Audio_forge] AEL_MSG_CMD_PAUSE
I (45928) AUDIO_ELEMENT: [OUT_iis] AEL_MSG_CMD_PAUSE
I (45928) ESP_AUDIO_CTRL: Exit media_ctrl_pause procedure, src:0, ret:0
I (45928) ESP_AUDIO_TASK: Recv Element[Audio_forge-0x3f809580] MSG,type:20000,cmd:8,len:4,status:AEL_STATUS_STATE_PAUSEDI (45928) CONSOLE_EXAMPLE: app_audio paused
esp32> I (45928) CONSOLE_EXAMPLE: esp_auido status:2,err:0

I (45948) ESP_AUDIO_TASK: Recv Element[OUT_iis-0x3f809040] MSG,type:20000,cmd:8,len:4,status:AEL_STATUS_STATE_PAUSED
I (51588) ESP_AUDIO_CTRL: Enter media_ctrl_resume procedure, src:0, st:PAUSED
I (51598) ESP_AUDIO_CTRL: Request_CMD_Queue CMD:4, Available:5, que:0x3ffe2068
I (51608) ESP_AUDIO_TASK: MEDIA_CTRL_EVT_RESUME, prev:0
I (51608) AUDIO_ELEMENT: [OUT_iis] AEL_MSG_CMD_RESUME,state:4
I (51618) I2S_STREAM: AUDIO_STREAM_WRITER
I (51618) AUDIO_ELEMENT: [Audio_forge] AEL_MSG_CMD_RESUME,state:4
I (51628) AUDIO_FORGE: audio_forge opened
I (51628) ESP_AUDIO_TASK: Recv Element[IN_file-0x3f80774c] MSG,type:20000,cmd:8,len:4,status:AEL_STATUS_STATE_RUNNING
I (51638) ESP_AUDIO_TASK: Recv Element[DEC_ogg-0x3f808314] MSG,type:20000,cmd:8,len:4,status:AEL_STATUS_STATE_RUNNING
I (51658) ESP_AUDIO_TASK: Recv Element[OUT_iis-0x3f809040] MSG,type:20000,cmd:8,len:4,status:AEL_STATUS_STATE_RUNNING
I (51668) ESP_AUDIO_TASK: Recv Element[Audio_forge-0x3f809580] MSG,type:20000,cmd:8,len:4,status:AEL_STATUS_STATE_RUNNING
I (51678) ESP_AUDIO_TASK: ESP_AUDIO status is AEL_STATUS_STATE_RUNNING, 0, src:0, is_stopping:0
I (51688) ESP_AUDIO_TASK: Func:media_ctrl_task, Line:984, MEM Total:4116191 Bytes, Inter:94103 Bytes, Dram:72003 Bytes
```

iloop2020, 2022-03-23:
@HengYongChao 

1. Which idf/adf version you are using ?
2. What is the meaning the log, that I shared before ?

Thank you.

HengYongChao, 2022-03-25:
Hi @iloop2020 

1. The latest ADF and the IDF release v4.4
2. If you play the ogg audio in the sdcard, then it means that the playing of the ogg is OK. We have to choose other directions to investigate.

iloop2020, 2022-03-26:
Hi @HengYongChao ,

1. We upgrade adf from 2.3 to 2.4, this issue is fixed. is there any bug fix for it between these two versions ?
2. We have another question about multiple audio coder, what we are doing is basically follow the diagram to setup the audio pipline for mp3, now we would like to suppot both MP3 and OGG. 
2.1 Wow can we switch between the 2 codec in the audio pipeline ?
2.2 Or we should create 2 independent audio pipeline ? if so, will it consume too many memory ?
[image]

Thank you.

Johboh, 2022-07-16:
I have the same issue in ESP-IDF v.4.4.1 with ESP-ADF v2.4:

```
I (9041) I2S: DMA Malloc info, datalen=blocksize=1200, dma_buf_count=3
I (9051) AUDIO_PIPELINE: link el->rb, el:0x3fda1b1c, tag:file/http, rb:0x3fdb82a8
I (9061) AUDIO_PIPELINE: link el->rb, el:0x3fdb6c80, tag:decrypt, rb:0x3fdb82e4
I (9071) AUDIO_PIPELINE: link el->rb, el:0x3fdb6d5c, tag:ogg_stripper, rb:0x3fdb8320
I (9081) AUDIO_PIPELINE: link el->rb, el:0x3fdb8104, tag:ogg_decoder, rb:0x3fdb835c
I (9091) AUDIO_ELEMENT: [file/http-0x3fda1b1c] Element task created
I (9101) AUDIO_THREAD: The decrypt task allocate stack on external memory
I (9111) AUDIO_ELEMENT: [decrypt-0x3fdb6c80] Element task created
I (9121) AUDIO_THREAD: The ogg_stripper task allocate stack on external memory
I (9131) AUDIO_ELEMENT: [ogg_stripper-0x3fdb6d5c] Element task created
I (9141) AUDIO_THREAD: The ogg_decoder task allocate stack on external memory
I (9151) AUDIO_ELEMENT: [ogg_decoder-0x3fdb8104] Element task created
I (9161) AUDIO_THREAD: The i2s task allocate stack on external memory
I (9171) AUDIO_ELEMENT: [i2s-0x3fdb50cc] Element task created
I (9181) AUDIO_PIPELINE: Func:audio_pipeline_run, Line:359, MEM Total:1817406 Bytes, Inter:71904 Bytes, Dram:71904 Bytes
I (9211) AUDIO_ELEMENT: [file/http] AEL_MSG_CMD_RESUME,state:1
I (9221) AUDIO_ELEMENT: [decrypt] AEL_MSG_CMD_RESUME,state:1
I (9231) AUDIO_ELEMENT: [ogg_stripper] AEL_MSG_CMD_RESUME,state:1
I (9241) AUDIO_ELEMENT: [ogg_decoder] AEL_MSG_CMD_RESUME,state:1
I (9251) VORBIS_DECODER: VORBIS codec
I (9251) CODEC_ELEMENT_HELPER: The element is 0x3fdb8104. The reserve data 2 is 0x0.
I (9251) AUDIO_ELEMENT: [i2s] AEL_MSG_CMD_RESUME,state:1
I (9271) I2S_STREAM: AUDIO_STREAM_WRITER
I (9261) VORBIS_DECODER: a new song playing
I (9271) AUDIO_PIPELINE: Pipeline started
I (9911) HTTPS_STREAM: total_bytes=4286424
I (10101) AUDIO_ELEMENT: [i2s] AEL_MSG_CMD_PAUSE
I (10101) AUDIO_ELEMENT: [i2s] AEL_MSG_CMD_RESUME,state:4
I (10111) I2S_STREAM: AUDIO_STREAM_WRITER
W (10101) AUDIO_ELEMENT: [i2s-0x3fdb50cc] RESUME timeout

I (53621) MAIN: handle_queue_entry(PlaybackAction::Pause) <------ Pausing
I (53691) AUDIO_ELEMENT: [file/http] AEL_MSG_CMD_PAUSE
I (53891) AUDIO_ELEMENT: [decrypt] AEL_MSG_CMD_PAUSE
I (54031) AUDIO_ELEMENT: [ogg_stripper] AEL_MSG_CMD_PAUSE
I (54111) VORBIS_DECODER: Closed
I (54111) AUDIO_ELEMENT: [ogg_decoder] AEL_MSG_CMD_PAUSE
I (54241) AUDIO_ELEMENT: [i2s] AEL_MSG_CMD_PAUSE

I (59951) MAIN: handle_queue_entry(PlaybackAction::Resume) <------ Resuming
I (59961) AUDIO_ELEMENT: [file/http] AEL_MSG_CMD_RESUME,state:4
I (59971) AUDIO_ELEMENT: [decrypt] AEL_MSG_CMD_RESUME,state:4
I (59981) AUDIO_ELEMENT: [ogg_stripper] AEL_MSG_CMD_RESUME,state:4
I (59991) AUDIO_ELEMENT: [ogg_decoder] AEL_MSG_CMD_RESUME,state:4
I (60001) VORBIS_DECODER: VORBIS codec
I (60001) VORBIS_DECODER: the song resume
I (60091) AUDIO_ELEMENT: [i2s] AEL_MSG_CMD_RESUME,state:4
I (60101) I2S_STREAM: AUDIO_STREAM_WRITER
I (61241) HTTPS_STREAM: total_bytes=4286424
I (61421) VORBIS_DECODER: Closed
I (61511) AUDIO_ELEMENT: IN-[i2s] AEL_IO_DONE,-2
W (61551) MUSIC_STREAMING: [ * ] Stop event received
E (61561) AUDIO_ELEMENT: [ogg_decoder] Element already stopped
E (61571) AUDIO_ELEMENT: [i2s] Element already stopped
W (61571) HTTPS_STREAM: No output due to stopping
W (61581) AUDIO_PIPELINE: There are no listener registered
I (61591) AUDIO_PIPELINE: audio_pipeline_unlinked
W (61591) AUDIO_ELEMENT: [file/http] Element has not create when AUDIO_ELEMENT_TERMINATE
W (61601) AUDIO_ELEMENT: [decrypt] Element has not create when AUDIO_ELEMENT_TERMINATE
W (61611) AUDIO_ELEMENT: [ogg_stripper] Element has not create when AUDIO_ELEMENT_TERMINATE
W (61621) AUDIO_ELEMENT: [ogg_decoder] Element has not create when AUDIO_ELEMENT_TERMINATE
I (61631) CODEC_ELEMENT_HELPER: The element is 0x3fdb8104. The reserve data 2 is 0x0.
W (61641) AUDIO_ELEMENT: [i2s] Element has not create when AUDIO_ELEMENT_TERMINATE
I (61651) I2S: DMA queue destroyed
```

In my case, I'm streaming an encrypted audio file over HTTPS, which I then decode before feeding it to the ogg decoder. Playback works, but pausing and resuming stops playback. There is a short 1-2 second audio on resume, but then it stops.

closed 2022-08-08 by jason-mao