espressif/esp-adf #789: How to use the mono function of I2S？

CAIJUN111, 2022-03-22:
_i2s_stream_cfg_t i2s_cfg = I2S_STREAM_CFG_DEFAULT();
    i2s_cfg.type = AUDIO_STREAM_READER;
    i2s_cfg.uninstall_drv = false;
#ifdef CONFIG_ESP_LYRAT_MINI_V1_1_BOARD
    i2s_cfg.i2s_port = 1;
#endif
    i2s_cfg.task_core = 1;
    i2s_cfg.i2s_config.sample_rate = I2S_SAMPLE_RATE;
    i2s_cfg.i2s_config.channel_format = I2S_CHANNEL_FMT_ONLY_RIGHT;
/*    In theory, mono is just a speaker does it sound.But actually both speakers have sound ,I find it pointless to modify it . */
    i2s_stream_reader = i2s_stream_init(&i2s_cfg);_

When I changed the "channel_format "parameter in I2S_STREAM_CFG_DEFAULT() to I2S_CHANNEL_FMT_ONLY_RIGHT, I noticed that the sound became sluggish .How can I solve this issue?

HengYongChao, 2022-03-25:
Hi @CAIJUN111 

In fact, the I2S stream reader and i2s stream writer are different. 

If you want to implement mono applications, please refer to the lyrat mini v1.2 development board routines.

FYI：https://github.com/espressif/esp-adf/blob/master/examples/recorder/element_cb_sdcard_amr/main/element_cb_sdcard_amr.c#L84

jason-mao, 2022-08-08:
@CAIJUN111 We have fixed some  I2S mono bug on release/v4.4 branch, you could test it

CAIJUN111, 2022-08-09:
> @CAIJUN111 We have fixed some I2S mono bug on release/v4.4 branch, you could test it

thank for your reply, i have already solved it!

closed 2022-08-09 by CAIJUN111