espressif/esp-adf #790: When I ran the routine play_tone_mp3_example , I found the problem.how to determine the prompt sound URI ?

CAIJUN111, 2022-03-23:
Guys, who has used esp32 to burn the prompt sound, how to determine the prompt sound URI ?
```
const char* tone_uri[] = 
{
    "flash://tone/0_Bt_Reconnect.mp3",
    "flash://tone/1_Wechat.mp3",
    "flash://tone/2_Welcome_To_Wifi.mp3",
    "flash://tone/3_New_Version_Available.mp3",
    "flash://tone/4_Bt_Success.mp3",
    "flash://tone/5_Freetalk.mp3",
    "flash://tone/6_Upgrade_Done.mp3",
    "flash://tone/7_shutdown.mp3",
    "flash://tone/8_Alarm.mp3",
    "flash://tone/9_Wifi_Success.mp3",
    "flash://tone/10_Under_Smartconfig.mp3",
    "flash://tone/11_Out_Of_Power.mp3",
    "flash://tone/12_server_connect.mp3",
    "flash://tone/13_hello.mp3",
    "flash://tone/14_new_message.mp3",
    "flash://tone/15_Please_Retry_Wifi.mp3",
    "flash://tone/16_please_setting_wifi.mp3",
    "flash://tone/17_Welcome_To_Bt.mp3",
    "flash://tone/18_Wifi_Time_Out.mp3",
    "flash://tone/19_Wifi_Reconnect.mp3",
    "flash://tone/20_server_disconnect.mp3",
};
```

HengYongChao, 2022-03-25:
Hi @CAIJUN111 

Please check the [pipeline_flash_tone](https://github.com/espressif/esp-adf/tree/master/examples/player/pipeline_flash_tone) example.  code belows:

```c
    ESP_LOGI(TAG, "[2.6] Set up  uri (file as tone_stream, mp3 as mp3 decoder, and default output is i2s)");
    audio_element_set_uri(tone_stream_reader, tone_uri[TONE_TYPE_HELLO]);
```

CAIJUN111, 2022-03-25:
The essence of this function "audio_element_set_uri" is to bootstrap to the URI,
For example, "flash://tone/0_Bt_Reconnect. Mp3 ", how can I confirm this URI from my synthesized .BIN file

HengYongChao, 2022-03-25:
Hi @CAIJUN111 

The answer is in the readme of [pipeline_flash_tone](https://github.com/espressif/esp-adf/blob/master/examples/player/pipeline_flash_tone/README.md) example. 

### Prerequisites

Required source file for the bin file address in flash has already been generated in the `/tools/audio-esp.bin` directory, and required source file for the audio file address in flash in the `/components/audio_flash_tone/` directory.

If you need to generate your own `audio-esp.bin`, you need to first put `mk_audio_bin.py` (located in $ADF_PATH/tools/audio_tone/mk_audio_tone.py) and the user prompt tone file in the same directory. Then, run the `mk_audio_bin.py` script with the command below, which will eventually generate the `audio-esp.bin` file and the `/audio_flash_tone/` folder in the upper directory of the script directory. The source file for the tone address in flash is stored in the `/components/audio_flash_tone/` folder.

The Python script command is as follows:

```
python $ADF_PATH/tools/audio_tone/mk_audio_tone.py tone_mp3_folder
```

The default `audio-esp.bin` of this example contains the following audio files:

```c
   "flash://tone/0_Bt_Reconnect.mp3",
   "flash://tone/1_Wechat.mp3",
   "flash://tone/2_Welcome_To_Wifi.mp3",
   "flash://tone/3_New_Version_Available.mp3",
   "flash://tone/4_Bt_Success.mp3",
   "flash://tone/5_Freetalk.mp3",
   "flash://tone/6_Upgrade_Done.mp3",
   "flash://tone/7_shutdown.mp3",
   "flash://tone/8_Alarm.mp3",
   "flash://tone/9_Wifi_Success.mp3",
   "flash://tone/10_Under_Smartconfig.mp3",
   "flash://tone/11_Out_Of_Power.mp3",
   "flash://tone/12_server_connect.mp3",
   "flash://tone/13_hello.mp3",
   "flash://tone/14_new_message.mp3",
   "flash://tone/15_Please_Retry_Wifi.mp3",
   "flash://tone/16_please_setting_wifi.mp3",
   "flash://tone/17_Welcome_To_Bt.mp3",
   "flash://tone/18_Wifi_Time_Out.mp3",
   "flash://tone/19_Wifi_Reconnect.mp3",
   "flash://tone/20_server_disconnect.mp3",
```

CAIJUN111, 2022-03-25:
Yes,I have already generated my .bin file,but I don't know how to define my uri like  "flash://tone/0_Bt_Reconnect.mp3".

jason-mao, 2022-08-08:
Hi @CAIJUN111, did you resolved your problem?

closed 2023-09-05 by jason-mao