espressif/esp-adf #791: Can't register to to Twilio SIP server.

charon22, 2022-03-23:
## Environment
Development Kit: ESP32-LyraT V4.3
Chip used: ESP32-WROVER-B
IDF version: v4.2.2-dirty
Build System: idf.py
Operating System: Windows 10
Using an IDE?: Power shell
Power Supply: USB / 5v

## Problem Description
I am going to use the Twilio SIP server to provide a VOIP service using ESP32 that was intended to be serviced using an existing landline phone. For this purpose, we are going to use the Twilio SIP server. When connecting to the Twilio SIP server, it is not registered.

## Question
I made the Twilio account free and created a test account and tested it.
No connection.
So, after sending the ESP log to Twilio, I received the following answer as a result of inquiring.
Please analyze this.
## ESP32 LOG)
[teraterm_COM6_20220309_153824.log.TXT](https://github.com/espressif/esp-adf/files/8335557/teraterm_COM6_20220309_153824.log.TXT)

## Twilio Response)
I am seeing multiple problems with the ESP32. Firstly, the ESP32 has a Request-URI which includes the user portion:

REGISTER sip:test@epicsafety.sip.twilio.com:5060 SIP/2.0

However, as per RFC 3261 section 10.2: Request-URI: The Request-URI names the domain of the location service for which the registration is meant (for example, "sip:chicago.com"). The "userinfo" and "@" components of the SIP URI MUST NOT be present.

This means that the ESP32 should be sending REGISTER sip:epicsafety.sip.twilio.com SIP/2.0 <-- this is from a working registration.

I also noticed that the digest response is incorrect from the ESP32.
This means that the ESP32 is doing something wrong when calculating the digest.
Here is a quick summary of the calculation:

HA1 = MD5(test:sip.twilio.com:Aaabbbcccdd1)
so, HA1 = 79b120bd899daf3d07c16be8e8c21d5a

HA2 = MD5(REGISTER:sip:epicsafety.sip.twilio.com)
so, HA2 = be16169f6b30b458e9534e741fd08174

Response = MD5(HA1:server_nonce:nc:cnonce:qop:HA2)
In other words, Response = MD5(79b120bd899daf3d07c16be8e8c21d5a:0WjLsbF6k7Z-ogcfJMIpsBxRoxu-AHBPPL31CQpy4MJ1lQal:00000001:9936de1b03473cf8:auth:be16169f6b30b458e9534e741fd08174)
In the above, server_nounce is the nounce that Twilio sent and cnounce is the nounce that your client generated
so, Response = fb3fd3f656959f84e7ca1e71948b0284, however, you ESP32 sent 7f2b20215d0c0a51eafeb3a9bf729e1f.

I don't know why your ESP32 is not calculating the correct digest response

ahhfzhang, 2022-03-31:
Hi @charon22,

1. About Request-URI, I think Twilio is right, `The "userinfo" and "@" components of the SIP URI MUST NOT be present.`

2. But about the digest, I don't think it should be a calculation error, The only difference is whether Twilio supports uri with port number `uri="sip:epicsafety.sip.twilio.com:5060"`. here, some open-source servers such as Freeswitch and FreePBX can be authenticated normally.

btw, Thanks for your report.

charon22, 2022-03-31:
Hi Ahhfzhang,
Thank you for the reply. I are going to test it with the new IDF version(v4.4) and ADF(v2.4) .

charon22, 2022-04-22:
Hi Ahhfzhang,

Would you like to hear the answer to my question?
I want to know how it goes.
Thanks,
Sungsu

charon22, 2022-12-08:
Can you close this issue ?

closed 2022-12-08 by charon22