espressif/esp-adf #792: master分支的pipeline_tcp_client 例程出现段错误

SeaHi-Mo, 2022-03-25:
开发板：lyrat_v4_3

HengYongChao, 2022-03-25:
Hi @SeaHi-Mo 

Looking at the log, it seems that the tcp open is not successful. Please check whether the tcp server is running correctly.

Bruce213, 2022-04-08:
Hi,
i met the same problem

I (27) boot: ESP-IDF v4.3.2-230-gab8b292407-dirty 2nd stage bootloader
I (27) boot: compile time 21:59:59
I (28) boot: chip revision: 3
I (32) boot_comm: chip revision: 3, min. bootloader chip revision: 0
I (39) boot.esp32: SPI Speed      : 40MHz
I (44) boot.esp32: SPI Mode       : DIO
I (48) boot.esp32: SPI Flash Size : 8MB
I (53) boot: Enabling RNG early entropy source...
I (58) boot: Partition Table:
I (62) boot: ## Label            Usage          Type ST Offset   Length
I (69) boot:  0 nvs              WiFi data        01 02 00009000 00004000
I (77) boot:  1 phy_init         RF data          01 01 0000d000 00001000
I (84) boot:  2 factory          factory app      00 00 00010000 00300000
I (92) boot: End of partition table
I (96) boot_comm: chip revision: 3, min. application chip revision: 0
I (103) esp_image: segment 0: paddr=00010020 vaddr=3f400020 size=211e8h (135656) map
I (160) esp_image: segment 1: paddr=00031210 vaddr=3ffb0000 size=03fa8h ( 16296) load
I (167) esp_image: segment 2: paddr=000351c0 vaddr=40080000 size=0ae58h ( 44632) load
I (186) esp_image: segment 3: paddr=00040020 vaddr=400d0020 size=9b768h (636776) map
I (416) esp_image: segment 4: paddr=000db790 vaddr=4008ae58 size=0c064h ( 49252) load
I (437) esp_image: segment 5: paddr=000e77fc vaddr=50000000 size=00010h (    16) load
I (448) boot: Loaded app from partition at offset 0x10000
I (448) boot: Disabling RNG early entropy source...
I (460) cpu_start: Pro cpu up.
I (460) cpu_start: Starting app cpu, entry point is 0x400813dc
I (0) cpu_start: App cpu up.
I (474) cpu_start: Pro cpu start user code
I (474) cpu_start: cpu freq: 160000000
I (474) cpu_start: Application information:
I (479) cpu_start: Project name:     tcp_client_example
I (485) cpu_start: App version:      v2.4-dirty
I (490) cpu_start: Compile time:     Apr  8 2022 21:59:45
I (496) cpu_start: ELF file SHA256:  43da715cfba3603e...
I (502) cpu_start: ESP-IDF:          v4.3.2-230-gab8b292407-dirty
I (509) heap_init: Initializing. RAM available for dynamic allocation:
I (516) heap_init: At 3FFAE6E0 len 00001920 (6 KiB): DRAM
I (522) heap_init: At 3FFB7E70 len 00028190 (160 KiB): DRAM
I (528) heap_init: At 3FFE0440 len 00003AE0 (14 KiB): D/IRAM
I (535) heap_init: At 3FFE4350 len 0001BCB0 (111 KiB): D/IRAM
I (541) heap_init: At 40096EBC len 00009144 (36 KiB): IRAM
I (548) spi_flash: detected chip: gd
I (552) spi_flash: flash io: dio
I (557) cpu_start: Starting scheduler on PRO CPU.
I (0) cpu_start: Starting scheduler on APP CPU.
I (601) TCP_CLIENT_EXAMPLE: [ 1 ] Start codec chip
I (621) TCP_CLIENT_EXAMPLE: [2.0] Create audio pipeline for playback
I (621) TCP_CLIENT_EXAMPLE: [2.1] Create i2s stream to write data to codec chip
I (651) TCP_CLIENT_EXAMPLE: [2.2] Create mp3 decoder to decode mp3 file
I (651) TCP_CLIENT_EXAMPLE: [2.2] Create tcp client stream to read data
I (661) TCP_CLIENT_EXAMPLE: [2.3] Register all elements to audio pipeline
I (661) TCP_CLIENT_EXAMPLE: [2.4] Link it together tcp-->mp3-->i2s
I (671) TCP_CLIENT_EXAMPLE: [ 3 ] Start and wait for Wi-Fi network
E (681) gpio: gpio_install_isr_service(460): GPIO isr service already installed
W (1631) PERIPH_WIFI: WiFi Event cb, Unhandle event_base:WIFI_EVENT, event_id:4
I (2601) TCP_CLIENT_EXAMPLE: [ 4 ] Set up  event listener
I (2601) TCP_CLIENT_EXAMPLE: [4.1] Listening event from all elements of pipeline
I (2601) TCP_CLIENT_EXAMPLE: [4.2] Listening event from peripherals
I (2611) TCP_CLIENT_EXAMPLE: [ 5 ] Start audio_pipeline
W (2711) wifi:<ba-add>idx:0 (ifx:0, f4:a5:9d:26:0e:1c), tid:0, ssn:1, winSize:64
W (2711) TCP_STREAM: TCP server actively closes the connection
E (2721) MP3_DECODER: failed to read audio data (line 117)
E (2721) AUDIO_ELEMENT: [mp3] AEL_STATUS_ERROR_OPEN,-1
W (2721) AUDIO_ELEMENT: [mp3] audio_element_on_cmd_error,7
W (2731) AUDIO_ELEMENT: IN-[i2s] AEL_IO_ABORT
W (2761) TCP_CLIENT_EXAMPLE: [ * ] Stop event received
I (2761) TCP_CLIENT_EXAMPLE: [ 6 ] Stop audio_pipeline
E (2771) AUDIO_ELEMENT: [tcp] Element already stopped
E (2771) AUDIO_ELEMENT: [mp3] Element already stopped
E (2781) AUDIO_ELEMENT: [i2s] Element already stopped
W (2781) AUDIO_PIPELINE: There are no listener registered
W (2791) AUDIO_ELEMENT: [tcp] Element has not create when AUDIO_ELEMENT_TERMINATE
W (2801) AUDIO_ELEMENT: [i2s] Element has not create when AUDIO_ELEMENT_TERMINATE
W (2811) AUDIO_ELEMENT: [mp3] Element has not create when AUDIO_ELEMENT_TERMINATE
W (2821) wifi:<ba-del>idx
W (2821) PERIPH_WIFI: Wi-Fi disconnected from SSID 1308, auto-reconnect disabled, reconnect after 1000 ms
W (2831) PERIPH_WIFI: WiFi Event cb, Unhandle event_base:WIFI_EVENT, event_id:3

Bruce213, 2022-04-08:
hi,
i alse met problem in this demo,

I (27) boot: ESP-IDF v4.3.2-230-gab8b292407-dirty 2nd stage bootloader
I (27) boot: compile time 21:59:59
I (28) boot: chip revision: 3
I (32) boot_comm: chip revision: 3, min. bootloader chip revision: 0
I (39) boot.esp32: SPI Speed      : 40MHz
I (44) boot.esp32: SPI Mode       : DIO
I (48) boot.esp32: SPI Flash Size : 8MB
I (53) boot: Enabling RNG early entropy source...
I (58) boot: Partition Table:
I (62) boot: ## Label            Usage          Type ST Offset   Length
I (69) boot:  0 nvs              WiFi data        01 02 00009000 00004000
I (77) boot:  1 phy_init         RF data          01 01 0000d000 00001000
I (84) boot:  2 factory          factory app      00 00 00010000 00300000
I (92) boot: End of partition table
I (96) boot_comm: chip revision: 3, min. application chip revision: 0
I (103) esp_image: segment 0: paddr=00010020 vaddr=3f400020 size=211e8h (135656) map
I (160) esp_image: segment 1: paddr=00031210 vaddr=3ffb0000 size=03fa8h ( 16296) load
I (167) esp_image: segment 2: paddr=000351c0 vaddr=40080000 size=0ae58h ( 44632) load
I (186) esp_image: segment 3: paddr=00040020 vaddr=400d0020 size=9b768h (636776) map
I (416) esp_image: segment 4: paddr=000db790 vaddr=4008ae58 size=0c064h ( 49252) load
I (437) esp_image: segment 5: paddr=000e77fc vaddr=50000000 size=00010h (    16) load
I (448) boot: Loaded app from partition at offset 0x10000
I (448) boot: Disabling RNG early entropy source...
I (460) cpu_start: Pro cpu up.
I (460) cpu_start: Starting app cpu, entry point is 0x400813dc
I (0) cpu_start: App cpu up.
I (474) cpu_start: Pro cpu start user code
I (474) cpu_start: cpu freq: 160000000
I (474) cpu_start: Application information:
I (479) cpu_start: Project name:     tcp_client_example
I (485) cpu_start: App version:      v2.4-dirty
I (490) cpu_start: Compile time:     Apr  8 2022 21:59:45
I (496) cpu_start: ELF file SHA256:  43da715cfba3603e...
I (502) cpu_start: ESP-IDF:          v4.3.2-230-gab8b292407-dirty
I (509) heap_init: Initializing. RAM available for dynamic allocation:
I (516) heap_init: At 3FFAE6E0 len 00001920 (6 KiB): DRAM
I (522) heap_init: At 3FFB7E70 len 00028190 (160 KiB): DRAM
I (528) heap_init: At 3FFE0440 len 00003AE0 (14 KiB): D/IRAM
I (535) heap_init: At 3FFE4350 len 0001BCB0 (111 KiB): D/IRAM
I (541) heap_init: At 40096EBC len 00009144 (36 KiB): IRAM
I (548) spi_flash: detected chip: gd
I (552) spi_flash: flash io: dio
I (557) cpu_start: Starting scheduler on PRO CPU.
I (0) cpu_start: Starting scheduler on APP CPU.
I (601) TCP_CLIENT_EXAMPLE: [ 1 ] Start codec chip
I (621) TCP_CLIENT_EXAMPLE: [2.0] Create audio pipeline for playback
I (621) TCP_CLIENT_EXAMPLE: [2.1] Create i2s stream to write data to codec chip
I (651) TCP_CLIENT_EXAMPLE: [2.2] Create mp3 decoder to decode mp3 file
I (651) TCP_CLIENT_EXAMPLE: [2.2] Create tcp client stream to read data
I (661) TCP_CLIENT_EXAMPLE: [2.3] Register all elements to audio pipeline
I (661) TCP_CLIENT_EXAMPLE: [2.4] Link it together tcp-->mp3-->i2s
I (671) TCP_CLIENT_EXAMPLE: [ 3 ] Start and wait for Wi-Fi network
E (681) gpio: gpio_install_isr_service(460): GPIO isr service already installed
W (1631) PERIPH_WIFI: WiFi Event cb, Unhandle event_base:WIFI_EVENT, event_id:4
I (2601) TCP_CLIENT_EXAMPLE: [ 4 ] Set up  event listener
I (2601) TCP_CLIENT_EXAMPLE: [4.1] Listening event from all elements of pipeline
I (2601) TCP_CLIENT_EXAMPLE: [4.2] Listening event from peripherals
I (2611) TCP_CLIENT_EXAMPLE: [ 5 ] Start audio_pipeline
W (2711) wifi:<ba-add>idx:0 (ifx:0, f4:a5:9d:26:0e:1c), tid:0, ssn:1, winSize:64
W (2711) TCP_STREAM: TCP server actively closes the connection
E (2721) MP3_DECODER: failed to read audio data (line 117)
E (2721) AUDIO_ELEMENT: [mp3] AEL_STATUS_ERROR_OPEN,-1
W (2721) AUDIO_ELEMENT: [mp3] audio_element_on_cmd_error,7
W (2731) AUDIO_ELEMENT: IN-[i2s] AEL_IO_ABORT
W (2761) TCP_CLIENT_EXAMPLE: [ * ] Stop event received
I (2761) TCP_CLIENT_EXAMPLE: [ 6 ] Stop audio_pipeline
E (2771) AUDIO_ELEMENT: [tcp] Element already stopped
E (2771) AUDIO_ELEMENT: [mp3] Element already stopped
E (2781) AUDIO_ELEMENT: [i2s] Element already stopped
W (2781) AUDIO_PIPELINE: There are no listener registered
W (2791) AUDIO_ELEMENT: [tcp] Element has not create when AUDIO_ELEMENT_TERMINATE
W (2801) AUDIO_ELEMENT: [i2s] Element has not create when AUDIO_ELEMENT_TERMINATE
W (2811) AUDIO_ELEMENT: [mp3] Element has not create when AUDIO_ELEMENT_TERMINATE
W (2821) wifi:<ba-del>idx
W (2821) PERIPH_WIFI: Wi-Fi disconnected from SSID 1308, auto-reconnect disabled, reconnect after 1000 ms
W (2831) PERIPH_WIFI: WiFi Event cb, Unhandle event_base:WIFI_EVENT, event_id:3

Get the esp32.mp3 size is 4072994
starting listen on ip 192.168.2.22, port 8881
waiting for client to connect
Traceback (most recent call last):
  File "tcp_server.py", line 79, in <module>
    start_tcp_server(socket.gethostbyname(socket.getfqdn(socket.gethostname())),SERVER_PORT)
  File "tcp_server.py", line 62, in start_tcp_server
    client.send(send_msg)
TypeError: a bytes-like object is required, not 'str'