espressif/esp-adf #793: What types of chips can run voip routines, except ESP32？

CAIJUN111, 2022-03-26:
[image]
[image]

None of the chips shown above can successfully compile routines ，except  ESP32.

CAIJUN111, 2022-03-26:
If I use ESP32S2 or another chip to connect to ES8388, can it replace ESP32?

HengYongChao, 2022-03-28:
Hi @CAIJUN111 

VIOP use esp-sr submodule, and esp-sr currently only supports esp32 and esp32s3 chip.

CAIJUN111, 2022-03-28:
> Hi @CAIJUN111
> 
> VIOP use esp-sr submodule, and esp-sr currently only supports esp32 and esp32s3 chip.
> 
> So, in addition to esp32, only esps3 supports this routine.

Thanks @HengYongChao, I select chip s3 to compile the routine,it won't be sucessful
..

closed 2022-03-28 by CAIJUN111