espressif/esp-adf #794: 语音识别和播放音乐冲突问题

Zhengqinze05, 2022-04-01:
你好，我想问一下，我移植了wwe功能，然后加上了audio_pipeline_run()的方式播放音乐，发现语音识别的准确率很低，说出不是命令词的指令都会被当初命令词。
我看到wwe里面也有播放功能esp_audio_sync_play()，并不会影响语音识别，使用pipeline播放如何能做到像esp_audio_sync_play()播放那样子的效果呢。
另外我需要播放特定的采样率，并非音乐本身的采样率，我在语音识别开启之后，使用了i2s_stream_set_clk()接口，但是这样的后果是会将整个系统卡死，我定位它卡死的是在I2S部分，请问如何解决I2S冲突的问题呢？
这是在ESPIDF的I2S卡住的地方，使用了i2s_stream_set_clk()就会卡在这个地方，[image]

HengYongChao, 2022-04-07:
> 你好，我想问一下，我移植了wwe功能，然后加上了audio_pipeline_run()的方式播放音乐，发现语音识别的准确率很低，说出不是命令词的指令都会被当初命令词。

这个你可以参考 adf 的 [dueros](https://github.com/espressif/esp-adf/tree/master/examples/dueros) 或 [esp_dispatcher_dueros](https://github.com/espressif/esp-adf/tree/master/examples/advanced_examples/esp_dispatcher_dueros) 的实现，两个例程均有唤醒的命令词的使用。

> 我看到wwe里面也有播放功能esp_audio_sync_play()，并不会影响语音识别，使用pipeline播放如何能做到像esp_audio_sync_play()播放那样子的效果呢。

[esp_audio_sync_play()](https://github.com/espressif/esp-adf-libs/blob/82b6d534e228f53d75aac825c5fdbdda33a8c33c/esp_audio/include/esp_audio.h#L253) 是阻塞播放， 它或播放直到音乐结束或者异常发生。esp_audio_play()  可以被插入打断， api 注释给出了解释。

> 另外我需要播放特定的采样率，并非音乐本身的采样率，我在语音识别开启之后，使用了i2s_stream_set_clk()接口，但是这样的后果是会将整个系统卡死，我定位它卡死的是在I2S部分，请问如何解决I2S冲突的问题呢？
> 这是在ESPIDF的I2S卡住的地方，使用了i2s_stream_set_clk()就会卡在这个地方，

关于 `另外我需要播放特定的采样率，并非音乐本身的采样率` ，我猜你是要使用 resample 吧， 可以参考 [pipeline_resample](https://github.com/espressif/esp-adf/tree/master/examples/audio_processing/pipeline_resample)  用法。

Zhengqinze05, 2022-04-12:
感谢您的回复resample是否只支持几种特定的播放采样率，如44100，8000，16000，但是像13257这种采样率是不支持的是吗

dan-vital, 2022-04-25:
@Zhengqinze05 哈喽，您好！我最近也在尝试语音识别控制音乐播放的功能，在移植的过程中遇到头文件找到的问题，您是否也遇到过呢？如果有，您是怎么解决的呢？如能告知，不胜感激！

在编译的时候就出错了
[image]

可是该头文件是存在的呀
[image]

HengYongChao, 2022-05-09:
> @Zhengqinze05 哈喽，您好！我最近也在尝试语音识别控制音乐播放的功能，在移植的过程中遇到头文件找到的问题，您是否也遇到过呢？如果有，您是怎么解决的呢？如能告知，不胜感激！
> 
> 在编译的时候就出错了 [image]
> 
> 可是该头文件是存在的呀 [image]

你编译的时候，是否指定了 $ADF_PATH 的 path 路径了呢？ 另一个你是否添加了相关的头文件。 建议检查下这两处地方。

jason-mao, 2022-08-10:
> 你好，我想问一下，我移植了wwe功能，然后加上了audio_pipeline_run()的方式播放音乐，发现语音识别的准确率很低，说出不是命令词的指令都会被当初命令词。
> 我看到wwe里面也有播放功能esp_audio_sync_play()，并不会影响语音识别，使用pipeline播放如何能做到像esp_audio_sync_play()播放那样子的效果呢。 [...]

i2s_stream_set_clk 卡住是 I2S 驱动的一个bug，目前已经在最新的IDF 版本上修复。

closed 2022-08-10 by jason-mao

meibao-real, 2023-10-11:
请问解决了吗，在最新版ESP-IDF和MASTER版本ADF使用了i2s_stream_set_clk接口后再用esp_audio_play接口播放会卡死，在寻找解决资料中