espressif/esp-adf #795: WWE的语音识别不能随时停止恢复吗？

Zhengqinze05, 2022-04-01:
你好，我用的是esp32_s3_korvo2_v3的板子，使用的是官方WWE工程，添加了停止和恢复语音识别的管道，当我想停下语音识别时，就出现以下报错，我的停止和恢复代码是这样的

[image]

                                               停止代码之后一直报这种错误

[image]

                                              恢复之后的log报错是这样的

[image]

HengYongChao, 2022-04-08:
Hi @Zhengqinze05 

你是要实现语音识别的 mute 功能吧， ADF 的 recorder 中有 API 可以帮你实现这个功能, 不必单独再搞一套 pipeline， 希望可以帮你。

code link:  https://github.com/espressif/esp-adf/blob/master/components/audio_recorder/include/audio_recorder.h#L152
```c
// 唤醒词使能
esp_err_t audio_recorder_wakenet_enable(audio_rec_handle_t handle, bool enable);

// 命令词使能
esp_err_t audio_recorder_multinet_enable(audio_rec_handle_t handle, bool enable);
```

Zhengqinze05, 2022-04-12:
感谢你的回复，其实我是想停止pipeline的，目前自己做了一个pipeline的停止

HengYongChao, 2022-04-15:
@Zhengqinze05 

可否提供 具体是指那些词汇呢？ 一般 menuconfig 中选择模型越大，效果越好，当然需要资源也越多些。

Zhengqinze05, 2022-04-18:
> @Zhengqinze05
> 
> 可具体是那些词条中选择模型，指越一般的菜单，当然可以当然需要一些资源也多。

像“ni hao;zuo zhuan;you zhuan;qian jin;hou tui;”这几个词，很容易识别错误，有esp-sr更新说两个词的识别提高，我更新了，确实识别两个单词的比之前好了一些，但是还是经常会出现识别错误，而且识别效率还比之前慢了一些，稍微远一点，有时候唤醒词也难以唤醒

Zhengqinze05, 2022-04-18:
并且唤醒词感觉没有之前那么灵敏

wangxiankang, 2022-11-10:
> 感谢你的回复，其实我是想停止pipeline的，目前自己做了一个pipeline的停止

> 
您好，请问您的随时停止和恢复实现代码可以分享一下吗

HongYan12345, 2022-12-06:
@Zhengqinze05 你好，能分享一下随时停止和回复管道的代码吗，我在做的项目遇到了同样的问题

Zhengqinze05, 2022-12-07:
> @Zhengqinze05 你好，能分享一下随时停止和回复管道的代码吗，我在做的项目遇到了同样的问题

停止pipeline的代码在上面图片中有，语音识别停止的接口在audio_recorder文件夹里的recorder_sr.c和audio_recorder.c里面有

closed 2023-09-05 by jason-mao