espressif/esp-adf #796: ESP32-S3-KORVO-2 V3 OV2640 Camera

JVKran, 2022-04-07:
Hi There!

Hope you're doing great today. I stumbled upon a problem with the ESP32-S3 Korvo 2 V3(.1?). When I connect an OV2640 to the corresponding FFC connector, the camera isn't recognized and I'm greeted with the following message.

``` bash
I (1411) s3 ll_cam: DMA Channel=4
I (1431) cam_hal: cam init ok
I (1441) sccb: pin_sda 17 pin_scl 18
E (1531) camera: Camera probe failed with error 0x105(ESP_ERR_NOT_FOUND)
W (1531) camera: Failed to initialize camera!
```

I think it might be related to another issue I've spotted. I think the CLK and SDA silk texts of the broken out pins are swapped; I can't connect sensors to it unless I connect SCL to SDA and vice versa. 😜 

Kind regards,

Jochem

HengYongChao, 2022-04-08:
Hi @JVKran 

It looks like a problem with the camera PIN links.

Which routine are you testing?  We have tested the 2640 before, the example [display/lcd_camera](https://github.com/espressif/esp-adf/tree/master/examples/display/lcd_camera) works fine, Please check it.

The camera model tested in this example is OV3660, for OV2640 model please refer to [esp32-camera](https://github.com/espressif/esp32-camera).

closed 2022-04-13 by JVKran

Curioplus, 2023-02-01:
> The issue was using an invalid clock frequency; I was using a frequency of 1MHz while the datasheet states a frequency between 6- and 20MHz is to be provided. Doing so fixed the described issue.

Hi
Do you have the code you changed to share?