espressif/esp-adf #798: Audio Player delay too long

lonelyjesse, 2022-04-09:
Hi, 
Our project is based on one of ADF demo projects, which is get-started/play_mp3_control. 
It's OK when we use the mp3 file from your project, but when it comes to our own mp3 file, which is 128kbps or 320kbps,
there is a delay. In worse situation, the delay reaches 10 seconds.
Would you please help me get out of this? It's an emergency.
I am really hope to receive your reply.

[delay_4S.txt](https://github.com/espressif/esp-adf/files/8456500/delay_4S.txt)

[music-16b-2c-44100hz.zip](https://github.com/espressif/esp-adf/files/8456503/music-16b-2c-44100hz.zip)

lonelyjesse, 2022-04-09:
PS:
It's OK when it runs in second round.
[source_code.txt](https://github.com/espressif/esp-adf/files/8456509/source_code.txt)

HengYongChao, 2022-04-15:
Hi @lonelyjesse 

[image]

This song has almost 2 seconds of silence data at the beginning, I still recommend you to test with different audio in #797 .

closed 2022-08-08 by jason-mao