espressif/esp-adf #799: algorithm_stream可以实现BSS的音频处理吗

LiTongXue98, 2022-04-12:
algorithm_stream目前好像只有NS，BSS后面会加入进来吗

HengYongChao, 2022-04-14:
Hi @LiTongXue98 

这个功能在 esp32s3 的演示视频上已经演示了 BBS 算法，请关注 [esp-skaine](https://github.com/espressif/esp-skainet) 和 [esp-sr](https://github.com/espressif/esp-sr) 仓库的更新。

LiTongXue98, 2022-04-14:
> Hi @LiTongXue98 
> 
> 这个功能在 esp32s3 的演示视频上已经演示了 BBS 算法，请关注 [esp-skaine](https://github.com/espressif/esp-skainet) 和 [esp-sr](https://github.com/espressif/esp-sr) 仓库的更新。

是的，我有进行使用skainet但是发现好像在afe feed的数据方式和adf的有一些区别，adf使用的是i2s流传入的，skainet是从spk的电路上采集的信号，这两种方式对于采集信号和回采信号的信号同步是不是有差异呢

LiTongXue98, 2022-04-14:
我用adf和skainet两种方式进行音频采集，然后进行ns和aec，发现adf里面使用的方式对于aec来说可以消除的更加干净，并且我也让他们使用了相同版本的esp-sr

jason-mao, 2022-08-08:
@LiTongXue98 你的这个问题处理的如何？

LiTongXue98, 2022-08-08:
ADF可以实现BSS，但是我的设备没有功放

jason-mao, 2022-08-09:
@LiTongXue98 谢谢更新，我会关闭这个问题。若遇到任何问题请新建Issue。

closed 2022-08-09 by jason-mao