espressif/esp-adf #802: 关于ADF例程“pipeline_raw_http”的问题

AdonisXie, 2022-04-21:
想问一下，我在运行例程：**pipeline_raw_http**的时候，
我已经成功实现了从板子录入声音，然后通过http协议传送到电脑中
文件在电脑中的后缀名为wav，
为什么当我用potplayer打开音频文件的时候，只有嘶嘶声、电流声，就是听不到任何可识别的声音
难道是Mic头录制不到声音吗

可我在使用dueros的时候，是可以准确识别我说的话的
到底是为什么呢？就我一个人运行这个例程是这种表现吗？
还望解答一下！谢谢！

HengYongChao, 2022-04-22:
Hi @AdonisXie 

[dueros](https://github.com/espressif/esp-adf/tree/master/examples/dueros) 例程录音可以，可以说明你的开发板应该没有问题。

[pipeline_raw_http](https://github.com/espressif/esp-adf/tree/master/examples/recorder/pipeline_raw_http) 例程不可以，你需要检查 http server 正确运行了，并且检查 PC 端录制的音频大小是否正常。我怀疑你的 PC 端的 http server 并没有正常的写录音文件。

AdonisXie, 2022-04-26:
谢谢你，的确是这样，现在已经解决了

lanistor, 2022-04-27:
@AdonisXie 怎么解决的？我的也是这个表现。server可以写文件，但是写出来的就是"呲呲"的声音。

AdonisXie, 2022-04-27:
首先，录音的代码段和dueros一样

```

static void voice_read_task(void *args){

    const int buf_len =  3 * 1024;
    bool runing = true;
    uint8_t *voiceData = audio_calloc(1, buf_len);

    while (runing){
        EventBits_t bits = xEventGroupWaitBits(smart_speaker_eventgroup, VOICE_REC_READING, false, true, portMAX_DELAY);
        if (bits & VOICE_REC_READING) {
            int ret = audio_recorder_data_read(recorder, voiceData , buf_len, portMAX_DELAY);
            ESP_LOGI(TAG_2, "The real recorde---data:%d",ret);
            if (ret == 0 || ret == -1) {
                xEventGroupClearBits(smart_speaker_eventgroup, VOICE_REC_READING);
            } else{
                speaker_voice_upload(voiceData, ret);
            }
        }
    }
    free(voiceData);
    xEventGroupClearBits(smart_speaker_eventgroup, VOICE_REC_READING);
    vTaskDelete(NULL);
}

```

然后upload函数是这样的

```
esp_err_t speaker_voice_upload(void *buf, int len){

    http_msg_t msg = {0};
    msg.type = SPEAKER_CMD_UPLOAD;
    msg.pdata = buf;
    msg.len = len;
    ESP_LOGW(TAG, "Queue Spaces Available: %d", uxQueueSpacesAvailable(http_que));
    
    xQueueSend(http_que, &msg, portMAX_DELAY);

    xEventGroupWaitBits(speaker_evt, SPEAKER_EVT_UPLOADED, true, true, pdMS_TO_TICKS(3000));
    return ESP_OK;

}
```

这个有两点说一下：
①xQueueSend就是向队列发送消息，消息中包含了录到的音的【地址】与【长度】
②xEventGroupWaitBits这句话一定要有，我之前就是改了这里，然后录的音质量提升了，最长等待时间设置为2s或3s都行。至于原因，我猜测是因为recorer录制声音的速度要远远大于esp32写http流的速度，所以需要在这里设置一个缓冲。

因为之前我还把等待的时间设置为最长时间 portMAX_DELAY，这就意味着，必须发送完一次http流，才能接着发下一次，但不理想，失真了很多，然后就印证了我的想法，写http流的速度太慢了，导致你写完之后，都许久没有进行audio_recorder_data_read了，所以失真。

但是不等也不行，因为你看录制的那里： audio_recorder_data_read(recorder, voiceData , buf_len, portMAX_DELAY);，它把录制的音频二进制放在了voiceData，这是一个起始地址，然后你发送http流的地址也是这个，试想一下，你要是不等的话，那么你xQueueSend(http_que, &msg, portMAX_DELAY);之后就马上return了，然后马上又得audio_recorder_data_read，你前一次的http流没发出去，又要把新一次的录音数据即起始地址voiceData这里，那怎么行呢

所以必须等待，稍微等待一会儿，这个就叫缓冲。

然后监听消息队列的任务，就会做这件事

```
if(msg.type == SPEAKER_CMD_UPLOAD){
                 speaker_http_voice_send(msg.pdata,msg.len);
                 xEventGroupSetBits(speaker_evt, SPEAKER_EVT_UPLOADED);
}
```

```
esp_err_t speaker_http_voice_send(void *data, int size){
    if(!client_opened){
        
        esp_http_client_set_method(client, HTTP_METHOD_POST);
        char dat[10] = {0};
        snprintf(dat, sizeof(dat), "%d", EXAMPLE_AUDIO_SAMPLE_RATE);
        esp_http_client_set_header(client, "x-audio-sample-rates", dat);
        memset(dat, 0, sizeof(dat));
        snprintf(dat, sizeof(dat), "%d", EXAMPLE_AUDIO_BITS);
        esp_http_client_set_header(client, "x-audio-bits", dat);
        memset(dat, 0, sizeof(dat));
        snprintf(dat, sizeof(dat), "%d", EXAMPLE_AUDIO_CHANNELS);
        esp_http_client_set_header(client, "x-audio-channel", dat);
        client_opened = true;

        err = esp_http_client_open(client, size);
        
        if (err != ESP_OK) {
            ESP_LOGE(TAG, "Failed to open HTTP connection: %s", esp_err_to_name(err));
        }

    }
    
    int wlen = sprintf(len_buf, "%x\r\n", size);
    if (esp_http_client_write(client, len_buf, wlen) <= 0) {
        ESP_LOGE(TAG, " [ 1 ]  Write Failed");
        return ESP_FAIL;
    }
    if (esp_http_client_write(client, data, size) <= 0) {
        ESP_LOGE(TAG, " [ 2 ]  Write Failed");
    }
    if (esp_http_client_write(client, "\r\n", 2) <= 0) {
        ESP_LOGE(TAG, " [ 3 ]  Write Failed");
    }
    total_write += size;
    ESP_LOGI(TAG, "  Total write %d ",total_write);
    
    return ESP_OK;
}
```

函数开头是判断有没有打开http_client，没打开的话就得设置头文件，然后打开client
之后的话就是发送了，这里连续写了三次，这三句话缺一不可
其中第二次发送，发送的才是真正的音频数据
为什么第一次和第三次要写/r/n，这其实就是和http服务端的通信协议了
如果你去看那个python代码，你就知道有一个函数是get_chunk_size，这第一次和第三次写的/r/n，就是为了那个服务的
仔细看看

之后呢就是不断地发送了，期间会调用speaker_http_voice_send多次
最后就是结束发送了，这个一定要有，否则服务端都不知道你发送完了

```
esp_err_t speaker_voice_over_uploaded(){

    if (esp_http_client_write(client, "0\r\n\r\n", 5) <= 0) {
        ESP_LOGE(TAG, " [ 4 ]  Write Failed");
        return ESP_FAIL;
    }

    int content_length = esp_http_client_fetch_headers(client);
        if (content_length < 0) {
            ESP_LOGE(TAG, " [ 5 ] HTTP client fetch headers failed");
        } else {
            int data_read = esp_http_client_read_response(client, output_buffer, MAX_HTTP_OUTPUT_BUFFER);
            if (data_read >= 0) {
                ESP_LOGI(TAG, "HTTP GET Status = %d, content_length = %d",
                    esp_http_client_get_status_code(client),
                    esp_http_client_get_content_length(client) );
                ESP_LOGI(TAG, "Got HTTP Response = %s", (char *)output_buffer);
            } else {
                ESP_LOGE(TAG, " [ 6 ] Failed to read response");
            }
            esp_http_client_close(client);
            client_opened = false;
        }

    return ESP_OK;
}

```

看上面函数的第一个write，esp_http_client_write(client, "0\r\n\r\n", 5) 
这个是为了什么
其实就是为了告诉服务端，我发送完了

之后就可以在电脑上听到声音了
用作语音识别已经足够了

你听到滋滋声，如果dueros例程没有问题的话，那么问题就是你发送给电脑的声音二进制，那个声音是不ok的
因为http流发送比录制的要慢，
所以你发送太快不行，发送太慢也不行，必须设置一个缓冲区
@[lanistor](https://github.com/lanistor)

lanistor, 2022-05-12:
@AdonisXie 感谢感谢。Demo可以提供一份吗？

AdonisXie, 2022-05-12:
Demo和dueros一样，一样的流程。而且我做的那个代码是为我的项目服务的，里面有很多你不需要用的

lanistor, 2022-05-12:
@AdonisXie 你用的开发板和adf idf的分支分别是什么？我跑Dueros的Demo，到语音交互那一步没有反应，手动点击Rec按键，会报warn: `W (355584) ESP_AUDIO_TASK: The old pipeline destroyed, FINISHED`。

我用的是`ESP32-Lyrat-Mini V1.2`板子，adf是release/v2.4，idf版本是`release/v4.4`。

HengYongChao, 2022-05-18:
> @AdonisXie 你用的开发板和adf idf的分支分别是什么？我跑Dueros的Demo，到语音交互那一步没有反应，手动点击Rec按键，会报warn: `W (661493) AUDIO_EVT: There is no space in external queue`。
> 
> 我用的是`ESP32-Lyrat-Mini V1.2`板子，adf是release/v2.4，idf版本是`release/v4.4`。

Dueros 的中文说明文档（https://github.com/espressif/esp-adf/blob/master/examples/dueros/README_CN.md）您看了吗？ 按照里面的步骤来，问题还是存在吗？

lanistor, 2022-05-19:
@HengYongChao 是的。可能是板子的问题，ESP32-LyraT V4.3的板子是可以的，我准备再买块板子测试下

closed 2022-08-08 by jason-mao