espressif/esp-adf #804: Broken I2S Internal DAC output

jitenshap, 2022-04-24:
I installed ESP-ADF using VS Code Extension today.

  
    C:\Users\admin\esp\.espressif\esp-adf>git branch --contains
        
    * master  
  
I built examples `/player/pipeline_play_mp3_with_dac_or_pwm` and flash it to my ESP32 board.  
I expected the sound of MP3 files to be played from GPIO25 and GPIO26, but the noise is terrible and the sound is distorted.  
The output waveform of GPIO25 is attached.  

[image]

HengYongChao, 2022-05-09:
Hi @jitenshap 

Looking at your picture, it seems there is not much detail in this picture.. I don't know what the terrible noise you are talking about is. When I run this example here, the sound is played normally.

I upload the log and sdkconfig to you, as well as a video of the operation. Hope it helps to solve your problem.

[sdkconfig.esp32.pipeline_play_mp3_with_dac_or_pwm.txt](https://github.com/espressif/esp-adf/files/8649304/sdkconfig.esp32.pipeline_play_mp3_with_dac_or_pwm.txt)

[pipeline_play_mp3_with_dac_or_pwm.log](https://github.com/espressif/esp-adf/files/8649305/pipeline_play_mp3_with_dac_or_pwm.log)

https://user-images.githubusercontent.com/4134628/167354693-912695af-7592-4358-b94d-39666b76e8fd.MP4

sylvainleris, 2022-05-12:
Hi,

I have the same problem as @jitenshap. @HengYongChao, on your video, I can see you are testing GPIO26 while @jitenshap talking about noise on GPIO25. My GPIO25 output waveform :
[image]

The sound on GPIO26 is horrible. You can see the output waveform :
[image]

@HengYongChao, I can also heard on your video that the sound played is not the original sound. It's as if it was played in fast motion...

jitenshap, 2022-05-12:
ESP-IDF v4.3.2 is good to play in internal DAC.  
```
C:\Users\admin\esp\esp-idf>git log -n1
commit 8bf14a9238329954c7c5062eeeda569529aedf75 (HEAD, tag: v4.3.2)
```    
  
ESP-IDF bundled in ESP-ADF is v4.4.  
ESP-IDF v4.4 is couln't play clearly sound in internal DAC.
  
```
C:\Users\admin\esp\.espressif\esp-adf\esp-idf>git log -n1  
commit 3c9657fa4e05ff6ee5a7ec450c408d8cfaa148ca (HEAD)
```  
I will post additional information over the weekend.

jitenshap, 2022-05-15:
There is a crunchy sound.  
https://user-images.githubusercontent.com/8150731/168475531-dd196b6f-5e5f-4ed7-9640-9e1795d8a206.mp4
.

jitenshap, 2022-05-15:
IDF v4.3.2 is low volume. But good sound.
https://user-images.githubusercontent.com/8150731/168478335-cb5635d3-07fe-4009-9d82-0530216ceac7.mp4

HengYongChao, 2022-05-16:
Hi @jitenshap 

Please try this change, https://github.com/espressif/esp-adf/pull/821

closed 2022-05-16 by jitenshap