espressif/esp-adf #807: 关于使用ADF高封装库 “ ESP Audio ” 的问题

AdonisXie, 2022-04-26:
大家好，
我在使用ESP Audio中出现了一个问题，不知道该怎么解决
我是在参考dueros例程的基础上写的，可在我调用esp_audio_play()函数的时候，出现了问题
这是我对 esp_audio_handle_t  ，也就是dueros例程中的player   初始化的代码

```

void *speaker_audio_setup_player(void){

    if (player) {
        return player;
    }

    esp_audio_cfg_t cfg = DEFAULT_ESP_AUDIO_CONFIG();
    audio_board_handle_t board_handle = audio_board_init();

    cfg.vol_handle = board_handle->audio_hal;
    cfg.vol_set = (audio_volume_set)audio_hal_set_volume;
    cfg.vol_get = (audio_volume_get)audio_hal_get_volume;
    cfg.resample_rate = 48000;
    cfg.prefer_type = ESP_AUDIO_PREFER_MEM;
    cfg.evt_que = xQueueCreate(3, sizeof(esp_audio_state_t));

    player = esp_audio_create(&cfg);
    audio_hal_ctrl_codec(board_handle->audio_hal, AUDIO_HAL_CODEC_MODE_BOTH, AUDIO_HAL_CTRL_START);

    xTaskCreate(esp_audio_state_task, "player_task", 3 * 1024, cfg.evt_que, 1, NULL);

    fatfs_stream_cfg_t fs_reader = FATFS_STREAM_CFG_DEFAULT();
    fs_reader.type = AUDIO_STREAM_READER;
    esp_audio_input_stream_add(player, fatfs_stream_init(&fs_reader));

    http_stream_cfg_t http_cfg = HTTP_STREAM_CFG_DEFAULT();
    http_cfg.event_handle = _http_stream_event_handle;
    http_cfg.type = AUDIO_STREAM_READER;
    http_cfg.enable_playlist_parser = true;
    audio_element_handle_t http_stream_reader = http_stream_init(&http_cfg);
  
    esp_audio_input_stream_add(player, http_stream_reader);

    audio_decoder_t auto_decode[] = {
        DEFAULT_ESP_MP3_DECODER_CONFIG(),
        DEFAULT_ESP_WAV_DECODER_CONFIG(),
        DEFAULT_ESP_AAC_DECODER_CONFIG(),
        DEFAULT_ESP_M4A_DECODER_CONFIG(),
        DEFAULT_ESP_TS_DECODER_CONFIG(),
    };
    esp_decoder_cfg_t auto_dec_cfg = DEFAULT_ESP_DECODER_CONFIG();
    auto_dec_cfg.task_stack = 6*1024;
    esp_audio_codec_lib_add(player, AUDIO_CODEC_TYPE_DECODER, esp_decoder_init(&auto_dec_cfg, auto_decode, 5));

    i2s_stream_cfg_t i2s_writer = I2S_STREAM_CFG_DEFAULT();
    i2s_writer.i2s_config.sample_rate = 48000;
    i2s_writer.i2s_config.bits_per_sample = CODEC_ADC_BITS_PER_SAMPLE;
    i2s_writer.need_expand = (CODEC_ADC_BITS_PER_SAMPLE != I2S_BITS_PER_SAMPLE_16BIT);
    i2s_writer.type = AUDIO_STREAM_WRITER;
    

    esp_audio_output_stream_add(player, i2s_stream_init(&i2s_writer));

    esp_audio_vol_set(player, 60);
    ESP_LOGI(TAG, "esp_audio instance is:%p", player);

    return player;

}

```

然后这是打印出来的错误代码

```
I (34284) HTTP_SERVICE_CALLBACK: HTTP_EVENT_DISCONNECTED
I (34291) HTTP_SERVICE:  [ 7 ] succeed to read response http://96.ierge.cn/15/235/471737.mp3
E (34299) AUDIO_WRAPPER: Playing music: http://96.ierge.cn/15/235/471737.mp3
I (34307) ESP_AUDIO_CTRL: Enter media_ctrl_stop procedure, src:0, type:0, st:FINISHED, phase:0
W (34316) ESP_AUDIO_CTRL: [media_ctrl_stop]-Already stopped, status is FINISHED, phase:0
I (34325) ESP_AUDIO_CTRL: Enter play procedure, src:0
I (34330) ESP_AUDIO_CTRL: Play procedure, URL is ok, src:0
I (34337) ESP_AUDIO_CTRL: Request_CMD_Queue CMD:0, Available:5, que:0x3ffe67c8
I (34344) ESP_AUDIO_CTRL: Func:_ctrl_play, Line:771, MEM Total:3966403 Bytes, Inter:28955 Bytes, Dram:28955 Bytes

I (34344) ESP_AUDIO_TASK: It's a decoder
I (34360) ESP_AUDIO_TASK: 1.CUR IN:[IN_http],CODEC:[DEC_auto],RESAMPLE:[48000],OUT:[OUT_iis],rate:0,ch:0,pos:0
I (34371) ESP_AUDIO_TASK: 2.Handles,IN:0x3f81bf60,CODEC:0x3f81c158,FILTER:0x3f83a4a8,OUT:0x3f81c304
I (34381) AUDIO_PIPELINE: audio_pipeline_unlinked
W (34386) AUDIO_PIPELINE: There are no listener registered
I (34392) ESP_AUDIO_TASK: 2.2 Update all pipeline
I (34398) ESP_AUDIO_TASK: 2.3 Linked new pipeline
I (34403) AUDIO_PIPELINE: link el->rb, el:0x3f81bf60, tag:IN_http, rb:0x3f841a20
I (34411) AUDIO_PIPELINE: link el->rb, el:0x3f81c158, tag:DEC_auto, rb:0x3f841a5c
I (34420) AUDIO_PIPELINE: link el->rb, el:0x3f83a4a8, tag:Audio_forge, rb:0x3f841a98
I (34428) ESP_AUDIO_TASK: 3. Previous starting...
W (34433) AUIDO_MEM: Can't support stack on external memory due to ESP32 chip is 1
I (34587) AUDIO_ELEMENT: [IN_http-0x3f81bf60] Element task created
I (34587) AUDIO_ELEMENT: [IN_http] AEL_MSG_CMD_RESUME,state:1
W (34590) AUIDO_MEM: Can't support stack on external memory due to ESP32 chip is 1
I (34597) AUDIO_ELEMENT: [DEC_auto-0x3f81c158] Element task created
I (34604) AUDIO_ELEMENT: [DEC_auto] AEL_MSG_CMD_RESUME,state:1
I (34611) ESP_DECODER: Ready to do audio type check, pos:0, (line 104)
I (34612) ESP_AUDIO_TASK: Blocking play until received AEL_MSG_CMD_REPORT_MUSIC_INFO
I (34906) HTTP_CLIENT: Body received in fetch header state, 0x3fffd441, 1795
I (34906) HTTP_STREAM: total_bytes=10377952
I (34909) ESP_AUDIO_TASK: Recv Element[IN_http-0x3f81bf60] MSG,type:20000,cmd:8,len:4,status:AEL_STATUS_STATE_RUNNING
I (34941) ESP_DECODER: Detect audio type is MP3
I (34942) MP3_DECODER: MP3 opened
I (34950) ESP_AUDIO_TASK: Recv Element[DEC_auto-0x3f81c158] MSG,type:20000,cmd:8,len:4,status:AEL_STATUS_STATE_RUNNING
I (35050) ESP_AUDIO_TASK: Received muisc info then on play
I (35050) ESP_AUDIO_TASK: On event play, status:UNKNOWN, 0
'''

上面的代码都是绿色的，即正常的，
可一到下面，直接变红，前面都说 I (34942) MP3_DECODER: MP3 opened
下面报错说：[Audio_forge] audio_thread_create failed
意思是进程创建失败，再看下面，又说
[Audio_forge] Element has not create when AUDIO_ELEMENT_RESUME
意思是在恢复过程中不能创建吗？
可我也没有调用esp_audio_resume()

'''
W (35052) AUIDO_MEM: Can't support stack on external memory due to ESP32 chip is 1
E (35060) AUDIO_THREAD: Error creating task Audio_forge
E (35066) AUDIO_ELEMENT: [Audio_forge] audio_thread_create failed
W (35072) AUDIO_ELEMENT: [Audio_forge] Element has not create when AUDIO_ELEMENT_RESUME
E (35081) AUDIO_THREAD: Error creating task OUT_iis
E (35087) AUDIO_ELEMENT: [OUT_iis] audio_thread_create failed
W (35093) AUDIO_ELEMENT: [OUT_iis] Element has not create when AUDIO_ELEMENT_RESUME
I (35102) ESP_AUDIO_CTRL: Exit play procedure, ret:0
I (35102) ESP_AUDIO_TASK: Recv Element[Audio_forge-0x3f83a4a8] MSG,type:20000,cmd:8,len:4,status:ESP_ERR_AUDIO_OPEN
E (35118) ESP_AUDIO_TASK: Got element[Audio_forge] error:ESP_ERR_AUDIO_OPEN, stop pipeline[0]
W (35150) AUDIO_ELEMENT: IN-[DEC_auto] AEL_IO_ABORT
W (35150) AUDIO_ELEMENT: OUT-[DEC_auto] AEL_IO_ABORT
W (35152) MP3_DECODER: output aborted -3
I (35155) MP3_DECODER: Closed
E (35270) AUDIO_ELEMENT: [IN_http] Element already stopped
E (35270) AUDIO_ELEMENT: [DEC_auto] Element already stopped
W (35271) ESP_AUDIO_TASK: Destroy the old pipeline, ELEMENT OCCUR ERROR
W (35292) AUDIO_ELEMENT: [OUT_iis] Element has not create when AUDIO_ELEMENT_TERMINATE, tick:300
W (35293) AUDIO_ELEMENT: [Audio_forge] Element has not create when AUDIO_ELEMENT_TERMINATE, tick:300
W (35302) ESP_AUDIO_TASK: The old pipeline destroyed, ELEMENT OCCUR ERROR
I (35309) ESP_AUDIO_TASK: Recv Element[OUT_iis-0x3f81c304] MSG,type:20000,cmd:8,len:4,status:ESP_ERR_AUDIO_OPEN
I (35320) ESP_AUDIO_TASK: Received last pos: 2115 bytes
I (35326) ESP_AUDIO_TASK: Recv Element[DEC_auto-0x3f81c158] MSG,type:20000,cmd:8,len:4,status:AEL_STATUS_STATE_STOPPED
I (35337) ESP_AUDIO_TASK: Recv Element[IN_http-0x3f81bf60] MSG,type:20000,cmd:8,len:4,status:AEL_STATUS_STATE_STOPPED
I (35348) ESP_AUDIO_TASK: Recv Element[IN_http-0x3f81bf60] MSG,type:20000,cmd:8,len:4,status:AEL_STATUS_STATE_STOPPED
I (35360) ESP_AUDIO_TASK: Recv Element[IN_http-0x3f81bf60] MSG,type:20000,cmd:8,len:4,status:AEL_STATUS_STATE_STOPPED
I (35371) ESP_AUDIO_TASK: Recv Element[DEC_auto-0x3f81c158] MSG,type:20000,cmd:8,len:4,status:AEL_STATUS_STATE_STOPPED
I (35383) ESP_AUDIO_TASK: Recv Element[Audio_forge-0x3f83a4a8] MSG,type:20000,cmd:8,len:4,status:AEL_STATUS_STATE_STOPPED
I (35394) ESP_AUDIO_TASK: Recv Element[OUT_iis-0x3f81c304] MSG,type:20000,cmd:8,len:4,status:AEL_STATUS_STATE_STOPPED
```

我用的代码段是我自己参考dueros的，基本上一样，也是要通过录音将音频录取然后通过http协议传送到后台，再返回来url进行播放。
因为dueros的代码段封装了，已经成.a文件，所以有很多核心功能代码段，我都不知道是怎么处理的

所以碰到这种问题，我又不能返回源代码去看，去学习
所以不知道是哪里出了问题，是否有哪个逻辑没写上去？

希望能得到帮助！谢谢！

esp-target：esp32
IDF： v4.4-329-g3c9657fa4e-dirty

HengYongChao, 2022-04-27:
Hi @AdonisXie 

我认为这个 线程创建失败的可能原因是 内存不足了， 你打印下系统内存剩余看看。

并且，我在你的代码 log 里面，也找到了这个警告打印：
```c
 AUIDO_MEM: Can't support stack on external memory due to ESP32 chip is 1
```
这个意思是说你的芯片版本是 version 1， 而使用 psram 作为 task 栈是需要 version 3 版本的。 

建议你使用最新的 version 3 版本的芯片，打开 PSRAM 测试。

AdonisXie, 2022-05-25:
@HengYongChao 很抱歉打扰你啊！我知道你在ESP32开发是很厉害的，所以想请教一下问题
能否帮我解答一下这个问题？
https://github.com/espressif/esp-adf/issues/829
本来应该更有耐心等你上线，只不过，我后天要答辩了wuwu😭😭
拜托了，拜托了

closed 2022-08-08 by jason-mao

Wantgitmoney, 2025-06-23:
AUIDO_MEM: Can't support stack on external memory due to ESP32 chip is 1
E (35060) AUDIO_THREAD: Error creating task Audio_forge
E (35066) AUDIO_ELEMENT: [Audio_forge] audio_thread_create failed
编译器都已经告诉你：是因为栈空间不够了，可能是线程的任务堆栈开小了