espressif/esp-adf #808: Shortage I2S in my use case

mike-2020, 2022-04-26:
I already have a camera which uses I2S for DMA.

I also need both audio input and output. 

Direct DAC output is not an option as the required two Pins are not exported on by board. So, I have to use the other I2S port for audio output.

It looks like this leaves me no option but have to choose audio input with ADC without I2S (without DMA).

Please share your recommendation. And question is: does ADF support ADC input without I2S?

HengYongChao, 2022-05-07:
Hi @mike-2020 

You can refer to the official development board hardware interface design, https://docs.espressif.com/projects/esp-adf/en/latest/design-guide/dev-boards/user-guide-esp32-s3-korvo-2.html

jason-mao, 2022-08-08:
@mike-2020 ADF does not support ADC without I2S.  Recommend use es8311 for you project. It's full-duplex for ADC and DAC with one I2S.

closed 2023-09-05 by jason-mao