espressif/esp-adf #811: examples\recorder\pipeline_wav_sdcard\main\record_wav_sdcard_example.c

kaituozhe, 2022-04-28:
I failed  to compile the example  record_wav_sdcard_example.c
D:/Cproject/ESP32Audio/Espressif/frameworks/esp-idf-v4.4.1/components/asio/asio/asio/src/asio.cpp
In file included from D:/Cproject/ESP32Audio/Espressif/frameworks/esp-idf-v4.4.1/components/asio/asio/asio/include/asio/error_code.hpp:25,
                 from D:/Cproject/ESP32Audio/Espressif/frameworks/esp-idf-v4.4.1/components/asio/asio/asio/include/asio/error.hpp:19,
                 from D:/Cproject/ESP32Audio/Espressif/frameworks/esp-idf-v4.4.1/components/asio/asio/asio/include/asio/impl/error.ipp:20,
                 from D:/Cproject/ESP32Audio/Espressif/frameworks/esp-idf-v4.4.1/components/asio/asio/asio/include/asio/impl/src.hpp:22,
                 from D:/Cproject/ESP32Audio/Espressif/frameworks/esp-idf-v4.4.1/components/asio/asio/asio/src/asio.cpp:11:
d:\cproject\esp32audio\espressif\frameworks\esp-idf-v4.4.1\tools\tools\xtensa-esp32-elf\esp-2021r2-patch3-8.4.0\xtensa-esp32-elf\xtensa-esp32-elf\include\c++\8.4.0\system_error:39:10: fatal error: bits/error_constants.h: No such file or directory
 #include <bits/error_constants.h>
          ^~~~~~~~~~~~~~~~~~~~~~~~
compilation terminated.

kaituozhe, 2022-04-28:
Copy bits/error_constants.h(from other directory,size 6k）， and put it in espressif\frameworks\esp-idf-v4.4.1\tools\tools\xtensa-esp32-elf\esp-2021r2-patch3-8.4.0\xtensa-esp32-elf\xtensa-esp32-elf\include\c++\8.4.0\bits

then it works well.  
It could be flashed to board.

HengYongChao, 2022-05-07:
Hi @kaituozhe 

I think you need reinstall you compile environment, please check the windows setup steps, https://docs.espressif.com/projects/esp-idf/en/latest/esp32/get-started/windows-setup.html

closed 2022-08-08 by jason-mao