espressif/esp-adf #812: ESP32-S3-KORVO-2 V3 lcd_jpeg example I2C error

cherishQAQ123, 2022-04-28:
Hi 
I compiled the lcd_ jepg example,but an error is reported after downloading and running

ESP-ROM:esp32s3-20210327
Build:Mar 27 2021
rst:0x1 (POWERON),boot:0x9 (SPI_FAST_FLASH_BOOT)
SPIWP:0xee
mode:DIO, clock div:1
load:0x3fcd0108,len:0x1660
load:0x403b6000,len:0xb7c
load:0x403ba000,len:0x2f70
entry 0x403b6248
[0;32mI (24) boot: ESP-IDF v4.4-329-g3c9657fa4e-dirty 2nd stage bootloader[0m
[0;32mI (25) boot: compile time 20:03:22[0m
[0;32mI (25) boot: chip revision: 0[0m
[0;32mI (28) boot.esp32s3: Boot SPI Speed : 80MHz[0m
[0;32mI (33) boot.esp32s3: SPI Mode       : DIO[0m
[0;32mI (38) boot.esp32s3: SPI Flash Size : 2MB[0m
[0;32mI (42) boot: Enabling RNG early entropy source...[0m
[0;32mI (48) boot: Partition Table:[0m
[0;32mI (51) boot: ## Label            Usage          Type ST Offset   Length[0m
[0;32mI (59) boot:  0 nvs              WiFi data        01 02 00009000 00006000[0m
[0;32mI (66) boot:  1 phy_init         RF data          01 01 0000f000 00001000[0m
[0;32mI (73) boot:  2 factory          factory app      00 00 00010000 00100000[0m
[0;32mI (81) boot: End of partition table[0m
[0;32mI (85) esp_image: segment 0: paddr=00010020 vaddr=3c030020 size=1f250h (127568) map[0m
[0;32mI (116) esp_image: segment 1: paddr=0002f278 vaddr=3fc94800 size=00da0h (  3488) load[0m
[0;32mI (118) esp_image: segment 2: paddr=00030020 vaddr=42000020 size=2103ch (135228) map[0m
[0;32mI (146) esp_image: segment 3: paddr=00051064 vaddr=3fc955a0 size=01a7ch (  6780) load[0m
[0;32mI (148) esp_image: segment 4: paddr=00052ae8 vaddr=40374000 size=107f4h ( 67572) load[0m
[0;32mI (167) esp_image: segment 5: paddr=000632e4 vaddr=50000000 size=00010h (    16) load[0m
[0;32mI (174) boot: Loaded app from partition at offset 0x10000[0m
[0;32mI (174) boot: Disabling RNG early entropy source...[0m
[0;32mI (187) cpu_start: Pro cpu up.[0m
[0;32mI (187) cpu_start: Starting app cpu, entry point is 0x40375360[0m
[0;32mI (0) cpu_start: App cpu up.[0m
[0;32mI (201) cpu_start: Pro cpu start user code[0m
[0;32mI (201) cpu_start: cpu freq: 240000000[0m
[0;32mI (201) cpu_start: Application information:[0m
[0;32mI (204) cpu_start: Project name:     lcd_jpeg[0m
[0;32mI (209) cpu_start: App version:      v2.4-4-geb2c5e0-dirty[0m
[0;32mI (215) cpu_start: Compile time:     Apr 28 2022 20:03:13[0m
[0;32mI (221) cpu_start: ELF file SHA256:  24ec365348456ea6...[0m
[0;32mI (227) cpu_start: ESP-IDF:          v4.4-329-g3c9657fa4e-dirty[0m
[0;32mI (234) heap_init: Initializing. RAM available for dynamic allocation:[0m
[0;32mI (241) heap_init: At 3FC98030 len 00047FD0 (287 KiB): D/IRAM[0m
[0;32mI (248) heap_init: At 3FCE0000 len 0000EE34 (59 KiB): STACK/DRAM[0m
[0;32mI (254) heap_init: At 3FCF0000 len 00008000 (32 KiB): DRAM[0m
[0;32mI (260) heap_init: At 600FE000 len 00002000 (8 KiB): RTCRAM[0m
[0;32mI (267) spi_flash: detected chip: generic[0m
[0;32mI (272) spi_flash: flash io: dio[0m
[0;33mW (275) spi_flash: Detected size(8192k) larger than the size in the binary image header(2048k). Using the size in the binary image header.[0m
[0;32mI (289) sleep: Configure to isolate all GPIO pins in sleep state[0m
[0;32mI (295) sleep: Enable automatic switching of GPIO sleep configuration[0m
[0;32mI (303) cpu_start: Starting scheduler on PRO CPU.[0m
[0;32mI (0) cpu_start: Starting scheduler on APP CPU.[0m
[0;31mE (363) TCA9554: IO expander device has not detected[0m
[0;31mE (383) I2C_BUS: /home/xh/share/02.esp32s3_display/esp-adf/components/esp_peripherals/driver/i2c_bus/i2c_bus.c:156 (i2c_bus_read_bytes):I2C Bus ReadReg Error[0m
[0;31mE (393) I2C_BUS: /home/xh/share/02.esp32s3_display/esp-adf/components/esp_peripherals/driver/i2c_bus/i2c_bus.c:101 (i2c_bus_write_bytes):I2C Bus WriteReg Error[0m
[0;31mE (413) I2C_BUS: /home/xh/share/02.esp32s3_display/esp-adf/components/esp_peripherals/driver/i2c_bus/i2c_bus.c:156 (i2c_bus_read_bytes):I2C Bus ReadReg Error[0m
[0;31mE (423) I2C_BUS: /home/xh/share/02.esp32s3_display/esp-adf/components/esp_peripherals/driver/i2c_bus/i2c_bus.c:101 (i2c_bus_write_bytes):I2C Bus WriteReg Error[0m
[0;31mE (443) I2C_BUS: /home/xh/share/02.esp32s3_display/esp-adf/components/esp_peripherals/driver/i2c_bus/i2c_bus.c:156 (i2c_bus_read_bytes):I2C Bus ReadReg Error[0m
[0;31mE (453) I2C_BUS: /home/xh/share/02.esp32s3_display/esp-adf/components/esp_peripherals/driver/i2c_bus/i2c_bus.c:101 (i2c_bus_write_bytes):I2C Bus WriteReg Error[0m
[0;31mE (473) I2C_BUS: /home/xh/share/02.esp32s3_display/esp-adf/components/esp_peripherals/driver/i2c_bus/i2c_bus.c:156 (i2c_bus_read_bytes):I2C Bus ReadReg Error[0m
[0;31mE (483) I2C_BUS: /home/xh/share/02.esp32s3_display/esp-adf/components/esp_peripherals/driver/i2c_bus/i2c_bus.c:101 (i2c_bus_write_bytes):I2C Bus WriteReg Error[0m
[0;31mE (503) I2C_BUS: /home/xh/share/02.esp32s3_display/esp-adf/components/esp_peripherals/driver/i2c_bus/i2c_bus.c:156 (i2c_bus_read_bytes):I2C Bus ReadReg Error[0m
[0;31mE (513) I2C_BUS: /home/xh/share/02.esp32s3_display/esp-adf/components/esp_peripherals/driver/i2c_bus/i2c_bus.c:101 (i2c_bus_write_bytes):I2C Bus WriteReg Error[0m
[0;31mE (543) I2C_BUS: /home/xh/share/02.esp32s3_display/esp-adf/components/esp_peripherals/driver/i2c_bus/i2c_bus.c:156 (i2c_bus_read_bytes):I2C Bus ReadReg Error[0m
[0;31mE (553) I2C_BUS: /home/xh/share/02.esp32s3_display/esp-adf/components/esp_peripherals/driver/i2c_bus/i2c_bus.c:101 (i2c_bus_write_bytes):I2C Bus WriteReg Error[0m
[0;32mI (563) gpio: GPIO[2]| InputEn: 0| OutputEn: 1| OpenDrain: 0| Pullup: 0| Pulldown: 0| Intr:0 [0m
[0;32mI (563) ImageDec: The image size is 156160 bytes, width:320, height:244[0m

HengYongChao, 2022-05-07:
Hi @cherishQAQ123 

This error means that the initialization of the audio hal layer is wrong, which is usually caused by the mismatch of the development board, usually a hardware problem. 

Please check whether the esp32_s3_korvo2_v3 development board is correctly selected in the menuconfig.

zx568073531, 2023-02-27:
Hi @HengYongChao 

我这边也有这个问题，用的 ESP32-S3-KORVO V2-V3.1版本硬件。设置HAL硬件为 ESP32-S3-KORVO V2。