espressif/esp-adf #814: ESP32-LyreT音频板子无法识别命令词问题

aithinker2222, 2022-05-10:
下面是我打印的串口log信息日志，跑的示例是speech_recognition/wwe    唤醒词是可以使用的 但是命令词是无法识别到 是否我操作方式不对
I (27) boot: ESP-IDF v4.4.1-117-g8c8f700c1d-dirty 2nd stage bootloader
I (27) boot: compile time 09:21:07
I (28) boot: chip revision: 3
I (32) boot_comm: chip revision: 3, min. bootloader chip revision: 0
I (39) qio_mode: Enabling default flash chip QIO
I (44) boot.esp32: SPI Speed      : 80MHz
I (49) boot.esp32: SPI Mode       : QIO
I (54) boot.esp32: SPI Flash Size : 4MB
I (58) boot: Enabling RNG early entropy source...
I (64) boot: Partition Table:
I (67) boot: ## Label            Usage          Type ST Offset   Length
I (74) boot:  0 nvs              WiFi data        01 02 00009000 00004000
I (82) boot:  1 otadata          OTA data         01 00 0000d000 00002000
I (89) boot:  2 phy_init         RF data          01 01 0000f000 00001000
I (97) boot:  3 ota_0            OTA app          00 10 00010000 00300000
I (104) boot:  4 flash_tone       Unknown data     01 27 00310000 00032000
I (112) boot: End of partition table
I (116) boot_comm: chip revision: 3, min. application chip revision: 0
I (123) esp_image: segment 0: paddr=00010020 vaddr=3f400020 size=178970h (1542512) map
I (557) esp_image: segment 1: paddr=00188998 vaddr=3ffb0000 size=02958h ( 10584) load
I (560) esp_image: segment 2: paddr=0018b2f8 vaddr=40080000 size=04d20h ( 19744) load
I (569) esp_image: segment 3: paddr=00190020 vaddr=400d0020 size=781bch (491964) map
I (707) esp_image: segment 4: paddr=002081e4 vaddr=40084d20 size=0ea30h ( 59952) load
I (726) esp_image: segment 5: paddr=00216c1c vaddr=50000000 size=00010h (    16) load
I (736) boot: Loaded app from partition at offset 0x10000
I (736) boot: Disabling RNG early entropy source...
I (749) psram: This chip is ESP32-D0WD
I (749) spiram: Found 64MBit SPI RAM device
I (749) spiram: SPI RAM mode: flash 80m sram 80m
I (752) spiram: PSRAM initialized, cache is in low/high (2-core) mode.
I (759) cpu_start: Pro cpu up.
I (763) cpu_start: Starting app cpu, entry point is 0x40081684
0x40081684: DPORT_REG_READ at /home/aithinker/esp/esp-adf/esp-idf/components/soc/esp32/include/soc/dport_access.h:79
 (inlined by) call_start_cpu1 at /home/aithinker/esp/esp-adf/esp-idf/components/esp_system/port/cpu_start.c:176

I (0) cpu_start: App cpu up.
I (1245) spiram: SPI SRAM memory test OK
I (1253) cpu_start: Pro cpu start user code
I (1253) cpu_start: cpu freq: 240000000
I (1253) cpu_start: Application information:
I (1256) cpu_start: Project name:     example_wwe
I (1261) cpu_start: App version:      v2.4-10-gd0f5bb6-dirty
I (1267) cpu_start: Compile time:     May 10 2022 09:51:30
I (1274) cpu_start: ELF file SHA256:  aaa8b724e4dad852...
I (1280) cpu_start: ESP-IDF:          v4.4.1-117-g8c8f700c1d-dirty
I (1287) heap_init: Initializing. RAM available for dynamic allocation:
I (1294) heap_init: At 3FFAE6E0 len 00001920 (6 KiB): DRAM
I (1300) heap_init: At 3FFB49C0 len 0002B640 (173 KiB): DRAM
I (1306) heap_init: At 3FFE0440 len 00003AE0 (14 KiB): D/IRAM
I (1313) heap_init: At 3FFE4350 len 0001BCB0 (111 KiB): D/IRAM
I (1319) heap_init: At 40093750 len 0000C8B0 (50 KiB): IRAM
I (1326) spiram: Adding pool of 4095K of external SPI memory to heap allocator
I (1334) spi_flash: detected chip: generic
I (1338) spi_flash: flash io: qio
W (1342) spi_flash: Detected size(8192k) larger than the size in the binary image header(4096k). Using the size in the binary image header.
I (1356) cpu_start: Starting scheduler on PRO CPU.
I (0) cpu_start: Starting scheduler on APP CPU.
I (1366) spiram: Reserving pool of 32K of internal memory for DMA/internal allocations
I (1376) gpio: GPIO[36]| InputEn: 1| OutputEn: 0| OpenDrain: 0| Pullup: 1| Pulldown: 0| Intr:3 
I (1386) gpio: GPIO[39]| InputEn: 1| OutputEn: 0| OpenDrain: 0| Pullup: 1| Pulldown: 0| Intr:3 
I (1396) gpio: GPIO[19]| InputEn: 1| OutputEn: 0| OpenDrain: 0| Pullup: 1| Pulldown: 0| Intr:3 
E (1416) gpio: gpio_install_isr_service(449): GPIO isr service already installed
W (1416) PERIPH_TOUCH: _touch_init
I (1426) gpio: GPIO[21]| InputEn: 0| OutputEn: 1| OpenDrain: 0| Pullup: 0| Pulldown: 0| Intr:0 
I (1426) ES8388_DRIVER: init,out:02, in:00
W (1436) AUDIO_BOARD: The board has already been initialized!

----------------------------- ESP Audio Platform -----------------------------
|                                                                            |
[... 75 lines cut ...]
I (260826) RECORDER_SR: MN dect quit
I (267216) wwe_example: rec_engine_cb - REC_EVENT_WAKEUP_START
W (267326) AUDIO_RECORDER: Not in speeching, return 0
W (267326) wwe_example: audio recorder read finished 0
I (268356) wwe_example: rec_engine_cb - REC_EVENT_VAD_START
W (268366) wwe_example: voice read begin
I (275616) RECORDER_SR: MN dect quit
I (277306) wwe_example: rec_engine_cb - REC_EVENT_VAD_STOP
W (277336) wwe_example: voice read stopped
I (278206) wwe_example: rec_engine_cb - REC_EVENT_WAKEUP_END
I (380826) wwe_example: rec_engine_cb - REC_EVENT_WAKEUP_START
I (381556) wwe_example: rec_engine_cb - REC_EVENT_VAD_START
W (381556) wwe_example: voice read begin
I (384526) wwe_example: rec_engine_cb - REC_EVENT_VAD_STOP
W (384536) wwe_example: voice read stopped
I (385026) wwe_example: rec_engine_cb - REC_EVENT_VAD_START
W (385026) wwe_example: voice read begin
I (389206) RECORDER_SR: MN dect quit
I (390226) wwe_example: rec_engine_cb - REC_EVENT_VAD_STOP
W (390246) wwe_example: voice read stopped
I (390726) wwe_example: rec_engine_cb - REC_EVENT_VAD_START
W (390736) wwe_example: voice read begin
I (392306) wwe_example: rec_engine_cb - REC_EVENT_VAD_STOP
W (392336) wwe_example: voice read stopped
I (393206) wwe_example: rec_engine_cb - REC_EVENT_WAKEUP_END

HengYongChao, 2022-05-11:
Hi @aithinker2222 

 我看到你的 log 中有 VAD 唤醒的事件。 新的 esp-sr 更新后，效果更好了，但是对算力要求更高了，esp32 在打开命令词模式下，会有 wdt 触发。

1. 现在回到你的问题， 你检查下  ` use multinet` 是否使能；

[image]

2. 再一个， 当前支持的免费的 命令词如下，检查下是否使用了正确的语音命令：

[image]

aithinker2222, 2022-05-11:
非常感激你的及时回复  我按照你给的配置设置好后可以成功识别到命令词了，问题点是我设置的wake word name hi lexin（wakenet5x2）导致识别率太慢  改成5x3后可以正常识别到了

closed 2022-08-08 by jason-mao