espressif/esp-adf #815: [Bug] i2s_alc_volume_set not working on esp-idf v4.4.1 except setting -64

JeyongOh-MAASASIA, 2022-05-10:
esp-idf v4.3.2 + esp-adf v2.4 -> i2s_alc_volume_set working
esp-idf v4.4.1 + esp-adf v2.4 -> i2s_alc_volume_set not working except -64 (which is mute.)

Setting i2s_alc_volume_set is not working except I set volume to -64.
When I set volume as -64, **sound is muted** which is expected behavior.
When I set volume as any value between -63 ~ 64, it will always speak max volume. When I get volume by i2s_alc_volume_get, the volume is set correctly as I set, but the sound is always max.

jason-mao, 2022-08-08:
@JeyongOh-MAASASIA How can I reproduce your issue? I have test i2s_alc_volume_set work well with v4.4.1.

jason-mao, 2023-09-05:
This topic has become inactive so I'm going to close the issue. Please test with latest ADF verison or reopen this if you have any questions or need any further assistance.

closed 2023-09-05 by jason-mao