espressif/esp-adf #816: VoIP issue, can not start SIP service, xTaskCreateRestrictedPinnedToCore

onelle, 2022-05-11:
Hello,

I have difficulties getting the VoIP example to work. I use:
-  IDF version v4.4-329-g3c9657fa4e
-  ADF version v2.4-10-gd0f5bb6
- esp32 LyraT Mini V1.2
-  VS Code on Win 64 
-  Asterisk server (communication between two phones works)

The following steps are completed successfully:
[1.0] Initialize peripherals management
[1.1] Initialize and start peripherals
[1.2] Create and start input key service
[ 1.3 ] Create display service instance
[ 2 ] Start codec chip
[ 3 ] Initialize tone player
[ 4 ] Create Wi-Fi service instance

BUT after LyraT connects to Wifi and tries to create sip service I keep getting the following error:

[image]

The same error still occurs after applying idf_v4.4_freertos.patch!

Any ideas what might be the problem or why it can not start SIP service?

Also, where do you find esp_sip.h (I cannot find it in espressif/esp-adf-libs) and is there a esp_sip.c as well?

HengYongChao, 2022-05-12:
Hi @onelle 

Please apply `https://github.com/espressif/esp-adf/blob/master/idf_patches/idf_v4.4_freertos.patch` to your IDF.

onelle, 2022-05-14:
Thank you for your reply!
But as I said before: **I did apply this patch but the error is still there**.
I even checked task.h and tasks.c to make sure the changes were applied.

Also, where do you find esp_sip.h and is there a esp_sip.c as well?

HengYongChao, 2022-05-16:
Hi @onelle 

Please check that your `idf_v4.4_freertos.patch` has been applied correctly.

Only from the analysis of the log, it is because the patch was not applied correctly, and then some tasks allocated on psram were not created. then the error happened.

onelle, 2022-05-16:
Hi @HengYongChao 

How do I correctly apply the patch? 
I did: 
`cd %IDF_PATH%`
`git apply %ADF_PATH%/idf_patches/idf_v4.4_freertos.patch`
it worked without error and in tasks.c and task.h it looks like the changes were applied correctly

These are the files after application:
[patched_files.zip](https://github.com/espressif/esp-adf/files/8700070/patched_files.zip)

HengYongChao, 2022-05-18:
> Hi @HengYongChao
>
> How do I correctly apply the patch? I did: `cd %IDF_PATH%` `git apply %ADF_PATH%/idf_patches/idf_v4.4_freertos.patch` it worked without error and in tasks.c and task.h it looks [...]

Yes, when git no report any message, it means successed. you can use that idf to compile your project.

Actualy, we do have the document here, https://github.com/espressif/esp-adf/blob/master/idf_patches/README.md.

onelle, 2022-05-18:
Good, thanks @HengYongChao 
Do you have any idea what the problem might be then? I still get this error.

HengYongChao, 2022-05-18:
HI @onelle 

Generally speaking, when you apply this patch in a clean idf repository, it will look like this. 

[image]

You first ensure that your patch is successfully applied. then, delete build, and then compile again.

onelle, 2022-05-20:
I couldn't get the example to work with with versions 4.4.1 or 4.3.2 but with esp-idf-v4.2.3 (downloaded with espressif-IDE) and corresponding patch it works just fine

jason-mao, 2022-08-08:
@onelle How about your issue now? Is it still can't work with version 4.4.1 or 4.3.2?

lenabux, 2022-08-09:
Hello @jason-mao !
It didn't work last time I tried and there were no changes since, so I assume it still doesn't. 
I haven't had the chance to try again since your message though.

jason-mao, 2022-08-09:
@lenabux Thank you for your info. We will try to reproduce your issue.

jason-mao, 2022-08-10:
@lenabux We are test and  work well as your ENV. So I think that you may not set the IDF_PATH correctly. 

I'm going to close issue. Please reopen this if you have any questions or need any further assistance.

closed 2022-08-10 by jason-mao