espressif/esp-adf #817: how to a2dp and spiffs mp3 play mixer

brilliantWorks, 2022-05-12:
I want to bt a2dp and play spiffs mp3 at the same time. 
Is there an example code?
Please advise me how to do it.

HengYongChao, 2022-05-12:
Hi @brilliantWorks 

There is no ready-made demo, you can refer to the [downmix_pipeline](https://github.com/espressif/esp-adf/tree/master/examples/advanced_examples/downmix_pipeline) for audio data processing.

[pipeline_a2dp_sink_stream](https://github.com/espressif/esp-adf/tree/master/examples/player/pipeline_a2dp_sink_stream) and [pipeline_spiffs_mp3](https://github.com/espressif/esp-adf/tree/master/examples/player/pipeline_spiffs_mp3) are the inputs  and  then the data flow should be as follows:
```c
          aadp_source ---> aadp_sink_stream ---> resample --->
                                                              |--->  down-mix ---> i2s_stream_writer ---> codec_chip ---> speaker
[flash] ---> spiffs_stream ---> mp3_decoder ---> resample --->
```

brilliantWorks, 2022-06-16:
Hi @HengYongChao 

I saw the downmix_pipeline example.

In downmix_pipeline example, raw_stream was connected before down-mix.

What exactly is raw_stream used for?

Is raw_stream required to use downmix?

I would appreciate it if you could explain the purpose of raw_stream.

HengYongChao, 2022-06-17:
Hi @brilliantWorks 

Raw Stream is just like a ringbuf, It is used to link the previous or next element, and it is used as an interface to output or input data.

closed 2023-09-05 by jason-mao