espressif/esp-adf #819: Dueros demo error: AUDIO_EVT: There is no space in external queue

lanistor, 2022-05-12:
I ran Dueros' Demo, and there was no response to the voice interaction step. When I manually clicked the Rec button, it would report warn: `W (661493) AUDIO_EVT: There is no space in external queue`.

Development board: ESP32-Lyrat-Mini V1.2
ADF version: `release/v2.4`
IDF version: `release/v4.4`

lanistor, 2022-05-12:
Full log:
```
--- idf_monitor on /dev/cu.SLAB_USBtoUART 115200 ---
--- Quit: Ctrl+] | Menu: Ctrl+T | Help: Ctrl+T followed by Ctrl+H ---
I (10) boot: ESP-IDF v4.4.1-91-g36823b45cc-dirty 2nd stage bootloader
I (10) boot: compile time 01:00:42
I (11) boot: chip revision: 3
I (14) boot_comm: chip revision: 3, min. bootloader chip revision: 0
I (21) qio_mode: Enabling default flash chip QIO
I (26) boot.esp32: SPI Speed      : 80MHz
I (31) boot.esp32: SPI Mode       : QIO
I (35) boot.esp32: SPI Flash Size : 4MB
I (40) boot: Enabling RNG early entropy source...
I (45) boot: Partition Table:
I (49) boot: ## Label            Usage          Type ST Offset   Length
I (56) boot:  0 nvs              WiFi data        01 02 00009000 00006000
I (64) boot:  1 phy_init         RF data          01 01 0000f000 00001000
I (71) boot:  2 factory          factory app      00 00 00010000 00310000
I (79) boot: End of partition table
I (83) boot_comm: chip revision: 3, min. application chip revision: 0
I (90) esp_image: segment 0: paddr=00010020 vaddr=3f400020 size=6b7f8h (440312) map
I (220) esp_image: segment 1: paddr=0007b820 vaddr=3ffb0000 size=041f0h ( 16880) load
I (225) esp_image: segment 2: paddr=0007fa18 vaddr=40080000 size=00600h (  1536) load
I (226) esp_image: segment 3: paddr=00080020 vaddr=400d0020 size=14c9f8h (1362424) map
I (610) esp_image: segment 4: paddr=001cca20 vaddr=40080600 size=16cb4h ( 93364) load
I (640) esp_image: segment 5: paddr=001e36dc vaddr=50000000 size=00010h (    16) load
I (652) boot: Loaded app from partition at offset 0x10000
I (652) boot: Disabling RNG early entropy source...
I (663) psram: This chip is ESP32-D0WD
I (664) spiram: Found 64MBit SPI RAM device
I (664) spiram: SPI RAM mode: flash 80m sram 80m
I (669) spiram: PSRAM initialized, cache is in low/high (2-core) mode.
I (676) cpu_start: Pro cpu up.
I (680) cpu_start: Starting app cpu, entry point is 0x4008163c
0x4008163c: call_start_cpu1 at /Users/Shared/SDK/esp/esp-idf/components/esp_system/port/cpu_start.c:160

I (0) cpu_start: App cpu up.
I (1161) spiram: SPI SRAM memory test OK
I (1170) cpu_start: Pro cpu start user code
I (1170) cpu_start: cpu freq: 240000000
I (1170) cpu_start: Application information:
I (1173) cpu_start: Project name:     esp_dueros
I (1178) cpu_start: App version:      v2.3-277-g4f84517-dirty
I (1184) cpu_start: Compile time:     May 13 2022 01:01:29
I (1191) cpu_start: ELF file SHA256:  470c97f7785df1c6...
I (1197) cpu_start: ESP-IDF:          v4.4.1-91-g36823b45cc-dirty
I (1204) heap_init: Initializing. RAM available for dynamic allocation:
I (1211) heap_init: At 3FFAE6E0 len 00001920 (6 KiB): DRAM
I (1217) heap_init: At 3FFB7458 len 00028BA8 (162 KiB): DRAM
I (1223) heap_init: At 3FFE0440 len 00003AE0 (14 KiB): D/IRAM
I (1230) heap_init: At 3FFE4350 len 0001BCB0 (111 KiB): D/IRAM
I (1236) heap_init: At 400972B4 len 00008D4C (35 KiB): IRAM
I (1243) spiram: Adding pool of 4084K of external SPI memory to heap allocator
I (1251) spi_flash: detected chip: gd
I (1255) spi_flash: flash io: qio
W (1259) spi_flash: Detected size(8192k) larger than the size in the binary image header(4096k). Using the size in the binary image header.
I (1273) cpu_start: Starting scheduler on PRO CPU.
I (0) cpu_start: Starting scheduler on APP CPU.
I (1288) spiram: Reserving pool of 32K of internal memory for DMA/internal allocations
I (1327) DUEROS: ADF version is v2.3-277-g4f84517-dirty
I (1333) DRV8311: ES8311 in Slave mode
I (1344) gpio: GPIO[21]| InputEn: 0| OutputEn: 1| OpenDrain: 0| Pullup: 0| Pulldown: 0| Intr:0 
W (1389) I2C_BUS: i2c_bus_create:57: I2C bus has been already created, [port:0]
I (1392) gpio: GPIO[13]| InputEn: 0| OutputEn: 1| OpenDrain: 0| Pullup: 0| Pulldown: 0| Intr:0 
I (1402) SDCARD: Using 1-line SD mode, 4-line SD mode,  base path=/sdcard
I (1474) SDCARD: CID name ASTC!

E (1895) DISPATCHER: exe first list: 0x0
I (1896) DISPATCHER: dispatcher_event_task is running...
I (1899) AUDIO_THREAD: The wifi_serv task allocate stack on external memory
[... 131 lines cut ...]
W (7059,tid:3ffdebb8) lightduer_engine.c( 697):data cache has not sent, pending..., dcache_len:2, qcache_len:4
W (7070,tid:3ffdebb8) lightduer_engine.c( 697):data cache has not sent, pending..., dcache_len:2, qcache_len:3
W (7080,tid:3ffdebb8) lightduer_engine.c( 697):data cache has not sent, pending..., dcache_len:2, qcache_len:2
W (7089,tid:3ffdebb8) lightduer_engine.c( 697):data cache has not sent, pending..., dcache_len:2, qcache_len:1
I (7894,tid:3ffdebb8) lightduer_dcs_router.c( 460):Directive name: ThrowException
E (7894,tid:3ffdebb8) lightduer_dcs_system.c( 314):error code: INVALID_REQUEST_EXCEPTION
E (7898,tid:3ffdebb8) lightduer_dcs_system.c( 315):error: didp product not exsists
I (7910,tid:3ffdebb8) lightduer_dcs_router.c( 460):Directive name: ThrowException
E (7914,tid:3ffdebb8) lightduer_dcs_system.c( 314):error code: INVALID_REQUEST_EXCEPTION
E (7921,tid:3ffdebb8) lightduer_dcs_system.c( 315):error: didp product not exsists
I (7935,tid:3ffdebb8) lightduer_dcs_router.c( 460):Directive name: DialogueFinished
W (7938,tid:3ffdebb8) lightduer_engine.c( 697):data cache has not sent, pending..., dcache_len:1, qcache_len:2
I (7955,tid:3ffdebb8) lightduer_dcs_router.c( 460):Directive name: DialogueFinished
W (7961,tid:3ffdebb8) lightduer_engine.c( 697):data cache has not sent, pending..., dcache_len:2, qcache_len:2
W (7975,tid:3ffdebb8) lightduer_engine.c( 697):data cache has not sent, pending..., dcache_len:2, qcache_len:1
W (33676) AUDIO_EVT: There is no space in external queue
W (36667) AUDIO_EVT: There is no space in external queue
W (38071) AUDIO_EVT: There is no space in external queue
W (43347) AUDIO_EVT: There is no space in external queue
I (43468) DUEROS: AUDIO_USER_KEY_VOL_UP [69]
W (43468) AUDIO_EVT: There is no space in external queue
W (47016) AUDIO_EVT: There is no space in external queue
W (48699) AUDIO_EVT: There is no space in external queue

```

HengYongChao, 2022-05-16:
Hi @lanistor 

> W (48699) AUDIO_EVT: There is no space in external queue

```c
esp_err_t audio_event_iface_sendout(audio_event_iface_handle_t evt, audio_event_iface_msg_t *msg)
{
    if (evt->external_queue) {
        if (xQueueSend(evt->external_queue, (void *)msg, 0) != pdPASS) {
            ESP_LOGW(TAG, "There is no space in external queue");
            return ESP_FAIL;
        }
    }
    return ESP_OK;
}
```

https://github.com/espressif/esp-adf/blob/master/components/audio_pipeline/audio_event_iface.c#L273

This means that your event receiving queue is full, you can see here [AUDIO_EVENT_IFACE_DEFAULT_CFG](https://github.com/espressif/esp-adf/blob/master/components/audio_pipeline/include/audio_event_iface.h#L67). Even if you set it bigger, it will be stuffed again. The most important thing is to clear the event receiving queue.

ayavilevich, 2022-12-30:
Happened to me when setup of the listener was missing.
I had to add

```
audio_event_iface_set_listener(esp_periph_set_get_event_iface(set), evt);
```