espressif/esp-adf #825: Failed to Build Example Codes for ESP32-S3-KORVO-2-V3

noyelseth, 2022-05-21:
Hi,

I compiled some examples to test and understand the code flow to use ESP32-S3-KORVO-2-V3.

I'm using **ESP-IDF v4.4.1 [Stable]** & the **ESP-ADF master repo**.

But when I try to build BT enable example **`pipeline_bt_sink, pipeline_hfp_stream & pipeline_bt_source`** that time I get an error of **`fatal error: esp32s2beta/rom/crc.h: No such file or directory**. I attached an image with this issue.

[image]

But in ESP-ADF as individual **`crc.h`** for every ESP32 SOC.

[image]

But If I change the error block in **`"D:/espressif-projects/esp/esp-adf/components/esp_peripherals/lib/blufi/blufi_security.c"`** with this
```c
#ifdef CONFIG_IDF_TARGET_ESP32
#include "esp32/rom/crc.h"
#elif CONFIG_IDF_TARGET_ESP32S2
#include "esp32s2/rom/crc.h"
#elif CONFIG_IDF_TARGET_ESP32S3
#include "esp32s3/rom/crc.h"
#endif // CONFIG_IDF_TARGET_ESP32
```

Then I get same problem in **`D:/espressif-projects/esp/esp-adf/components/wifi_service/blufi_config/blufi_security.c:45:10: fatal error: esp32s2beta/rom/crc.h: No such file or directory`**
after fix the same error into this file then I get this error **`undefined reference`**. and I did not get any way to resolve

```
Building in: D:\espressif-projects\eclipse-adf-workspace\pipeline_bt_sink\build
cmake --build . -- -v
[1/9] cmd.exe /C "cd /D D:\espressif-projects\eclipse-adf-workspace\pipeline_bt_sink\build\bootloader && C:\Users\noyelseth\.espressif\tools\cmake\3.20.3\bin\cmake.exe --build ."
[1/1] cmd.exe /C "cd /D D:\espressif-projects\eclipse-adf-workspace\pipeline_bt_sink\build\bootloader\esp-idf\esptool_py && python D:/espressif-projects/esp/esp-idf/components/partition_table/check_sizes.py --offset 0x8000 bootloader 0x0 D:/espressif-projects/eclipse-adf-workspace/pipeline_bt_sink/build/bootloader/bootloader.bin"
Bootloader binary size 0x5190 bytes. 0x2e70 bytes (36%) free.
[2/7] ccache C:\Users\noyelseth\.espressif\tools\xtensa-esp32s3-elf\esp-2021r2-patch3-8.4.0\xtensa-esp32s3-elf\bin\xtensa-esp32s3-elf-gcc.exe -DADF_VER=\"\" -DMBEDTLS_CONFIG_FILE=\"mbedtls/esp_config.h\" -Iconfig -ID:/espressif-projects/esp/esp-adf/components/wifi_service/include -ID:/espressif-projects/esp/esp-adf/components/wifi_service/inc -ID:/espressif-projects/esp/esp-idf/components/newlib/platform_include -ID:/espressif-projects/esp/esp-idf/components/freertos/include -ID:/espressif-projects/esp/esp-idf/components/freertos/include/esp_additions/freertos -ID:/espressif-projects/esp/esp-idf/components/freertos/port/xtensa/include -ID:/espressif-projects/esp/esp-idf/components/freertos/include/esp_additions -ID:/espressif-projects/esp/esp-idf/components/esp_hw_support/include -ID:/espressif-projects/esp/esp-idf/components/esp_hw_support/include/soc -ID:/espressif-projects/esp/esp-idf/components/esp_hw_support/include/soc/esp32s3 -ID:/espressif-projects/esp/esp-idf/components/esp_hw_support/port/esp32s3/. -ID:/espressif-projects/esp/esp-idf/components/esp_hw_support/port/esp32s3/private_include -ID:/espressif-projects/esp/esp-idf/components/heap/include -ID:/espressif-projects/esp/esp-idf/components/log/include -ID:/espressif-projects/esp/esp-idf/components/lwip/include/apps -ID:/espressif-projects/esp/esp-idf/components/lwip/include/apps/sntp -ID:/espressif-projects/esp/esp-idf/components/lwip/lwip/src/include -ID:/espressif-projects/esp/esp-idf/components/lwip/port/esp32/include -ID:/espressif-projects/esp/esp-idf/components/lwip/port/esp32/include/arch -ID:/espressif-projects/esp/esp-idf/components/soc/include -ID:/espressif-projects/esp/esp-idf/components/soc/esp32s3/. -ID:/espressif-projects/esp/esp-idf/components/soc/esp32s3/include -ID:/espressif-projects/esp/esp-idf/components/hal/esp32s3/include -ID:/espressif-projects/esp/esp-idf/components/hal/include -ID:/espressif-projects/esp/esp-idf/components/hal/platform_port/include -ID:/espressif-projects/esp/esp-idf/components/esp_rom/include -ID:/espressif-projects/esp/esp-idf/components/esp_rom/include/esp32s3 -ID:/espressif-projects/esp/esp-idf/components/esp_rom/esp32s3 -ID:/espressif-projects/esp/esp-idf/components/esp_common/include -ID:/espressif-projects/esp/esp-idf/components/esp_system/include -ID:/espressif-projects/esp/esp-idf/components/esp_system/port/soc -ID:/espressif-projects/esp/esp-idf/components/esp_system/port/public_compat -ID:/espressif-projects/esp/esp-idf/components/xtensa/include -ID:/espressif-projects/esp/esp-idf/components/xtensa/esp32s3/include -ID:/espressif-projects/esp/esp-idf/components/driver/include -ID:/espressif-projects/esp/esp-idf/components/driver/esp32s3/include -ID:/espressif-projects/esp/esp-idf/components/esp_pm/include -ID:/espressif-projects/esp/esp-idf/components/esp_ringbuf/include -ID:/espressif-projects/esp/esp-idf/components/efuse/include -ID:/espressif-projects/esp/esp-idf/components/efuse/esp32s3/include -ID:/espressif-projects/esp/esp-idf/components/vfs/include -ID:/espressif-projects/esp/esp-idf/components/esp_wifi/include -ID:/espressif-projects/esp/esp-idf/components/esp_event/include -ID:/espressif-projects/esp/esp-idf/components/esp_netif/include -ID:/espressif-projects/esp/esp-idf/components/esp_eth/include -ID:/espressif-projects/esp/esp-idf/components/tcpip_adapter/include -ID:/espressif-projects/esp/esp-idf/components/esp_phy/include -ID:/espressif-projects/esp/esp-idf/components/esp_phy/esp32s3/include -ID:/espressif-projects/esp/esp-idf/components/esp_ipc/include -ID:/espressif-projects/esp/esp-idf/components/app_trace/include -ID:/espressif-projects/esp/esp-idf/components/esp_timer/include -ID:/espressif-projects/esp/esp-idf/components/bt/common/osi/include -ID:/espressif-projects/esp/esp-idf/components/bt/include/esp32s3/include -ID:/espressif-projects/esp/esp-idf/components/bt/common/api/include/api -ID:/espressif-projects/esp/esp-idf/components/bt/common/btc/profile/esp/blufi/include -ID:/espressif-projects/esp/esp-idf/components/bt/common/btc/profile/esp/include -ID:/espressif-projects/esp/esp-idf/components/bt/host/bluedroid/api/include/api -ID:/espressif-projects/esp/esp-idf/components/nvs_flash/include -ID:/espressif-projects/esp/esp-idf/components/spi_flash/include -ID:/espressif-projects/esp/esp-idf/components/mbedtls/port/include -ID:/espressif-projects/esp/esp-idf/components/mbedtls/mbedtls/include -ID:/espressif-projects/esp/esp-idf/components/mbedtls/esp_crt_bundle/include -ID:/espressif-projects/esp/esp-adf/components/audio_sal/include -ID:/espressif-projects/esp/esp-adf/components/esp_dispatcher/include -ID:/espressif-projects/esp/esp-adf/components/esp_actions/include -mlongcalls -ffunction-sections -fdata-sections -Wall -Werror=all -Wno-error=unused-function -Wno-error=unused-variable -Wno-error=deprecated-declarations -Wextra -Wno-unused-parameter -Wno-sign-compare -ggdb -Og -fmacro-prefix-map=D:/espressif-projects/eclipse-adf-workspace/pipeline_bt_sink=. -fmacro-prefix-map=D:/espressif-projects/esp/esp-idf=IDF -fstrict-volatile-bitfields -Wno-error=unused-but-set-variable -fno-jump-tables -fno-tree-switch-conversion -std=gnu99 -Wno-old-style-declaration -D_GNU_SOURCE -DIDF_VER=\"v4.4.1\" -DESP_PLATFORM -D_POSIX_READER_WRITER_LOCKS -MD -MT esp-idf/wifi_service/CMakeFiles/__idf_wifi_service.dir/blufi_config/blufi_security.c.obj -MF esp-idf\wifi_service\CMakeFiles\__idf_wifi_service.dir\blufi_config\blufi_security.c.obj.d -o esp-idf/wifi_service/CMakeFiles/__idf_wifi_service.dir/blufi_config/blufi_security.c.obj -c D:/espressif-projects/esp/esp-adf/components/wifi_service/blufi_config/blufi_security.c
[3/7] cmd.exe /C "cd . && C:\Users\noyelseth\.espressif\tools\cmake\3.20.3\bin\cmake.exe -E rm -f esp-idf\wifi_service\libwifi_service.a && C:\Users\noyelseth\.espressif\tools\xtensa-esp32s3-elf\esp-2021r2-patch3-8.4.0\xtensa-esp32s3-elf\bin\xtensa-esp32s3-elf-ar.exe qc esp-idf\wifi_service\libwifi_service.a  esp-idf/wifi_service/CMakeFiles/__idf_wifi_service.dir/src/esp_wifi_setting.c.obj esp-idf/wifi_service/CMakeFiles/__idf_wifi_service.dir/src/wifi_service.c.obj esp-idf/wifi_service/CMakeFiles/__idf_wifi_service.dir/src/wifi_ssid_manager.c.obj esp-idf/wifi_service/CMakeFiles/__idf_wifi_service.dir/smart_config/smart_config.c.obj esp-idf/wifi_service/CMakeFiles/__idf_wifi_service.dir/blufi_config/blufi_config.c.obj esp-idf/wifi_service/CMakeFiles/__idf_wifi_service.dir/blufi_config/blufi_security.c.obj esp-idf/wifi_service/CMakeFiles/__idf_wifi_service.dir/airkiss_config/airkiss_config.c.obj && C:\Users\noyelseth\.espressif\tools\xtensa-esp32s3-elf\esp-2021r2-patch3-8.4.0\xtensa-esp32s3-e [... cut ...]
      .bss  size:   16528 bytes

Used Flash size :  702319 bytes

      .text     :  547547 bytes

      .rodata   :  154516 bytes

Total image size:  792810 bytes (.bin may be padded larger)

C:\Users\noyelseth\.espressif\python_env\idf4.4_py3.8_env\Scripts\python.exe D:\espressif-projects\esp\esp-idf/components/partition_table/gen_esp32part.py D:/espressif-projects/eclipse-adf-workspace/pipeline_bt_sink/build/partition_table/partition-table.bin
Total time taken to build the project: 20,769 ms
```

HengYongChao, 2022-05-24:
Hi @noyelseth 

Are you working on using classic bluetooth on your  ESP32-S3-KORVO-2-V3 board? 

esp32s3 does not support the classic bluetooth function, it supports ble 5.0 and WiFi.

FYI： https://www.espressif.com/en/products/socs/esp32-s3

closed 2022-08-08 by jason-mao

Baavro, 2024-09-18:
How did you resolve the error. Facing the same issue changed the config to use ble5.0 but still same error