espressif/esp-adf #826: hardcoded sdcard spi pins

tacrazymage, 2022-05-22:
## The problem
the SPI pin numbers hardcoded in "./components/esp_peripherals/lib/sdcard/sdcard.c" as it shown below.
> #define PIN_NUM_MISO 2
#define PIN_NUM_MOSI 15
#define PIN_NUM_CLK  14
#define PIN_NUM_CS   13

should we make it configurable using Kconfig file?

tacrazymage, 2022-05-23:
also the sdcard_mount function not working when the mode==SD_MODE_SPI (runtime crash)

HengYongChao, 2022-05-24:
Hi @tacrazymage 

The reason is that these sdcard pins are fixed on the official development board. 

I don't know your board type, I have tested to support SD_MODE_SPI mode on official board.

danny2jenny, 2022-05-27:
Also, in sdcard_mount function , spi mode code is not working.

spi port is not initialized，and should use esp_vfs_fat_sdspi_mount but not esp_vfs_fat_sdmmc_mount to mount.

tacrazymage, 2022-05-29:
> you need to modify the initialization function `audio_board_sdcard_init` in your board hal.

as far as I know (and now I test it again) #define keyword is preprocessor and preprocessors instructs the compiler to do required pre-processing before the actual compilation (and also before linking)

so there is *no way* defining pin numbers in board hal overwrites ones defined here: ./components/esp_peripherals/lib/sdcard/sdcard.c

tacrazymage, 2022-05-29:
if nobody assigned to this, I can make a PR (fixing #define and esp_vfs_fat_sdmmc_mount)
just let me know

tacrazymage, 2022-05-30:
@danny2jenny did you managed to play mp3 from sdcard in spi mode?
I correct the sdcard_mount function code but when I run "pipeline play sdcard music" example, it panics due to dead lock ocured in spi master

HengYongChao, 2022-06-08:
> @HengYongChao if nobody assigned to this, I can make a PR (fixing #define and esp_vfs_fat_sdmmc_mount) just let me know

Yes please, if you can give the PR.

and-elf, 2022-09-17:
This question/feature request keeps popping up, and should be fixed. The module code should not be board related, and as such, is a bug. Why was this PR closed?

closed 2023-07-07 by espressif-bot