espressif/esp-adf #827: 关于pipeline_aws_polly产生的错误

HongYan12345, 2022-05-23:
[image]
[image]
我在运行examples里面的examples/cloud_services/pipeline_aws_polly_mp3出现错误
我没有修改例子代码里面的任何东西，我在menuconfig里面设置了wifi密码和aws账号和密码，最后运行卡在了这里，这个错误是因为什么引起的呢？代码里面有哪些地方还需要改的吗？

HengYongChao, 2022-05-24:
Hi @HongYan12345 

HTTP 返回的错误码是 403（禁止访问），检查下你的 aws账号和密码是否正确。

第二张图是退出当前例程，WiFi 资源被 destroy 掉了。

HongYan12345, 2022-05-24:
> Hi @HongYan12345
> 
> HTTP 返回的错误码是 403（禁止访问），检查下你的 aws账号和密码是否正确。
> 
> 第二张图是退出当前例程，WiFi 资源被 destroy 掉了。

我是第一次使用aws，昨天刚创建出来，我不知道是不是直接用我注册的邮箱和密码放到代码里面，还是说要在aws网站里重新申请个密钥什么的吗？

HengYongChao, 2022-05-31:
> > Hi @HongYan12345
> > HTTP 返回的错误码是 403（禁止访问），检查下你的 aws账号和密码是否正确。
> > 第二张图是退出当前例程，WiFi 资源被 destroy 掉了。
> 
> 我是第一次使用aws，昨天刚创建出来，我不知道是不是直接用我注册的邮箱和密码放到代码里面，还是说要在aws网站里重新申请个密钥什么的吗？

把申请的密钥放在 `menuconfig` 配置里， [README.md](https://github.com/espressif/esp-adf/blob/master/examples/cloud_services/pipeline_aws_polly_mp3/README_CN.md) 文档已经提供了步骤，照着操作。

closed 2023-09-05 by jason-mao