espressif/esp-adf #829: 使用esp_audio_play()时一直出现Task watchdog got triggered

AdonisXie, 2022-05-25:
我在使用esp_audio_play()播放网络HTTP音乐时，一直出现喂不了狗的情况，
只要在放歌，就会一直出现，虽然能够放歌，但是我感觉是esp32在“坚持”着放歌，
放出来的歌也是会小卡，而且有杂音。
不放歌的时候就没什么事。
然后下面的是看门狗触发时打印的：【看门狗触发提示】 和 【CPU 1 回溯】

几次看门狗触发的提示：
```
E (448650) task_wdt: Task watchdog got triggered. The following tasks did not reset the watchdog in time:
E (448650) task_wdt:  - IDLE (CPU 1)
E (448650) task_wdt: Tasks currently running:
E (448650) task_wdt: CPU 0: afe_mase
E (448650) task_wdt: CPU 1: DEC_auto
E (448650) task_wdt: Print CPU 0 (current core) backtrace

E (453694) task_wdt: Task watchdog got triggered. The following tasks did not reset the watchdog in time:
E (453694) task_wdt:  - IDLE (CPU 1)
E (453694) task_wdt: Tasks currently running:
E (453694) task_wdt: CPU 0: afe_mase
E (453694) task_wdt: CPU 1: Audio_forge
E (453694) task_wdt: Print CPU 0 (current core) backtrace

E (386557) task_wdt: Task watchdog got triggered. The following tasks did not reset the watchdog in time:
E (386557) task_wdt:  - IDLE (CPU 1)
E (386557) task_wdt: Tasks currently running:
E (386557) task_wdt: CPU 0: OUT_iis
E (386557) task_wdt: CPU 1: IDLE
E (386557) task_wdt: Print CPU 0 (current core) backtrace

** CPU1是IDLE的情况下，很少出现，基本是CPU 1: Audio_forge 和 CPU 1: DEC_auto这两种情况**

```

打印的CPU 1 回溯如下所示，每次看门狗触发时，内容都是一样的
```

E (359957) task_wdt: Print CPU 1 backtrace

0x40168b3e: task_wdt_isr at /home/aaa/esp/esp-adf/esp-idf/components/esp_system/task_wdt.c:183 (discriminator 3)

0x40084bad: _xt_lowint1 at /home/aaa/esp/esp-adf/esp-idf/components/freertos/port/xtensa/xtensa_vectors.S:1111

0x4013b41b: fa_decimate_process_dual_dec at /builds/adf/esp-adf-libs-source/esp_processing/esp-resample/src_fa/src/fa_resample.c:877
 (inlined by) fa_resample_process at /builds/adf/esp-adf-libs-source/esp_processing/esp-resample/src_fa/src/fa_resample.c:2242

0x40139f7e: esp_resample_run at /builds/adf/esp-adf-libs-source/esp_processing/esp-resample/src/esp_resample.c:23

0x400daeb1: rsp_filter_process at /home/aaa/esp/esp-adf/components/esp-adf-libs/esp_codec/filter_resample.c:149

0x400dbc5e: audio_element_process_running at /home/aaa/esp/esp-adf/components/audio_pipeline/audio_element.c:336
 (inlined by) audio_element_task at /home/aaa/esp/esp-adf/components/audio_pipeline/audio_element.c:483

```

我不清楚是为什么，是因为内存不够还是CPU太过繁忙，我想请问一下这是什么情况？
我能在哪些方面去改善我的代码？
希望能够得到帮助！谢谢！

IDF: V4.4

HengYongChao, 2022-06-02:
Hi @AdonisXie 

 Task WDT 触发了， 检查代码有没有一直 while(1) 死循环运行的代码。 一般出现这个情况是，某个 task  一直占据 cpu， 其他 task 得不到执行， 触发此问题。

AdonisXie, 2022-06-03:
好的谢谢

closed 2022-08-08 by jason-mao