espressif/esp-adf #830: ESP32-A1S from Ai Thinker is Not Working

shridattdudhat, 2022-05-25:
Hi, I have two ESP32-A1S Audio Kit, We want to use as Intercom between two kits,

We tried this kit with default ADF example by changing the I2C and I2S pins suggested by the Ai Thinker. Listed below:
`esp_err_t get_i2c_pins(i2c_port_t port, i2c_config_t *i2c_config)
{
    AUDIO_NULL_CHECK(TAG_AI, i2c_config, return ESP_FAIL);
    if (port == I2C_NUM_0 || port == I2C_NUM_1) {
        i2c_config->sda_io_num = GPIO_NUM_33; // GPIO_NUM_18
        i2c_config->scl_io_num = GPIO_NUM_32; // GPIO_NUM_23
    } else {
        i2c_config->sda_io_num = -1;
        i2c_config->scl_io_num = -1;
        KIT_LOGE( "i2c port %d is not supported", port);
        return ESP_FAIL;
    }

    KIT_LOGI("i2c port: %d", port);
    KIT_LOGI("i2c sda: %d", i2c_config->sda_io_num);
    KIT_LOGI("i2c scl: %d", i2c_config->scl_io_num);

    return ESP_OK;
}

esp_err_t get_i2s_pins(i2s_port_t port, i2s_pin_config_t *i2s_config)
{
    AUDIO_NULL_CHECK(TAG_AI, i2s_config, return ESP_FAIL);
    if (port == I2S_NUM_0 || port == I2S_NUM_1) {
        i2s_config->bck_io_num = GPIO_NUM_27;
        i2s_config->ws_io_num = GPIO_NUM_25;
        i2s_config->data_out_num = GPIO_NUM_26;
        i2s_config->data_in_num = GPIO_NUM_35;
    } else {
        memset(i2s_config, -1, sizeof(i2s_pin_config_t));
        KIT_LOGE( "i2s port %d is not supported", port);
        return ESP_FAIL;
    }
    return ESP_OK;
}`

The Module is working with the arduino [audio kit lib](https://github.com/pschatzmann/arduino-audio-tools) by @pschatzmann with board number 5 but the ADF's example is not working, It is giving tap tap tap sound continuous  from speaker.

Can Someone help me regarding the issue.

HengYongChao, 2022-06-02:
Hi @shridattdudhat 

ESP32-A1S is not an espressif officially development board, it is a third-party product, so esp-adf does not currently support it, you can port it according to the [audio hal]() component.

pschatzmann, 2022-06-02:
I never used ADF, but I suggest to use the following site to get help: https://github.com/Ai-Thinker-Open/ESP32-A1S-AudioKit

shridattdudhat, 2022-06-06:
Hi @HengYongChao @pschatzmann 

I have successfully added AiThinker Audio Board in the Custom board component in ADF. There are some changes needs to done in the @pschatzmann 's audio board file to work the microphone. After applying those changes the AiThinker Audio Kit is working perfectly with ADF.

I will share the detailed step here by this week and then will close the issue.

HengYongChao, 2022-06-08:
> Hi @HengYongChao @pschatzmann
>
> I have successfully added AiThinker Audio Board in the Custom board component in ADF. There are some changes needs to done in the @pschatzmann 's audio board file to [...]

@shridattdudhat , It would be wonderful if you can share some steps here.

MrBhola, 2022-06-30:
> 

Hello sir, waiting for the solution. Thanks

pierluigizagaria, 2022-06-30:
@MrBhola if you want to use the line in jack, desolder the two microphones from the board and set LINE1

closed 2022-08-10 by jason-mao

trombik, 2023-07-24:
JFYI: here is my component: https://github.com/trombik/esp-adf-component-ai-thinker-esp32-a1s