espressif/esp-adf #831: ESP_AUDIO_CTRL: Exit play procedure, ret:81001

mike-2020, 2022-05-29:
ESP32, One pipeline, add two streams for input (spiffs and HTTP), one decoder (mp3), one output stream (I2S).

when play mp3 file from HTTP server, it works well.

when play mp3 file from local flash partition, it does not output any sound. It prints this message (as information, not error):

**_ESP_AUDIO_CTRL: Exit play procedure, ret:81001_**

what could cause the non-zero return code? and cause no sound output?

HengYongChao, 2022-05-31:
Hi @mike-2020 

> ESP_AUDIO_CTRL: Exit play procedure, ret:81001

According to the macro definition prompt `[ESP_ERR_AUDIO_NO_INPUT_STREAM](https://github.com/espressif/esp-adf-libs/blob/82b6d534e228f53d75aac825c5fdbdda33a8c33c/esp_audio/include/audio_def.h#L20)` , this error is no input stream, you need to check the multi-input configuration.

The usage routine of multiple input can be seen here, FYI: https://github.com/espressif/esp-adf/tree/master/examples/audio_processing/pipeline_audio_forge

mike-2020, 2022-05-31:
Thanks for the response. I have no intention to do audio forge.

Here is my case:
Generally, my ESP32 will play mp3 by connecting to remote HTTP server. Just in case there is issue with network, my ESP32 will play mp3 in flash (spiffs partition) to tell user there is something wrong.

So, I only created one pipeline, but added two input streams (spiffs and HTTP) to this pipeline. Using this pipeline, it works well when I play mp3 over HTTP with URL looks like this:

http://server/get.mp3

But it returns error above when play mp3 in flash partition with URL looks like this:

file://spiflash/mp3/error.mp3.

BTW: I use function "esp_audio_sync_play".

mike-2020, 2022-05-31:
Here is code for adding two streams:

`    // Create readers and add to esp_audio
    http_stream_cfg_t http_cfg = HTTP_STREAM_CFG_DEFAULT();
    http_cfg.event_handle = _http_stream_event_handle;
    http_cfg.type = AUDIO_STREAM_READER;
    http_cfg.enable_playlist_parser = true;
    http_stream_reader = http_stream_init(&http_cfg);
    esp_audio_input_stream_add(handle, http_stream_reader);

    //create spiffs reader
    spiffs_stream_cfg_t spiffs_cfg = SPIFFS_STREAM_CFG_DEFAULT();
    spiffs_cfg.type = AUDIO_STREAM_READER;
    audio_element_handle_t fs_stream_reader = spiffs_stream_init(&spiffs_cfg);
    rc = esp_audio_input_stream_add(handle, fs_stream_reader);
    if(rc!=ESP_ERR_AUDIO_NO_ERROR){
        ESP_LOGE(TAG, "Failed to add spiffs input stream (%d).", rc);
    }`

Here are logs:

I (1911) HTTP_AUDIO_PLAYER: http_audio_play: play file://spiffs/mp3/power_on.mp3.
I (1911) wifi:I (1921) ESP_AUDIO_CTRL: Enter play procedure, src:0
I (1921) ESP_AUDIO_CTRL: Play procedure, URL is ok, src:0
I (1931) wifi:Init data frame dynamic rx buffer num: 32
I (1931) ESP_AUDIO_CTRL: Request_CMD_Queue CMD:0, Available:5, que:0x3ffd38e8
I (1941) wifi:I (1941) ESP_AUDIO_CTRL: Func:_ctrl_play, Line:771, MEM Total:4322919 Bytes, Inter:209835 Bytes, Dram:207151 Bytes
I (1941) ESP_AUDIO_TASK: It's a decoder
I (1991) ESP_AUDIO_TASK: 1.CUR IN:[IN_file],CODEC:[DEC_mp3],RESAMPLE:[48000],OUT:[OUT_iis],rate:0,ch:0,pos:0
W (2001) ESP_AUDIO_TASK: No more in stream handle
I (2011) ESP_AUDIO_CTRL: Exit play procedure, ret:81001
I (2011) ESP_AUDIO_TASK: Func:media_ctrl_task, Line:782, MEM Total:4244399 Bytes, Inter:166155 Bytes, Dram:163471 Bytes

closed 2022-05-31 by mike-2020